kizniche/Mycodo #1049: Error when calibrating Atlas Scientific pH probe

lu-krin, 2021-07-06:
1. Setup -> Calibration and Setup. 
2. Choose AS pH, 3 point. Input is deactivated.
3. Set the temperature in Celsius (26). Click next.
4. Things break:

`Error 500: Internal Server Error

Something bad happened but it's probably not your fault. Letting the developers know about these issues is crucial to supporting Mycodo. Please submit a new issue on GitHub with the following diagnostic information and error traceback (copy the entire traceback):

Version: 8.11.0
Database: 43c72773dbe8
Model: Raspberry Pi 3 Model B Plus Rev 1.3

Release:
Distributor ID: Raspbian
Description: Raspbian GNU/Linux 10 (buster)
Release: 10
Codename: buster

Firmware:
May 27 2021 14:05:31
Copyright (c) 2012 Broadcom
version 7d9a298cda813f747b51fe17e1e417e7bf5ca94d (clean) (release) (start_x)

Error (Full Traceback):

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1952, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_restx/api.py", line 652, in error_router
    return original_handler(e)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1821, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/_compat.py", line 39, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_login/utils.py", line 272, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_calibration.py", line 515, in setup_atlas_ph
    current_stage='temperature', next_stage=next_stage)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_calibration.py", line 710, in dual_commands_to_sensor
    first_cmd, set_amount=set_amount)
  File "/home/pi/Mycodo/mycodo/utils/atlas_calibration.py", line 103, in calibrate
    err, msg = self.send_command('T,{:.2f}'.format(set_amount))
ValueError: Unknown format code 'f' for object of type 'str'
`

Raspberry Pi 3B+ (1 GB RAM)
Latest Raspbian, latest Mycodo

kizniche, 2021-07-06:
Hi. Calibration from the calibration page will be deprecated in the next release. Therefore, I am not fixing any issues related to it. All calibration should be performed from the Input module itself on the Input page.

Duplicate of #1022

kizniche, 2021-07-06:
We can keep this open. There have been a few of this issue duplicated, so the longer it stays at the top the better.

closed 2021-08-30 by kizniche