kizniche/Mycodo #1051: Error 500: Internal Server Error

keefer223, 2021-07-11:
I was working on a NEW Custom Output for the EMC2101 Fan Controller with temperature monitoring.

I was looking at the example_output_dummy.py and to check out what Mycodo was doing with some of the code:

'custom_actions': [
        {
            'id': 'input_value',
            'type': 'float',
            'name': 'Value Name',
            'phrase': 'A description for this input'
        },
        {
            'id': 'input_button',
            'type': 'button',
            'name': 'Button Name'
        }
    ],

So, I renamed the program name to test_output_dummy.py, no other changes were made. Then imported the file into Mycodo; 
          Configuration > Custom Outputs > Import Output Module
it went in with no errors reported.

Then I went to Setup > Output and tried to add "example_output_dummy" and got the following instead.

**Something bad happened but it's probably not your fault. Letting the developers know about these issues is crucial to supporting Mycodo. Please submit a new issue on GitHub with the following diagnostic information and error traceback (copy the entire traceback):**

Version: 8.11.0
Database: 37509b846483
Model: Raspberry Pi 4 Model B Rev 1.2

Release:
Distributor ID: Raspbian
Description: Raspbian GNU/Linux 10 (buster)
Release: 10
Codename: buster

Firmware:
May 27 2021 14:03:06
Copyright (c) 2012 Broadcom
version 7d9a298cda813f747b51fe17e1e417e7bf5ca94d (clean) (release) (start_x)

Error (Full Traceback):

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1952, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_restx/api.py", line 652, in error_router
    return original_handler(e)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1821, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/_compat.py", line 39, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_login/utils.py", line 272, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_page.py", line 1860, in page_output
    for each_channel in dict_outputs[each_output.output_type]['channels_dict']:
KeyError: 'channels_dict'

I tried to delete example_output_dummy from: Configuration > Custom Outputs > Import Output Module

As soon a I hit delete, I get the following warning:

**Error: Import Output: Cannot delete Output Module if there are still Output entries using it. Delete all Output entries that use this module before deleting the module.**

I did a restart on both the front and backend, which didn't help. I still can't get to the setup > output page. I can still get to the input and function page but I'm locked out of the output page with the above error.

How do we get rid of the error without doing a restore? My backup is more than a week old.

kizniche, 2021-07-11:
Delete all Outputs from the Configure/Diagnostics page

kizniche, 2021-07-11:
Ways export your settings from the Import/Export page before doing anything experimental.

closed 2021-07-14 by kizniche