kizniche/Mycodo #1052: Atlas Dissolved Oxygen

jkstant, 2021-07-11:
### Describe the problem/bug

After adding an Atlas DO: Dissolved Oxygen [I2C] as an Input, I attempt to Calibrate(Air) the sensor and receive the following error, "Error: Action, Controller: 'NoneType' object is not subscriptable".

### Versions:

 - Mycodo Version: 8.11.0
 - Raspberry Pi Version: 4 
 - Raspbian OS Version: Buster (v10)

### Reproducibility

1. Navigate to Setup --> Input
2. Click on 'Input: Select One' and select 'Atlas Scientific: Atlas DO: Dissolved Oxygen... [I2C]'
3. Click 'Add' button
4. Click 'Activate' on new input, 'Atlas DO'
4. Click ' + ' on 'Atlas DO' input to expand settings
5. Click 'Calibrate (Air)' or 'Calibrate (0 mg/L)'
6. Receive Message "Error: Action, Controller: 'NoneType' object is not subscriptable"
7. Unable to calibrate

### Expected behavior

Performs Calibration of the sensor for Air or (0 mg/L)  

### Additional context

The following are entries from the Daemon Log upon adding & activating the Atlas Scientific DO Input:

2021-07-11 19:12:08,055 - INFO - mycodo.devices.atlas_scientific_i2c_105_1 - Atlas Scientific Board: DC, Rev: 2, Firmware: 2.14
2021-07-11 19:12:09,563 - ERROR - mycodo.controllers.controller_input_38a2ebfb - initialize_variables() Exception: could not convert string to float: '0.00,0 0.0'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/base_controller.py", line 74, in run
    self.initialize_variables()
  File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 317, in initialize_variables
    self.measure_input = input_loaded.InputModule(self.input_dev)
  File "/home/pi/Mycodo/mycodo/inputs/atlas_do.py", line 118, in __init__
    self.initialize_input()
  File "/home/pi/Mycodo/mycodo/inputs/atlas_do.py", line 130, in initialize_input
    self.get_measurement()
  File "/home/pi/Mycodo/mycodo/inputs/atlas_do.py", line 168, in get_measurement
    do = float(ec_str)
ValueError: could not convert string to float: '0.00,0 0.0'
2021-07-11 19:12:09,566 - INFO - mycodo.controllers.controller_input_38a2ebfb - Activated in 2277.3 ms
2021-07-11 19:12:09,566 - ERROR - mycodo.controllers.controller_input_38a2ebfb - Mycodo is attempting to acquire measurement(s) from an Input that has already critically errored. Review the log lines following Input Activation to investigate why this happened.

And this is the entry in the Daemon Log when I click 'Calibrate (Air)':

Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 417, in custom_button_exec_function
    run_action = getattr(self.measure_input, button_id)
AttributeError: 'NoneType' object has no attribute 'calibrate_air'

kizniche, 2021-07-12:
So the error occurs before you even attempt to calibrate?

kizniche, 2021-07-12:
>2021-07-11 19:12:08,055 - INFO - mycodo.devices.atlas_scientific_i2c_105_1 - Atlas Scientific Board: DC, Rev: 2, Firmware: 2.14

Isn't the default DO board set to use the I2C address 97 (0x61), not 105? Also, why is the board returning "DC" instead of "DO"?

>ValueError: could not convert string to float: '0.00,0 0.0'

For whatever reason, your board is returning this string in response from a simple query for a measurement. Any idea why it's not returning a single value as the datasheet says it should?

kizniche, 2021-07-12:
Page 58 of the datasheet indicates you may have % saturation enabled, which would cause two values to be returned in a query. Can you disable % sat and see if you can acquire measurements after that?

kizniche, 2021-07-12:
I just pushed a change to the Input that ensures only DO is enabled and % sat is disabled during Input initialization. If you make the [above change](https://github.com/kizniche/Mycodo/commit/8ec4704ace1c79df765d6636cdbfdf703449d204#diff-696f374098798cd9be6bdc827cfe9a818dd5d7a66d7e55286967579c4c249316) to your input file, then restart the Pi, it should be working.

jkstant, 2021-07-12:
To answer your questions, 
 - Yes, the Daemon shows errors upon adding and activation, before calibration
 - Yes, the I2C default address is 97, I was trying to identify the issue and had changed it to 105 as part of troubleshooting, error occurs on any I2C address I use including 97. I dont know why the board is showing DC instead of DO. The only changes I made were to the I2C mode, the board Name and then the new (97) address. 
 -  I will make the changes you suggested to my input file and let you know the result.
Stay tuned! Thanks!!

jkstant, 2021-07-12:
I updated atlas_do.py,  deleted the DO input, reboot PI and added it back as an input. Before rebooting I also set the I2C address back to default, 97, using python script outside of mycodo. .

When I activated it, I received the following in Daemon Log (note it still shows DC rather than DO):
2021-07-12 07:48:17,271 - INFO - mycodo.daemon - Mycodo daemon started in 8.490 seconds
2021-07-12 07:48:17,274 - INFO - mycodo.daemon - 50.69 MB RAM in use
2021-07-12 07:49:55,986 - INFO - mycodo.devices.atlas_scientific_i2c_97_1 - Atlas Scientific Board: DC, Rev: 2, Firmware: 2.14
2021-07-12 07:49:55,989 - INFO - mycodo.inputs.atlas_do_8c53cae1 - Parameters enabled: None
2021-07-12 07:49:57,501 - ERROR - mycodo.controllers.controller_input_8c53cae1 - initialize_variables() Exception: could not convert string to float: '0.00,0( &'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/base_controller.py", line 74, in run
    self.initialize_variables()
  File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 317, in initialize_variables
    self.measure_input = input_loaded.InputModule(self.input_dev)
  File "/home/pi/Mycodo/mycodo/inputs/atlas_do.py", line 118, in __init__
    self.initialize_input()
  File "/home/pi/Mycodo/mycodo/inputs/atlas_do.py", line 136, in initialize_input
    self.get_measurement()
  File "/home/pi/Mycodo/mycodo/inputs/atlas_do.py", line 174, in get_measurement
    do = float(ec_str)
ValueError: could not convert string to float: '0.00,0( &'
2021-07-12 07:49:57,510 - INFO - mycodo.controllers.controller_input_8c53cae1 - Activated in 2329.3 ms
2021-07-12 07:49:57,511 - ERROR - mycodo.controllers.controller_input_8c53cae1 - Mycodo is attempting to acquire measurement(s) from an Input that has already critically errored. Review the log lines following Input Activation to investigate why this happened.

I also hit Calibration button again and it showed the following errors (as I would expect give above), in Daemon Log:

2021-07-12 07:55:25,537 - ERROR - mycodo.controllers.controller_input_8c53cae1 - Mycodo is attempting to acquire measurement(s) from an Input that has already critically errored. Review the log lines following Input Activation to investigate why this happened.
2021-07-12 07:55:39,131 - ERROR - mycodo.controllers.controller_input_8c53cae1 - Error executing button press function 'calibrate_air'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 417, in custom_button_exec_function
    run_action = getattr(self.measure_input, button_id)
AttributeError: 'NoneType' object has no attribute 'calibrate_air'
2021-07-12 07:55:40,472 - ERROR - mycodo.controllers.controller_input_8c53cae1 - Mycodo is attempting to acquire measurement(s) from an Input that has already critically errored. Review the log lines following Input Activation to investigate why this happened.

jkstant, 2021-07-12:
i discovered my new Atlas Scientific EC sensor was not showing up in i2cdetect and on a hunch, I removed it from my Tentacle T3 and now the DO is working and showing the correct device type, DO.

I purchased both of these and installed at the same time however I had not tested the EC due to the DO issues I encountered. I wonder if I have a bad EC  embedded circuit. I will do some further troubleshooting to determine if its the board.

stay tuned.
thanks

kizniche, 2021-07-12:
You can see from the error your response is malformed. You should always test your devices in isolation if you encounter issues. Your EC board is likely set to the same I2C address as the DO board.

jkstant, 2021-07-12:
I can see how it would appear that way but they aren't actually. That is the strange thing about this. They are both on their default addresses and it still occurs 0x61(DO) and 0x64(EC). I am going to test the EC on a separate rasb-pi in isolation as you said and see what happens

jkstant, 2021-07-15:
turns out it was a bad DO board. Thanks for the help.

closed 2021-07-15 by jkstant