kizniche/Mycodo #1053: MQTT input stop receiving input after a few days

garudaonekh, 2021-07-14:
Has anyone experience the same issue?

My MQTT input stop receiving data after a few days.
I restart the input and it works normally.

My Input MQTT Client ID is unique. And the MQTT server is working fine and the sensor(Tasmota) keep sending the telemetry data.

kizniche, 2021-07-14:
loop_start() handles reconnects, per the 9documentation](https://pypi.org/project/paho-mqtt/#network-loop).

https://github.com/kizniche/Mycodo/blob/fe38478a81c1aed54ffd017480e7ecb0cc6fb5eb/mycodo/inputs/mqtt_paho.py#L185

garudaonekh, 2021-07-14:
It works after I change the client_id again.
I think the most likely reason is I restart this input some days ago and it's not properly disconnect from the MQTT server so when I restart the input again it connect with the same client_id that still left connected/subscribed.

garudaonekh, 2021-07-24:
still unable to find the root cause. Do as what you said but nothing abnormal. Restart Mycodo daemon and it works fine till now.

kizniche, 2021-07-24:
You're saying there are no disconnects logged with Log Level: Debug enabled?

garudaonekh, 2021-07-24:
No, it didn't disconnect but simply quiet. Yes, I enable debug and see the connected and subscribed log...

garudaonekh, 2021-12-17:
I still facing the issue and still no clue.
Here's my scenario:
A controller publish mqtt mesage of multiple sensors(SHT30, Light and Pressure Sensor), all in the same mqtt message.
For ease of use on the Mycodo side, I create 3 MQTT Input(1 for SHT30, 1 for Light and 1 for pressure). Both Light and Pressure sensor work fine, except the SHT30 mqtt Input(It stop receiving data after sometimes until I restart the mqtt input again).

dylandn, 2022-01-02:
I am having MQTT input issues as well where the input stops receiving anything until I Deactivate/Activate it again. 
In the log:
2022-01-02 13:07:23,973 - INFO - mycodo.function.bang_bang_0ef487c1 - Input: 55.1, output: on, target: 55.0, hyst: 0.1
2022-01-02 13:07:25,568 - INFO - mycodo.function.bang_bang_7b881fb3 - Input: 18.0, output: on, target: 18.0, hyst: 1.0
2022-01-02 13:07:25,901 - DEBUG - mycodo.inputs.mqtt_paho_b4af85b4 - Disconnected. Return code: 0
2022-01-02 13:07:25,902 - INFO - mycodo.controllers.controller_input_b4af85b4 - Deactivated in 842.2 ms
2022-01-02 13:07:27,698 - DEBUG - mycodo.inputs.mqtt_paho_b4af85b4 - Client created with ID mycodo_mqtt_client
2022-01-02 13:07:27,698 - DEBUG - mycodo.controllers.controller_input_b4af85b4 - Detected as listener. Starting listener thread.
2022-01-02 13:07:27,699 - DEBUG - mycodo.inputs.mqtt_paho_b4af85b4 - Connecting MQTT callback functions
2022-01-02 13:07:27,699 - DEBUG - mycodo.inputs.mqtt_paho_b4af85b4 - MQTT callback functions connected
2022-01-02 13:07:27,708 - INFO - mycodo.inputs.mqtt_paho_b4af85b4 - Connected to localhost as mycodo_mqtt_client
2022-01-02 13:07:27,709 - DEBUG - mycodo.inputs.mqtt_paho_b4af85b4 - Subscribing to MQTT topic 'Dew Temperature'
2022-01-02 13:07:27,711 - INFO - mycodo.controllers.controller_input_b4af85b4 - Activated in 364.2 ms
2022-01-02 13:07:27,711 - DEBUG - mycodo.inputs.mqtt_paho_b4af85b4 - Connected to 'None'. Return code: 0
2022-01-02 13:07:27,712 - DEBUG - mycodo.inputs.mqtt_paho_b4af85b4 - Subscribed to mqtt topic: None, 1, (0,)
2022-01-02 13:07:28,864 - INFO - mycodo.function.bang_bang_0ef487c1 - Input: 54.8, output: on, target: 55.0, hyst: 0.1
2022-01-02 13:07:28,881 - DEBUG - mycodo.inputs.mqtt_paho_b4af85b4 - Disconnected. Return code: 1
2022-01-02 13:07:29,900 - DEBUG - mycodo.inputs.mqtt_paho_b4af85b4 - Connected to 'None'. Return code: 0
2022-01-02 13:07:30,465 - INFO - mycodo.function.bang_bang_7b881fb3 - Input: 18.0, output: on, target: 18.0, hyst: 1.0
2022-01-02 13:07:33,868 - INFO - mycodo.function.bang_bang_0ef487c1 - Input: 54.5, output: off, target: 55.0, hyst: 0.1

Some times it breaks before receiving any data, other times it runs for a few minutes before getting disconnected:
2022-01-02 13:12:33,796 - DEBUG - mycodo.inputs.mqtt_paho_b4af85b4 - Received message: topic: espAC/airDew, payload:  9.1
2022-01-02 13:12:33,797 - DEBUG - mycodo.inputs.mqtt_paho_b4af85b4 - Found channel 0 with topic 'espAC/airDew'
2022-01-02 13:12:33,803 - DEBUG - mycodo.inputs.mqtt_paho_b4af85b4 - Payload represents a float: 9.1
2022-01-02 13:12:33,814 - DEBUG - mycodo.inputs.mqtt_paho_b4af85b4 - Adding measurement to influxdb: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 9.1, 'timestamp_utc': datetime.datetime(2022, 1, 2, 21, 12, 33, 797312)}}
2022-01-02 13:12:34,007 - DEBUG - mycodo.inputs.mqtt_paho_b4af85b4 - Received message: topic: espAC/airDew, payload:  9.1
2022-01-02 13:12:34,007 - DEBUG - mycodo.inputs.mqtt_paho_b4af85b4 - Found channel 0 with topic 'espAC/airDew'
2022-01-02 13:12:34,007 - DEBUG - mycodo.inputs.mqtt_paho_b4af85b4 - Payload represents a float: 9.1
2022-01-02 13:12:34,011 - DEBUG - mycodo.inputs.mqtt_paho_b4af85b4 - Adding measurement to influxdb: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 9.1, 'timestamp_utc': datetime.datetime(2022, 1, 2, 21, 12, 34, 7611)}}
2022-01-02 13:12:34,024 - INFO - mycodo.function.bang_bang_0ef487c1 - Input: 55.1, output: off, target: 55.0, hyst: 0.1
2022-01-02 13:12:35,933 - INFO - mycodo.function.bang_bang_7b881fb3 - Input: 18.3, output: on, target: 18.0, hyst: 1.0
2022-01-02 13:12:36,416 - DEBUG - mycodo.inputs.mqtt_paho_b4af85b4 - Received message: topic: espAC/airDew, payload:  9.1
2022-01-02 13:12:36,416 - DEBUG - mycodo.inputs.mqtt_paho_b4af85b4 - Found channel 0 with topic 'espAC/airDew'
2022-01-02 13:12:36,417 - DEBUG - mycodo.inputs.mqtt_paho_b4af85b4 - Payload represents a float: 9.1
2022-01-02 13:12:36,422 - DEBUG - mycodo.inputs.mqtt_paho_b4af85b4 - Adding measurement to influxdb: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 9.1, 'timestamp_utc': datetime.datetime(2022, 1, 2, 21, 12, 36, 416705)}}
2022-01-02 13:12:36,619 - DEBUG - mycodo.inputs.mqtt_paho_b4af85b4 - Received message: topic: espAC/airDew, payload:  9.2
2022-01-02 13:12:36,620 - DEBUG - mycodo.inputs.mqtt_paho_b4af85b4 - Found channel 0 with topic 'espAC/airDew'
2022-01-02 13:12:36,620 - DEBUG - mycodo.inputs.mqtt_paho_b4af85b4 - Payload represents a float: 9.2
2022-01-02 13:12:36,626 - DEBUG - mycodo.inputs.mqtt_paho_b4af85b4 - Adding measurement to influxdb: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 9.2, 'timestamp_utc': datetime.datetime(2022, 1, 2, 21, 12, 36, 620291)}}
2022-01-02 13:12:38,799 - DEBUG - mycodo.inputs.mqtt_paho_b4af85b4 - Received message: topic: espAC/airDew, payload:  9.2
2022-01-02 13:12:38,799 - DEBUG - mycodo.inputs.mqtt_paho_b4af85b4 - Found channel 0 with topic 'espAC/airDew'
2022-01-02 13:12:38,800 - DEBUG - mycodo.inputs.mqtt_paho_b4af85b4 - Payload represents a float: 9.2
2022-01-02 13:12:38,804 - DEBUG - mycodo.inputs.mqtt_paho_b4af85b4 - Adding measurement to influxdb: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 9.2, 'timestamp_utc': datetime.datetime(2022, 1, 2, 21, 12, 38, 799866)}}
2022-01-02 13:12:38,937 - INFO - mycodo.function.bang_bang_0ef487c1 - Input: 55.2, output: off, target: 55.0, hyst: 0.1
2022-01-02 13:12:38,955 - DEBUG - mycodo.inputs.mqtt_paho_b4af85b4 - Disconnected. Return code: 1
2022-01-02 13:12:39,974 - DEBUG - mycodo.inputs.mqtt_paho_b4af85b4 - Connected to 'None'. Return code: 0
2022-01-02 13:12:40,582 - INFO - mycodo.function.bang_bang_7b881fb3 - Input: 18.3, output: on, target: 18.0, hyst: 1.0
(no more messages are received at this point)

dylandn, 2022-01-02:
Using an MQTT client on my phone, I can see that the sensor is indeed publishing the payload, while Mycodo stops receiving them after the Disconnections.

dylandn, 2022-01-02:
Hmm.. I have 4 MQTT inputs, the other 4 work find and they have unique 'Client ID' set, the 4th one had the default Client ID which wasnt being used by any other, but changing the 4th Client ID to something else and so far it's hasn't stopped receiving data yet.

garudaonekh, 2022-01-03:
> Hmm.. I have 4 MQTT inputs, the other 3 work fine and they have unique 'Client ID' set, the 4th one had the default Client ID which wasn't being used by any other, but changing the 4th Client ID to something else and so far it's hasn't stopped receiving data yet.

For me, it's not duplicate client_id issue. My client id is unique.