kizniche/Mycodo #1054: AJAX forms don't work on mobile Safari browser

dookaloosy, 2021-07-18:
### Describe the problem/bug

AJAX-enabled interface doesn't work on mobile Safari browser (iOS 12.1.4). Inputs cannot be activated or deactivated from the browser.

### Versions:

 - Mycodo Version: 8.11.0 + master branch commit [8d46745](https://github.com/kizniche/Mycodo/commit/8d46745b8439abad903f49eb9da0e27756e8fdca)
 - Raspberry Pi Version: 3B
 - Raspbian OS Version: Linux raspberrypi 5.10.17-v7+ #1403 SMP Mon Feb 22 11:29:51 GMT 2021 armv7l GNU/Linux

### Reproducibility

Please list specific setup details that are involved and the steps to reproduce the behavior:

1. Upgrade to master branch commit [8d46745](https://github.com/kizniche/Mycodo/commit/8d46745b8439abad903f49eb9da0e27756e8fdca)
2. Browse to Inputs page on mobile Safari browser (iOS 12.1.4)
3. Click on Activate or Deactivate.
4. Notice that the activation status of the Input does not change.

### Expected behavior

Activate and Deactivate should work as on non-AJAX interface.

### Screenshots

N/A

### Additional context

N/A

kizniche, 2021-07-19:
This has been a real pain, but I finally got a mac virtual machine running and found a solution. It's going to require refactoring all the AJAX inputs and form event functions.

kizniche, 2021-07-19:
I just committed fixes for the Input/Output/Function pages for Safari. Let me know if that fixes everything.

dookaloosy, 2021-07-20:
Upgraded to this commit. It appears to fix everything. Awesome work!

kizniche, 2021-07-20:
Great! Also, how is the usability of the new system? Any comments or suggestions?

dookaloosy, 2021-07-20:
Absolutely better in my opinion: speed and responsiveness are way higher. I'm sure there are more bugs to sort through as we go but it's definitely working well so far.

closed 2021-07-23 by kizniche