kizniche/Mycodo #1055: AJAX interface does not show pylint output

dookaloosy, 2021-07-18:
### Describe the problem/bug

AJAX-enabled interface does not show the pylint output when saving Conditional Functions, only the pylint return code. This makes it more difficult to debug the Conditional python code.

### Versions:

 - Mycodo Version: 8.11.0 + master branch commit [8d46745](https://github.com/kizniche/Mycodo/commit/8d46745b8439abad903f49eb9da0e27756e8fdca)
 - Raspberry Pi Version: 3B
 - Raspbian OS Version: Linux raspberrypi 5.10.17-v7+ #1403 SMP Mon Feb 22 11:29:51 GMT 2021 armv7l GNU/Linux

### Reproducibility

Please list specific setup details that are involved and the steps to reproduce the behavior:

1. Upgrade to master branch commit [8d46745](https://github.com/kizniche/Mycodo/commit/8d46745b8439abad903f49eb9da0e27756e8fdca)
2. Edit Conditional Function code.
3. Click on Save after injecting error into python code.
4. If pylint has a non-zero return code, it does not show the pylint output.

### Expected behavior

The pylint output should show as on non-AJAX interface, identifying the lines where errors were detected.

### Screenshots

N/A

### Additional context

N/A

kizniche, 2021-07-18:
Duplicate of #1054, a bug in Safari

kizniche, 2021-07-18:
Ah, I see the, issue. I'll fix shortly.

closed 2021-07-18 by kizniche