kizniche/Mycodo #1056: AJAX interface unreliable when adding Conditions and Actions in Conditional Function

dookaloosy, 2021-07-18:
### Describe the problem/bug

AJAX-enabled interface breaks when adding Conditions while editing Conditional Function. While a new Condition is added, the form doesn't display a dropdown for the new Condition. A Ctrl+R page refresh must be forced in the browser (Chrome) to display the new dropdown. 

When adding multiple Actions in the Conditional Function, a new dropdown is displayed for the new Action. After setting and saving this first action, if a second Action is added, it clears the previous dropdown's displayed setting. A Ctrl+R forced page refresh will cause the previous dropdown to display correctly. 

### Versions:

 - Mycodo Version: 8.11.0 + master branch commit [8d46745](https://github.com/kizniche/Mycodo/commit/8d46745b8439abad903f49eb9da0e27756e8fdca)
 - Raspberry Pi Version: 3B
 - Raspbian OS Version: Linux raspberrypi 5.10.17-v7+ #1403 SMP Mon Feb 22 11:29:51 GMT 2021 armv7l GNU/Linux

### Reproducibility

Please list specific setup details that are involved and the steps to reproduce the behavior:

1. Upgrade to master branch commit [8d46745](https://github.com/kizniche/Mycodo/commit/8d46745b8439abad903f49eb9da0e27756e8fdca)
2. Browse to Conditional Function and edit it.
3. Add a new Condition. Notice that the dropdown is missing.
4. Force a page refresh in the browser. The dropdown appears.
5. Add a new Action. The dropdown appears, set and save it.
6. Add another Action. Notice the setting in the previously saved dropdown is cleared.
7. Force a page refresh in the browser. The previously saved dropdown is displayed correctly.

### Expected behavior

Dropdowns should appear when added and be stable when saved.

### Screenshots

N/A

### Additional context

N/A

kizniche, 2021-07-18:
Duplicate of #1054, a bug in Safari

dookaloosy, 2021-07-18:
Actually, this was encountered on desktop Chrome (Windows7).

kizniche, 2021-07-18:
Just found and fixed the issue. Should work as expected now.

closed 2021-07-18 by kizniche