kizniche/Mycodo #1060: Issue Regarding Float Switch and Relay

justbest23, 2021-07-20:
Thank you for making this interface, it's exactly what I was looking for. 

I am having trouble using a float switch to activate a relay. Using both the GPIO On/Off Output Function and the Python function gives the same result. When the float switch is down (0 bool) the relay activates for 0.8 seconds and then turns off for 2 seconds, and then repeats.

Current path is GPIO State > PID Controller (Raising from 0 to 1) > Outputs mentioned above

Any help would be greatly appreciated!

kizniche, 2021-07-20:
I don't understand the use of a PID controller. Can you provide more information? What you're trying to achieve, specific settings, etc.?

justbest23, 2021-07-20:
I'm trying to use a float switch to activate a relay which will in turn open a solenoid and start refilling the system. It is extremely likely that I am just not seeing the correct option in the Funtions tab, I'm very new to this whole IoT thing. Is there a better way to do what I'm doing? I tried to follow your tutorial (which is extremely good) but you only used your float switch to send an email. Attached is the PID settings page. Not sure if that helps
[image]

kizniche, 2021-07-21:
For that purpose, a PID is not what you should be using. A PID requires an input with a wide range of variability that you want to stabilize. What you are trying to do is simply turn on an output (open water valve) when a circuit (float switch) is closed (or opened) until that circuit changes to the opposite state. The most simple way to accomplish this is the following:

 - Input: Raspberry Pi GPIO State, set to the pin for your float sensor
 - Output: GPIO On/Off, set to the pin of your water valve
 - Function: Conditional Controller
   - Set the Period to how often you want to check (and possibly turn on the valve)
   - Condition: Measurement (Single, Last), set to your Input
   - Action: Output (On/Off/Duration), set to your Output, State to On, and Duration to the number of seconds you want the output on.
   - Conditional Statement:

```python
measurement = self.condition("{55ffc791}")  # Replace with the ID shown next to your Condition
if measurement is not None and measurement == 1:  # Change to 0 if 1 produces the opposite of the desired effect
    self.run_action("{7bd0129a}")  # Replace with the ID shown next to your Action
```

justbest23, 2021-07-21:
Ah yea I figured I was probably using the wrong thing. I'll give this a go in the morning. Thanks! Do you have a discord or similar that would make communication easier? Or should I just be posting here if I have further questions?

kizniche, 2021-07-21:
The forum or discord is a better place for questions (see below). Try to reserve bugs for github issues.

https://github.com/kizniche/Mycodo#discussion

closed 2021-08-30 by kizniche