kizniche/Mycodo #1061: Serial Bridge with MQTT

garudaonekh, 2021-07-22:
Hi,
I have sensors which communicate with ESP32 thru serial. I used Tasmota on that ESP32. Here's the reading process:
1. Send the reading command thru HTTP command to the sensors via ESP32.
2. The tasmota on esp32 will response the result back thru MQTT publishing.

What's the suggestion to implement this? I see the MQTT Input does the job of receiving the MQTT response but we miss the sending the HTTP request part. Can we use the get_measurement() of input module to issue the HTTP request but get_measurement need to return data but in this scenario there's no data yet?

My idea is to add get_measurement() into the MQTT Input but get_measurement here will have no data yet.

maecki-maecki, 2021-07-24:
In one case I use a bash command input for my tasmota sensors with mycodo (without mqtt) perhaps this might work for you in a similar manner?

`
curl -sS http://tasmota-temperatur-1.ka.maerkle.de/cm?cmnd=status%2010 | jq -r '.StatusSNS."DS18B20-1".Temperature' `

garudaonekh, 2021-07-24:
Thanks for your suggestion, but Tasmota serial bridge will response in mQTT not HTtp response. Anyhow, I now use Tasmota rule in combination with MQTT

closed 2021-08-03 by garudaonekh