kizniche/Mycodo #1064: Atlas Scientific PH and EC sensor temperature compensation

treesnsnacks, 2021-08-02:
### Describe the problem/bug

Atlas scientific PH and EC sensors display wrong value when using temperature compensation from a sensor who has had its value converting from Celsius to Fahrenheit in the input measurement settings.

### Versions:

 - Mycodo Version: 8.11.0
 - Raspberry Pi Version: 4B 4gb
 - Raspbian OS Version: Buster 5.10.17
### Reproducibility

Please list specific setup details that are involved and the steps to reproduce the behavior:

1. Setup the Atlas Scientific PH EC and Temperature sensors as normal and get readings on calibration solutions
2. Add the temperature sensor as temperature compensation measurement for the PH and EC inputs 
3. Change the temperature sensor to convert from Celsius to Fahrenheit  
4. Watch the readings change to inaccurate levels, particularly the EC which was almost halved in my case

### Expected behavior

Expected the EC and PH to vary slightly compared to the measurement pre temperature compensation. The PH might be fine as it only changed .2 and I have no way to double check it but the EC halved and I checked with a separate tester which confirmed it was way off. Switching back to the unconverted Celsius shows EC readings where they should be.  I suspect its treating the new Fahrenheit values as still Celsius or an error in the conversion somewhere. 

### Screenshots

If applicable, add screenshots to help explain your problem.

### Additional context

There is also a setup bug for Atlas scientific EC sensors that might prevent sensor from reporting any value which could prevent testing of this bug, writing a separate bug report.

kizniche, 2021-08-02:
Thanks for the bug report. I can confirm there is no conversion performed in the Input module. Until a fix can be made, use the unit Celsius.

kizniche, 2021-08-07:
I just committed changes that should fix this. If you upgrade to master to test, let me know if it works as expected. Thanks.

treesnsnacks, 2021-08-14:
Tried today and it looks right, but haven't messed with it too much or double checked readings with another sensor.

closed 2021-08-30 by kizniche