kizniche/Mycodo #1065: Atlas Scientific EC sensor setup bug

treesnsnacks, 2021-08-02:
### Describe the problem/bug

Atlas scientific EC sensor wont report a value on its default settings and doesn't support all the measurements the sensor can report.

### Versions:

 - Mycodo Version: 8.11.0
 - Raspberry Pi Version: 4B 4gb
 - Raspbian OS Version: 5.10.17
### Reproducibility

Please list specific setup details that are involved and the steps to reproduce the behavior:

1. Connect EC sensor and verified it shows up on i2c device list
2. Attempt to get reading and show no value

### Expected behavior

Expected conductivity, total dissolved solids, salinity, and specific gravity values to be reported as that's what the data sheet online shows the circuit reports. Need to update Atlas Scientific EC input to support multiple values and maybe a note somewhere about the difference in default settings between newer and older board which prevents using older boards with their default settings.

### Screenshots

If applicable, add screenshots to help explain your problem.

### Additional context

After accidently getting it to report on the humidity sensor input that supports multiple values it became clear that the EC input was only expecting one value which was why it wouldn't work. After reading the datasheet more I saw that newer boards only report EC by default until you enable the other values so I disabled the all the values except EC on my board and was able to get it to work under the proper atlas scientific EC input but only showing the EC value.

kizniche, 2021-08-02:
Thanks for the bug report. What firmware version is your sensor? This can be found in the daemon log after activating the input. I can add an additional Input for newer firmwares that include the additional values.

treesnsnacks, 2021-08-02:
Its 2.15 but from my understanding they all support the same values just any firmware above 2.10 by DEFAULT only shows EC and you need to manually turn on the rest, I must have switched the settings in the past as I have a newer firmware that should have only reported EC and I messed with the sensor a bit before finding your software. So not sure if necessary to make a new input maybe just update the old one to have the option to turn off certain values like you have in the humidity sensor.

kizniche, 2021-08-02:
The new Input will select the values to return upon activation (for example, like the Atlas RGB sensor), based on what measurements the user selects to record. The original input cannot be just modified to include new measurements, as there are a number of users already using this input. Additionally, there cannot be the option to select measurements that don't exist for the old sensor firmware. And since Mycodo cannot know the firmware version prior to activation, there's no other way for Mycodo to render the appropriate options to the user unless there are two different Inputs that the user can select from.

kizniche, 2021-08-07:
I just committed changes that should fix this. If you upgrade to master to test, test with a single measurement enabled as well as multiple measurements, as well as in different combinations, and let me know if it works as expected. Thanks.

kizniche, 2021-08-07:
You'll need to delete your current Input and add a new Input to see the new measurement selections.

treesnsnacks, 2021-08-14:
Tried it out today and its still not working. EC alone still works but once u add other measurements none report anything. It looks like it is sending the command and configuring the sensor to output the right measurements, every time i changed settings and reactivated i was checking the values on atlas's sample code and they would show the new measurements. But on mycodo live view the measurements all show 0 unless i go back to only using EC.

kizniche, 2021-08-14:
Can you provide the debug log line that shows what's returned from the sensor when all measurements are enabled?

kizniche, 2021-08-14:
I just committed a change to the Input that adds a few more debug logging lines, if you want to update to master and run the Input with Log Level: Debug enabled to see some more information about which measurements get enabled and the return values.

closed 2021-08-30 by kizniche