kizniche/Mycodo #1066: support MQTT publish ?

haidv1992, 2021-08-08:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

haidv1992, 2021-08-08:
Currently I don't see mqtt publish support
, can I add it myself ? hope you can support the source material to add this feature

kizniche, 2021-08-09:
Search "MQTT" at https://kizniche.github.io/Mycodo/

haidv1992, 2021-08-09:
Thank you very much

closed 2021-08-09 by kizniche