kizniche/Mycodo #1067: Error 502: Bad Gateway Error

juleszimmermann, 2021-08-09:
Describe the problem/bug

Unable to have an access to the mycodo frontend. I have already one mycodo system which works perfectly but this new one fail 

Versions:

Mycodo Version: 8.11.0
Raspberry Pi Version: 4B 4gb
Raspbian OS Version: 5.10.17

Reproducibility
Please list specific setup details that are involved and the steps to reproduce the behavior:

Install Mycodo from scratch with a new SD card on a Raspberry pi 4 with the official software

Screenshots

[image]

After typing the command to get logs i'm getting that. I guess the issue come from the unreachable "localhost".

pi@raspberrypi:~ $ journalctl -u mycodoflask | tail -n 50
-- Logs begin at Fri 2021-05-07 16:07:11 BST, end at Tue 2021-08-10 00:00:12 BST. --
Aug 09 23:46:14 raspberrypi systemd[1]: Started gunicorn daemon for Mycodo flask web user interface.
Aug 09 23:46:14 raspberrypi systemd[3125]: mycodoflask.service: Failed to execute command: No such file or directory
Aug 09 23:46:14 raspberrypi systemd[3125]: mycodoflask.service: Failed at step EXEC spawning /var/mycodo-root/env/bin/gunicorn: No such file or directory
Aug 09 23:46:14 raspberrypi systemd[1]: mycodoflask.service: Main process exited, code=exited, status=203/EXEC
Aug 09 23:46:14 raspberrypi systemd[1]: mycodoflask.service: Failed with result 'exit-code'.
pi@raspberrypi:~ $

kizniche, 2021-08-09:
Your virtualenv is not set up correctly if gunicorn doesn't exist. Try installing again.

kizniche, 2021-08-09:
Just run ```sudo ~/Mycodo/install/setup.sh```

juleszimmermann, 2021-08-09:
I just tried, same as before, it's trying to connect to localhost but it fails
[image]

kizniche, 2021-08-09:
What are the errors in the setup log?

kizniche, 2021-08-09:
~/Mycodo/install/setup.log
Please don't paste the entire log here, attach it.

juleszimmermann, 2021-08-09:
[setup.log](https://github.com/kizniche/Mycodo/files/6957975/setup.log)

kizniche, 2021-08-10:
It appears to be an issue installing the virtualenv package:

```
#### Virtualenv doesn't exist. Creating...
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting virtualenv
  ERROR: HTTP error 404 while getting https://www.piwheels.org/simple/virtualenv/virtualenv-20.7.1-py2.py3-none-any.whl#sha256=f101753071fe8df994f78049016a98f9d2af65443df96e31cabe999d21eef0ca (from https://www.piwheels.org/simple/virtualenv/) (requires-python:!=3.0.*,!=3.1.*,!=3.2.*,!=3.3.*,!=3.4.*,>=2.7)
ERROR: Could not install requirement virtualenv from https://www.piwheels.org/simple/virtualenv/virtualenv-20.7.1-py2.py3-none-any.whl#sha256=f101753071fe8df994f78049016a98f9d2af65443df96e31cabe999d21eef0ca because of HTTP error 404 Client Error: Not Found for url: https://www.piwheels.org/simple/virtualenv/virtualenv-20.7.1-py2.py3-none-any.whl for URL https://www.piwheels.org/simple/virtualenv/virtualenv-20.7.1-py2.py3-none-any.whl#sha256=f101753071fe8df994f78049016a98f9d2af65443df96e31cabe999d21eef0ca (from https://www.piwheels.org/simple/virtualenv/) (requires-python:!=3.0.*,!=3.1.*,!=3.2.*,!=3.3.*,!=3.4.*,>=2.7)
```

juleszimmermann, 2021-08-10:
yep ! i'm trying to reinstall it, it seems that some package are unreachable, so i installed it manually from the github repo

juleszimmermann, 2021-08-10:
basic install with "sudo pip3 install virtualenv" failed, i'm not sure if it comes from my side or not

pi@raspberrypi:~ $ sudo pip3 install virtualenv
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting virtualenv
  ERROR: HTTP error 404 while getting https://www.piwheels.org/simple/virtualenv/virtualenv-20.7.1-py2.py3-none-any.whl#sha256=f101753071fe8df994f78049016a98f9d2af65443df96e31cabe999d21eef0ca (from https://www.piwheels.org/simple/virtualenv/) (requires-python:!=3.0.*,!=3.1.*,!=3.2.*,!=3.3.*,!=3.4.*,>=2.7)
ERROR: Could not install requirement virtualenv from https://www.piwheels.org/simple/virtualenv/virtualenv-20.7.1-py2.py3-none-any.whl#sha256=f101753071fe8df994f78049016a98f9d2af65443df96e31cabe999d21eef0ca because of HTTP error 404 Client Error: Not Found for url: https://www.piwheels.org/simple/virtualenv/virtualenv-20.7.1-py2.py3-none-any.whl for URL https://www.piwheels.org/simple/virtualenv/virtualenv-20.7.1-py2.py3-none-any.whl#sha256=f101753071fe8df994f78049016a98f9d2af65443df96e31cabe999d21eef0ca (from https://www.piwheels.org/simple/virtualenv/) (requires-python:!=3.0.*,!=3.1.*,!=3.2.*,!=3.3.*,!=3.4.*,>=2.7)

kizniche, 2021-08-10:
See if this works:

```
~/Mycodo/env/bin/pip install --upgrade setuptools
~/Mycodo/env/bin/pip install virtualenv
```

juleszimmermann, 2021-08-10:
Same error as before

#### Checking python 3 virtualenv
#### Virtualenv doesn't exist. Creating...
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting virtualenv
  ERROR: HTTP error 404 while getting https://www.piwheels.org/simple/virtualenv/virtualenv-20.7.1-py2.py3-none-any.whl#sha256=f101753071fe8df994f78049016a98f9d2af65443df96e31cabe999d21eef0ca (from https://www.piwheels.org/simple/virtualenv/) (requires-python:!=3.0.*,!=3.1.*,!=3.2.*,!=3.3.*,!=3.4.*,>=2.7)
ERROR: Could not install requirement virtualenv from https://www.piwheels.org/simple/virtualenv/virtualenv-20.7.1-py2.py3-none-any.whl#sha256=f101753071fe8df994f78049016a98f9d2af65443df96e31cabe999d21eef0ca because of HTTP error 404 Client Error: Not Found for url: https://www.piwheels.org/simple/virtualenv/virtualenv-20.7.1-py2.py3-none-any.whl for URL https://www.piwheels.org/simple/virtualenv/virtualenv-20.7.1-py2.py3-none-any.whl#sha256=f101753071fe8df994f78049016a98f9d2af65443df96e31cabe999d21eef0ca (from https://www.piwheels.org/simple/virtualenv/) (requires-python:!=3.0.*,!=3.1.*,!=3.2.*,!=3.3.*,!=3.4.*,>=2.7)
/home/pi/Mycodo/mycodo/scripts/upgrade_commands.sh: line 200: virtualenv: command not found

#### Updating pip
/home/pi/Mycodo/mycodo/scripts/upgrade_commands.sh: line 443: /home/pi/Mycodo/env/bin/pip3: No such file or directory

#### Installing pip requirements

## Error: Virtualenv doesn't exist. Create with /home/pi/Mycodo/mycodo/scripts/upgrade_commands.sh setup-virtualenv
--2021-08-10 02:36:13--  https://project-downloads.drogon.net/wiringpi-latest.deb
Resolving project-downloads.drogon.net (project-downloads.drogon.net)... 188.246.205.22, 2a03:9800:10:7b::2
Connecting to project-downloads.drogon.net (project-downloads.drogon.net)|188.246.205.22|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 52260 (51K) [application/x-debian-package]
Saving to: ‘wiringpi-latest.deb’

kizniche, 2021-08-10:
That doesn't appear to be the output of the two commands I provided, as it has the attempted execution of "/home/pi/Mycodo/mycodo/scripts/upgrade_commands.sh" in the middle.

Can you copy the output from the two commands?

kizniche, 2021-08-10:
virtualenv 20.7.1 was released 12 hours ago, and I don't see any issues on its github. It appears to be some issue with that package, not pip or Mycodo. You can attempt to install the previous version rather than the latest by doing the following:

Change this line:

https://github.com/kizniche/Mycodo/blob/8fab275ae4997509be240231c98704baa2f3b310/mycodo/scripts/upgrade_commands.sh#L198

to this:

```python
            pip install virtualenv==20.7.0
```

Save, then run setup.sh.

jess-parent, 2021-08-10:
I was experiencing the same issue, and updating to install virtualenv=20.7.0 worked for me, seems to be up and running now. Thanks!

juleszimmermann, 2021-08-10:
Installing virtualenv=20.7.0 works for me too, it works perfectly now , thanks !

closed 2021-08-30 by kizniche