kizniche/Mycodo #1072: Camera widget cannot get most recent timelapse image if custom path is defined

kimbernator, 2021-08-18:
### Describe the problem/bug

I have a camera defined with a custom timelapse path. When I configure a camera widget on my dashboard and select "most recent timelapse image", it will never find a valid one because it doesn't flex on the custom path. 

The logic in `routes_general.camera_img_latest_timelapse()` will search for the filename successfully found by `utils_general.get_camera_image_info()` within `"INSTALL_DIRECTORY/cameras/{camera_uid}"`.

### Versions:

 - Mycodo Version: *8.11.0*
 - Raspberry Pi Version: *4*
 - Raspbian OS Version: *buster*

### Reproducibility

Please list specific setup details that are involved and the steps to reproduce the behavior:

1. Create camera via More->Camera
2. Set the timelapse path for the camera to any valid path on the system
3. Create a "Camera" widget on a dashboard that refers to the new camera
4. For "Image Display Type", select "Display Latest Timestamp Image"
### Expected behavior

The most recent timelapse image would show in the widget

kizniche, 2021-08-18:
Thanks for the bug report. I'll prepare fixes that will be available in the next release.

kizniche, 2021-08-19:
Actually, this was suposedly fixed with https://github.com/kizniche/Mycodo/commit/cca16f60d8ea3efd57e3034edae3bae96e1a10a4#diff-6d1d7d9a1ecfaec5100bf049d2ac42413e199e7d979151a31f649510c1afce94

Neither the current release nor master branch has "INSTALL_DIRECTORY/cameras/{camera_uid}" in utils_general.get_camera_image_info().

This appears to be an issue with camera_img_latest_timelapse() in routes_general.py:

https://github.com/kizniche/Mycodo/blob/803f3eb1c003c4af8e88e7b9246fd03a6966fe11/mycodo/mycodo_flask/routes_general.py#L179-L181

kizniche, 2021-08-19:
It's a bit more involved that that, I'm finding, but I'll have it solved soon enough.

kizniche, 2021-08-19:
Just committed a fix. Thanks again for the bug report.

kimbernator, 2021-08-19:
Thank you for the quick fix!

closed 2021-08-30 by kizniche