kizniche/Mycodo #1073: EZO PMP error 2

cDurham, 2021-08-22:
Describe the problem/bug
I am connected to an Atlas Scientific EZO-PMP. It is the only device listed (only device plugged in) when I run 'i2cdetect -y 1' and am able to run certain commands to the pump. However, whenever I try to run some 'D,x' command, I get an 'error 2' which means syntax. However, my syntax seems correct to me. What 

Versions:
Mycodo Version: 8.11.0
Raspberry Pi Version: 4 Model B
Raspbian OS Version: 10 (buster)

Reproducibility
Install mycodo
plug in ezo pmp on 3.3v line
connect via atlas_scientific_i2c.py
run any command except dispense and verify working
Run some dispense company
get error 2

Screenshots
[image]
[image]
[image]

Additional context
I am able to read data over the other i2c devices when they are plugged in. 

I feel like I might have missed some crucial setup step.

The only thing connected to the pump is a 3.3v line. Do I need to add additional power?

Thanks for your help!

kizniche, 2021-08-22:
Does it dispense when instructed to on the Output page?

cDurham, 2021-08-23:
I get a success message saying the command was sent, but the pump does not dispense. 

I am able to change the I2C address from this page, FWIW. Further, I ensured pylibftdi was installed.

Below are the logs from this command, and a screenshot of my setup. 

`2021-08-22 18:52:03,800 - DEBUG - mycodo.outputs.pump_atlas_ezo_pmp_e44ffbd9 - output_on_off(on, 0, sec, 5.0, 0.0, True)
2021-08-22 18:52:03,812 - DEBUG - mycodo.outputs.pump_atlas_ezo_pmp_e44ffbd9 - Output e44ffbd9-91ff-41e0-a092-cbbd6cc783ed CH0 (32) on for 5.0 seconds. Output returned: None
2021-08-22 18:52:03,814 - DEBUG - mycodo.outputs.pump_atlas_ezo_pmp_e44ffbd9 - EZO-PMP command: D,*
2021-08-22 18:52:08,807 - DEBUG - mycodo.outputs.pump_atlas_ezo_pmp_e44ffbd9 - output_on_off(off, 0, None, 0.0, 0.0, True)
2021-08-22 18:52:08,807 - DEBUG - mycodo.outputs.pump_atlas_ezo_pmp_e44ffbd9 - EZO-PMP command: X
2021-08-22 18:52:08,808 - DEBUG - mycodo.outputs.pump_atlas_ezo_pmp_e44ffbd9 - Output e44ffbd9-91ff-41e0-a092-cbbd6cc783ed CH0 (32) OFF at 2021-08-22 18:52:08. Output returned: None`

[image]

kizniche, 2021-08-23:
Are you powering the pump?

kizniche, 2021-08-23:
Many calibration options no longer work so the calibration page has deprecated (#1022) and will be replaced by calibration on the Input/Output itself that are verified to work, in the next release.

closed 2021-09-02 by cDurham