kizniche/Mycodo #1074: Controllers and Inputs not functioning after reboot

kharberts, 2021-08-24:
STOP right now, and please first look to see if the issue you're about to submit is already an open or recently closed issue at https://github.com/kizniche/Mycodo/issues

Please DO NOT OPEN AN ISSUE:

 - If your Mycodo version is not the latest release version, please update your device before submitting your issue (unless your issue is related to not being able to upgrade). Your problem might already be solved.

 - If your issue has been addressed before. If you have any new information that may aid in solving the issue, post it in the issue that already exists.

If you are going to post a new issue, next read How to Write a Good Bug Report at https://forum.kylegabriel.com/t/how-to-write-a-good-bug-report/71

Please complete as many of the sections below, if applicable, to provide the most information that may help with investigating your issue. Replace the text in brackets with your own text describing the issue. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help those providing assistance by reducing the time required to help you.

### Describe the problem/bug

Inputs and functions show active, but not working after a reboot. Inputs and functions have to be manually disabled/enabled before they work.

### Versions:

 - Mycodo Version: 8.11.0
 - Raspberry Pi Version: 4B
 - Raspbian OS Version: Raspberry Pi OS Lite kernel 5.10.17

### Reproducibility

Please list specific setup details that are involved and the steps to reproduce the behavior:

1. Reboot/halt system and restart
2. Setup --> Input shows all inputs active
3. Setup --> Function shows all previously enabled functions active
4. Live data shows no sensor data
5. Disabling the input or function produces an error indicating that the input or function ID was not found (see screenshot)
6. Enabling the input or  function is successful without error
7. Data now appears in the Live Data screen for the enabled input and the function is now running

### Expected behavior

All previously enabled inputs and functions should start and provide data after reboot/halt/power failure.

### Screenshots

[image]

### Additional context

I want to avoid having to manually deactivate/active inputs and functions after a reboot and avoid trouble in the greenhouse after a power failure.

kizniche, 2021-08-24:
This is not the expected behavior Please provide the Daemon Log (attach a .txt file) from the daemon startup until the errors are produced.

kharberts, 2021-08-24:
Thanks Kyle,

The log is attached and follows the procedure to reproduce the issue outlined in my case notes.

2021-08-24 13:39:46,933 - INFO - mycodo.daemon - Mycodo daemon v8.11.0 starting
2021-08-24 13:39:47,205 - ERROR - mycodo.daemon - Could not query output state: 'NoneType' object has no attribute 'output_state'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 849, in output_state
    return self.controller['Output'].output_state(output_id, output_channel)
AttributeError: 'NoneType' object has no attribute 'output_state'
2021-08-24 13:39:47,263 - ERROR - mycodo.daemon - Could not query output state: '37352c46-6a4f-4d08-bf6a-278e248c0ede'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 849, in output_state
    return self.controller['Output'].output_state(output_id, output_channel)
  File "/var/mycodo-root/mycodo/controllers/controller_output.py", line 320, in output_state
    return self.output[output_id].output_state(output_channel)
KeyError: '37352c46-6a4f-4d08-bf6a-278e248c0ede'
2021-08-24 13:39:47,271 - ERROR - mycodo.daemon - Could not query output state: '395a4215-9006-4340-846b-ce85ecbdc098'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 849, in output_state
    return self.controller['Output'].output_state(output_id, output_channel)
  File "/var/mycodo-root/mycodo/controllers/controller_output.py", line 320, in output_state
    return self.output[output_id].output_state(output_channel)
KeyError: '395a4215-9006-4340-846b-ce85ecbdc098'
2021-08-24 13:39:47,325 - ERROR - mycodo.daemon - Could not query output state: '5476ee8f-ff92-4459-89eb-2ec932368f15'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 849, in output_state
    return self.controller['Output'].output_state(output_id, output_channel)
  File "/var/mycodo-root/mycodo/controllers/controller_output.py", line 320, in output_state
    return self.output[output_id].output_state(output_channel)
KeyError: '5476ee8f-ff92-4459-89eb-2ec932368f15'
2021-08-24 13:39:47,330 - ERROR - mycodo.daemon - Could not query output state: '9e784eed-0f27-4aaa-8b1f-693276a74fcc'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 849, in output_state
    return self.controller['Output'].output_state(output_id, output_channel)
  File "/var/mycodo-root/mycodo/controllers/controller_output.py", line 320, in output_state
    return self.output[output_id].output_state(output_channel)
KeyError: '9e784eed-0f27-4aaa-8b1f-693276a74fcc'
2021-08-24 13:39:47,381 - ERROR - mycodo.daemon - Could not query output state: '19eb4247-f21d-4b77-8c79-8c91dbe55f1d'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 849, in output_state
    return self.controller['Output'].output_state(output_id, output_channel)
  File "/var/mycodo-root/mycodo/controllers/controller_output.py", line 320, in output_state
    return self.output[output_id].output_state(output_channel)
KeyError: '19eb4247-f21d-4b77-8c79-8c91dbe55f1d'
2021-08-24 13:39:47,395 - ERROR - mycodo.daemon - Could not query output state: '19eb4247-f21d-4b77-8c79-8c91dbe55f1d'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 849, in output_state
    return self.controller['Output'].output_state(output_id, output_channel)
  File "/var/mycodo-root/mycodo/controllers/controller_output.py", line 320, in output_state
    return self.output[output_id].output_state(output_channel)
KeyError: '19eb4247-f21d-4b77-8c79-8c91dbe55f1d'
2021-08-24 13:39:47,439 - ERROR - mycodo.daemon - Could not query output state: 'ceacf4e6-dae4-4449-92d2-16781382cb51'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 849, in output_state
    return self.controller['Output'].output_state(output_id, output_channel)
  File "/var/mycodo-root/mycodo/controllers/controller_output.py", line 320, in output_state
    return self.output[output_id].output_state(output_channel)
KeyError: 'ceacf4e6-dae4-4449-92d2-16781382cb51'
2021-08-24 13:39:47,445 - ERROR - mycodo.daemon - Could not query output state: '09f69997-d7bb-41c7-9660-828e808c2d26'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 849, in output_state
    return self.controller['Output'].output_state(output_id, output_channel)
  File "/var/mycodo-root/mycodo/controllers/controller_output.py", line 320, in output_state
    return self.output[output_id].output_state(output_channel)
KeyError: '09f69997-d7bb-41c7-9660-828e808c2d26'
2021-08-24 13:39:47,496 - ERROR - mycodo.daemon - Could not query output state: '323f2571-9635-48c2-aa8f-85ebd9ded846'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 849, in output_state
    return self.controller['Output'].output_state(output_id, output_channel)
[... 165 lines cut ...]
2021-08-24 13:42:03,422 - ERROR - mycodo.inputs.atlas_ec_70b68111 - Calibration measurement not found within the past 120 seconds
2021-08-24 13:42:06,968 - INFO - mycodo.controllers.controller_input_0f0849aa - Activated in 181.2 ms
2021-08-24 13:42:08,730 - DEBUG - mycodo.controllers.controller_input_0f0849aa - Adding measurements to InfluxDB with ID 0f0849aa-ff4d-4122-ab4a-79a376415e51: {0: {'measurement': None, 'unit': 'F', 'value': 71.7125, 'timestamp_utc': None}}
[... cut ...]
2021-08-24 13:42:54,518 - ERROR - mycodo.controllers.controller_input_44f0baed - Mycodo is attempting to acquire measurement(s) from an Input that has already critically errored. Review the log lines following Input Activation to investigate why this happened.
2021-08-24 13:43:08,717 - DEBUG - mycodo.controllers.controller_input_0f0849aa - Adding measurements to InfluxDB with ID 0f0849aa-ff4d-4122-ab4a-79a376415e51: {0: {'measurement': None, 'unit': 'F', 'value': 71.7125, 'timestamp_utc': None}}

kizniche, 2021-08-24:
I meant to attack it via the website comment box. I want to avoid very long posts due to pasting, and it appears your email reply to the conversation with the attachment pasted the entire contents of the attached log.

kizniche, 2021-08-24:
From your log, it appears you are attempting to deactivate the Input and PID controllers before they are even activated by the daemon at startup. From your log:

```
2021-08-24 13:41:03,078 - ERROR - mycodo.daemon - PID controller with ID 8ae846be-5fc9-4d1e-ac5d-780327a5e414 not found
2021-08-24 13:41:03,350 - ERROR - mycodo.daemon - Input controller with ID 0aaf8390-cf63-4f4f-880b-def00fb415b4 not found
...
2021-08-24 13:42:10,289 - INFO - mycodo.controllers.controller_input_0aaf8390 - Activated in 388.9 ms
2021-08-24 13:42:37,184 - INFO - mycodo.controllers.controller_pid_8ae846be - Activated in 121.2 ms
```

kharberts, 2021-08-24:
Sorry about the long post. Apparently it took the contents on my attachment and dumped them right into the main body of the reply.

I did manually disable/enable them at startup for this test. I have tried waiting, but the inputs never produce any data without deactivating/activating them. I can provide logs from a regular reboot without touching anything for comparison if you like.

kizniche, 2021-08-24:
Again, the issue is you're attempting to deactivate a controller that has not been activated yet because the daemon is still starting up and hasn't completed. I don't see in your log that the daemon has completed starting up (example, below).

```
2021-08-09 21:21:38,991 - INFO - mycodo.daemon - Mycodo daemon v8.11.0 starting
2021-08-09 21:21:40,377 - INFO - mycodo.outputs.mqtt_a86e7f09 - Initialized in 366.4 ms
2021-08-09 21:21:40,597 - INFO - mycodo.outputs.mqtt_05c92581 - Initialized in 217.3 ms
2021-08-09 21:21:40,816 - INFO - mycodo.outputs.on_off_gpio_3aae6bf2 - Output setup on pin 20 and turned OFF (ON=HIGH)
2021-08-09 21:21:40,817 - INFO - mycodo.outputs.on_off_gpio_3aae6bf2 - Initialized in 217.4 ms
2021-08-09 21:21:41,069 - INFO - mycodo.outputs.on_off_gpio_d5e34308 - Output setup on pin 21 and turned OFF (ON=HIGH)
2021-08-09 21:21:41,070 - INFO - mycodo.outputs.on_off_gpio_d5e34308 - Initialized in 250.8 ms
2021-08-09 21:21:41,070 - INFO - mycodo.controllers.controller_output - Activated in 1670.9 ms
2021-08-09 21:21:41,967 - INFO - mycodo.controllers.controller_conditional_1e492347 - Activated in 158.1 ms
2021-08-09 21:21:42,289 - INFO - mycodo.controllers.controller_conditional_94efb8d3 - Activated in 106.7 ms
2021-08-09 21:21:42,290 - INFO - mycodo.daemon - All activated Conditional controllers started
2021-08-09 21:21:42,653 - INFO - mycodo.controllers.controller_trigger_97acf4b7 - Activated in 134.3 ms
2021-08-09 21:21:43,027 - INFO - mycodo.controllers.controller_trigger_b4f78c6c - Activated in 139.2 ms
2021-08-09 21:21:43,028 - INFO - mycodo.daemon - All activated Trigger controllers started
2021-08-09 21:21:43,716 - INFO - mycodo.controllers.controller_input_3ca3da6e - Activated in 458.0 ms
2021-08-09 21:21:44,369 - INFO - mycodo.controllers.controller_input_65905428 - Activated in 327.5 ms
2021-08-09 21:21:46,123 - INFO - mycodo.controllers.controller_input_436953bf - Activated in 753.5 ms
2021-08-09 21:21:46,843 - INFO - mycodo.controllers.controller_input_de3eb65c - Activated in 352.5 ms
2021-08-09 21:21:47,543 - INFO - mycodo.controllers.controller_input_da5ec399 - Activated in 421.6 ms
2021-08-09 21:21:48,158 - INFO - mycodo.controllers.controller_input_ce967ed4 - Activated in 374.0 ms
2021-08-09 21:21:48,803 - INFO - mycodo.controllers.controller_input_7d9be460 - Activated in 386.3 ms
2021-08-09 21:21:49,423 - INFO - mycodo.controllers.controller_input_1a5a3be1 - Activated in 372.2 ms
2021-08-09 21:21:50,148 - INFO - mycodo.controllers.controller_input_9d40dd88 - Activated in 449.0 ms
2021-08-09 21:21:51,163 - INFO - mycodo.controllers.controller_input_68b6ceda - Activated in 745.0 ms
2021-08-09 21:21:51,167 - INFO - mycodo.daemon - All activated Input controllers started
2021-08-09 21:21:51,168 - INFO - mycodo.daemon - All activated Math controllers started
2021-08-09 21:21:51,174 - INFO - mycodo.daemon - All activated PID controllers started
2021-08-09 21:21:51,179 - INFO - mycodo.daemon - All activated LCD controllers started
2021-08-09 21:21:51,183 - INFO - mycodo.daemon - All activated Function controllers started
2021-08-09 21:21:52,595 - INFO - mycodo.controllers.controller_widget - Activated in 1410.8 ms
2021-08-09 21:21:53,597 - INFO - mycodo.daemon - Mycodo daemon started in 14.604 seconds
```

kharberts, 2021-08-24:
Understood. I see that in my logs now. The daemon hasn't started even after 45 min of waiting.

kizniche, 2021-08-24:
Okay, so we see the effects of the daemon not starting fully, now we need to figure out why. Deactivate everything that can be deactivated (Inputs/Functions/LCDs/Math controllers), then reboot the Pi. Determine if the daemon can start fully. If it can, then selectively Activate each component, then restart the Backend/Daemon from the Config dropdown and watch the log to see at what point is it no longer able to start.

kharberts, 2021-08-25:
Okay, this log output is with all inputs and functions disabled. The daemon took 3604.965 seconds to start with everything disabled.
[Daemon log.txt](https://github.com/kizniche/Mycodo/files/7042821/Daemon.log.txt)

kizniche, 2021-08-25:
Whatever code you're using for this Output is your issue:

```
mycodo.outputs.on_off_python_ceacf4e6 - Initialized in 3600294.1 ms
```

kharberts, 2021-08-25:
Bingo! I had a sleep 3600 line in there to suspend pumps and functions during a solution purge. I've swapped all of that out for triggers. Daemon now starts in 15 seconds and everything works after a reboot. Thanks a ton for a lesson in what to look for.

kizniche, 2021-08-25:
Great! Glad to hear.

closed 2021-08-25 by kizniche