kizniche/Mycodo #1078: HighCharts Widgets are not displayed,

pe2269, 2021-08-30:
STOP right now, and please first look to see if the issue you're about to submit is already an open or recently closed issue at https://github.com/kizniche/Mycodo/issues

Please DO NOT OPEN AN ISSUE:

 - If your Mycodo version is not the latest release version, please update your device before submitting your issue (unless your issue is related to not being able to upgrade). Your problem might already be solved.

 - If your issue has been addressed before. If you have any new information that may aid in solving the issue, post it in the issue that already exists.

If you are going to post a new issue, next read How to Write a Good Bug Report at https://forum.kylegabriel.com/t/how-to-write-a-good-bug-report/71

Please complete as many of the sections below, if applicable, to provide the most information that may help with investigating your issue. Replace the text in brackets with your own text describing the issue. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help those providing assistance by reducing the time required to help you.

### Describe the problem/bug
Hi Kyle,
I have  problems with the new highcharts widgets from the new version (8.12) of Mycodo.
The input show good measurements (CPU/GPU temperature), i can displayed it with the measurement widget for example. 
But if i use the highcharts dependancies widget (Gauges and Graph), after the installtion of the Highcharts dependencies the widgets are not dsiplayed (see picture).

Thanks for your advices

PE

### Versions:

 - Mycodo Version: 8.12.0
 - Raspberry Pi 4 model B version 1.4
-  Python Version: 3.7.3 (default, Jan 22 2021, 20:04:44) [GCC 8.3.0]
 - Raspbian OS Version 10 Buster: *[e.g. Buster Lite]*

### Reproducibility

Please list specific setup details that are involved and the steps to reproduce the behavior:

1. add Dashboard widget
2. Gauge (Angular) Highcharts or Gauge Highcharts (Solid) or or Graph (synchronous highstock)
3. Select the input (raspberry CPU/GPU temperature)
4. The widget displayed nothing

### Expected behavior

### Screenshots

[image]

### Additional context

*[Anything that should be added to make it easier to address this issue - replace this with your own text]*

kizniche, 2021-08-30:
Can you inspect the page in your browser and see if any 404 errors appear in the console?

kizniche, 2021-08-30:
And did you indeed add a new widget after the upgrade or just save a previously-created widget?

pe2269, 2021-08-30:
> Can you inspect the page in your browser and see if any 404 errors appear in the console?

Yes i have some 404 errors i join you the log 
[192.168.0.68-1630339719296.log](https://github.com/kizniche/Mycodo/files/7078001/192.168.0.68-1630339719296.log)

pe2269, 2021-08-30:
> And did you indeed add a new widget after the upgrade or just save a previously-created widget?

I try both ... (and on 2 differents raspberry)

pe2269, 2021-08-30:
> > Can you inspect the page in your browser and see if any 404 errors appear in the console?
> 
> Yes i have some 404 errors i join you the log
> [192.168.0.68-1630339719296.log](https://github.com/kizniche/Mycodo/files/7078001/192.168.0.68-1630339719296.log)

[192.168.0.68-1630339719296.log](https://github.com/kizniche/Mycodo/files/7078015/192.168.0.68-1630339719296.log)
[192.168.0.68-1630339935272.log](https://github.com/kizniche/Mycodo/files/7078016/192.168.0.68-1630339935272.log)
[192.168.0.68-1630339956631.log](https://github.com/kizniche/Mycodo/files/7078017/192.168.0.68-1630339956631.log)
[192.168.0.68-1630339961583.log](https://github.com/kizniche/Mycodo/files/7078018/192.168.0.68-1630339961583.log)
[192.168.0.68-1630339968439.log](https://github.com/kizniche/Mycodo/files/7078019/192.168.0.68-1630339968439.log)
[192.168.0.68-1630339980079.log](https://github.com/kizniche/Mycodo/files/7078020/192.168.0.68-1630339980079.log)

pe2269, 2021-08-30:
If i add on the same dashboard the Greph widget, the Jauge (angular and solid) are displayed but the graph no ...
[image]

kizniche, 2021-08-30:
Create a new dashboard with only one graph

pe2269, 2021-08-30:
> Create a new dashboard with only one graph

The graph Widget is displayed !!
[image]

kizniche, 2021-08-30:
Can you investigate widget combinations to find what specifically causes the issue?

pe2269, 2021-08-30:
> Can you investigate widget combinations to find what specifically causes the issue?

1) graph then angular then solid = ok for all
[image]

2) Solid then graph ten angular 
[image]

3) Angular then grap then solid
[image]

kizniche, 2021-08-30:
I just release v8.12.1 that fixes this issue

closed 2021-08-30 by kizniche

pe2269, 2021-08-31:
Thanks a lot !!