kizniche/Mycodo #1080: Widget back colour and size pb

pe2269, 2021-08-31:
### Describe the problem/bug

Hi Kyle,

I upgrade my system to the V8.12.2 in order to fix the displayed widgets problems (it's works :))
But now i have some new issues :
 
                 1°) The back colour of the Solid and Angular Gauge widget remains white in the darkly theme i use.
                 2°) When I add in the same dashboard that the gauges widget, a graph widget, the back colour problem is fixed
                 3°) i must to save the widget after change in the size of the widget (maybe it's not a bug)                 

### Versions:

 - Mycodo Version: *v8.12.2
 - Raspberry Pi Version: 4 B 
 - Raspbian OS Version: 10 Buster

### Reproducibility

Please list specific setup details that are involved and the steps to reproduce the behavior:

1. *[Go to...]*
2. *[Click on...]*
3. *[Scroll down to...]*
4. *[See error...]*
5. *[replace this list with your own text]*

### Expected behavior

*[A clear and concise description of what you expected to happen - replace this with your own text]*

### Screenshots

[image]

### Additional context

*[Anything that should be added to make it easier to address this issue - replace this with your own text]*

pe2269, 2021-08-31:
Great it's working !!

kizniche, 2021-08-31:
You can leave this open until the fix is released, otherwise it's likely someone else will open the same issue.

closed 2021-08-31 by kizniche