kizniche/Mycodo #1081: Dashboard layout broken after update to 8.12.4

constantm, 2021-09-01:
### Describe the problem/bug

After updating to the latest version of MyCodo, my dashboard layouts seem to be broken in Firefox. On Webkit  histograms correctly take up the full width of the screen, but the layout of smaller graphs are also broken.

### Versions:

 - Mycodo Version: 8.12.4
 - Raspberry Pi Version: 4
 - Raspbian OS Version: Raspbian GNU/Linux 10 (buster)

### Screenshots

Firefox:
[image]

Webkit:
[image]

kizniche, 2021-09-01:
I exclusively use Firefox for developement/testing and don't experience this issue on any of my systems running 8.12.4.

Are you sure this isn't some sort of caching issue?

Can you provide any additional info like errors in the inspection console?

Ikzy, 2021-09-01:
I'm having this same problem.

on Safari; however I'm getting this error in inspector:

Error: Highcharts error #16: www.highcharts.com/errors/16/

and I remember the update throwing errors when trying to update Highchart / highstock dependency..

checking the dependency page gives me a bunch of errors and an internal server error..

pe2269, 2021-09-01:
I have the same issues with google chrome and in my 7'' touchscreen with chromium
[image]
[image]

Ikzy, 2021-09-01:
There is still some errors popping up here, but I managed to downgrade / reinstall and was able to re-make my dashboard again.. so I don't know if its fixed. The dependencies page is still erroneous here ( I opened a new issue for that) 
[image]

kizniche, 2021-09-01:
That's fine, there are no maps and it should be blocking external content

pe2269, 2021-09-01:
My dashboard seems to be fixed with the V8.12.5 with chrome and chromium !!
Thans again !!

constantm, 2021-09-02:
The recent update fixed the width issue on the synchronous graph. However, the gauge widgets rendering remained broken. 

Good news though - I could create a new gauge widget that rendered correctly. 

The only difference I could see between the broken widgets the the new, correctly rendering widget was the height property. The newly created widget had a height of 183px, whereas the broken widgets had a height of 93px, which I believe causes Highcharts to render the squished gauges.

So I reckon this bug has something to do with how the heights of widgets are stored.

[image]

kizniche, 2021-09-02:
Yes, the gauge will not display the value if it's too small. Just drag the widget lerger.

kizniche, 2021-09-02:
And I believe your grid width issue was a caching issue. The update increased the grid width value and I recall this exact issue occurred when the previous version grid file was cached.

constantm, 2021-09-03:
I had no idea you could drag widgets to make them larger! 🥲 Thank you for pointing this out, and thank you for your awesome work on Mycodo.

closed 2021-09-03 by constantm

kizniche, 2021-09-03:
>I had no idea you could drag widgets to make them larger! 🥲

I remember when I first got gridstack working, which replaced the old clunky system of setting width/height values, and I got to drag to resize a widget. It's a great feeling!

pe2269, 2021-09-16:
Hi,
From time to time my dashboard layout breaks again.

I use the v8.12.6 with jauges on dashboard.
I have a 7'' touchscreen which display continually  with chromium the dashboard and a second access on google chrome too.
The two display show the same broken layout but just the "first" dashboard with jauges not the others.
I resize the dashboard and few days after the layout brokes again ....

I dont't know how to documented this bug better :/

stardawg, 2022-01-12:
Is this issue being worked on?  It's been going on for several versions now.  I use chrome and firefox browsers.  It appears to be an issue with chrome.  Once I make any changes to the default dashboard in chrome it jacks the whole thing up rendering it virtually useless.  I've been relying on the live data view and a second dashboard with asynchronous graphs as a workaround which is a major pain especially since the graphs render terribly on mobile displays so I actually have to use a laptop to ensure my system is operating within parameters.

kizniche, 2022-01-12:
See the changelog

stardawg, 2022-01-12:
I'm sorry...is this the upcoming fix "Fix Desktop Widgets resizing when viewing on mobile browsers"?

That would be awesome.  BTW...@kizniche I really appreciate this product.  I can't begin to imagine the amount of time and effort you put into this.  I wish I had more time to contribute as well.  Perhaps one day.  Again...thank you so much!!!

kizniche, 2022-01-12:
Yes, the yet-to-be-released version. I'll try to get it pushed out soon. In the meantime, you can [upgrade to master](https://github.com/kizniche/Mycodo/wiki/Updating-to-Github-Master-Branch) if you want to test the fix.