kizniche/Mycodo #1082: 8.12.5 Dependencies page internal server error

Ikzy, 2021-09-01:
STOP right now, and please first look to see if the issue you're about to submit is already an open or recently closed issue at https://github.com/kizniche/Mycodo/issues

Please DO NOT OPEN AN ISSUE:

 - If your Mycodo version is not the latest release version, please update your device before submitting your issue (unless your issue is related to not being able to upgrade). Your problem might already be solved.

 - If your issue has been addressed before. If you have any new information that may aid in solving the issue, post it in the issue that already exists.

If you are going to post a new issue, next read How to Write a Good Bug Report at https://forum.kylegabriel.com/t/how-to-write-a-good-bug-report/71

Please complete as many of the sections below, if applicable, to provide the most information that may help with investigating your issue. Replace the text in brackets with your own text describing the issue. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help those providing assistance by reducing the time required to help you.

### Describe the problem/bug
When opening options and the dependencies page (mycodo/admin/dependencies/) page takes 20 seconds to load and throws an internal server error (500) *see other context

### Versions:

 - Mycodo Version: *[8.12.5]*
 - Raspberry Pi Version: *[4]*
 - Raspbian OS Version: *[ Buster]*

### Reproducibility

Please list specific setup details that are involved and the steps to reproduce the behavior:

1. *[Go to options]*
2. *[Click on dependencies]*
3.  PROFIT

### Additional context
Full Traceback:

**
Error 500: Internal Server Error

Something bad happened but it's probably not your fault. Letting the developers know about these issues is crucial to supporting Mycodo. Please submit a new issue on GitHub with the following diagnostic information and error traceback (copy the entire traceback):

Version: 8.12.5 
Database: 6e394f2e8fec 
Model: Raspberry Pi 4 Model B Rev 1.2

Release:
Distributor ID: Raspbian
Description: Raspbian GNU/Linux 10 (buster)
Release: 10
Codename: buster

Firmware:
Apr 30 2021 13:46:42 
Copyright (c) 2012 Broadcom
version d7f29d96450abfc77cd6cf011af1faf1e03e5e56 (clean) (release) (start_x)

Error (Full Traceback):

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1952, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_restx/api.py", line 652, in error_router
    return original_handler(e)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1821, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/_compat.py", line 39, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_login/utils.py", line 272, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_admin.py", line 379, in admin_dependencies
    if each_dep not in unmet_list:
TypeError: unhashable type: 'list'
**

kizniche, 2021-09-03:
This issue has been mentioned on **DIY Automation Forum**. There might be relevant details there:

https://forum.kylegabriel.com/t/reinstalling-an-unmet-dependency/351/1

closed 2021-09-04 by kizniche

zachwoodill, 2022-05-11:
Hi @kizniche I just wanted to mention I am having the same issue.

kizniche, 2022-05-12:
@zachwoodill This has been fixed and will be in the next release.