kizniche/Mycodo #1083: Unable to access Inputs UI after upgrade to 8.12.5

emuehlstein, 2021-09-01:
### Describe the problem/bug

After upgrade from 8.9.2 to 8.12.5 when Setup->Input is selected from the menu mycodo returns a 500 error.
### Versions:

 - Mycodo Version: 8.12.5
 - Raspberry Pi Version: 4
 - Raspbian OS Version: Raspbian Buster

### Reproducibility

Please list specific setup details that are involved and the steps to reproduce the behavior:

1. Upgrade from 8.9.2 to 8.12.5
2. Click on Setup -> Input
3. See error.

### Expected behavior

Input screen displayed.

### Screenshots

### Additional context

Version: 8.12.5
Database: 6e394f2e8fec
Model: Raspberry Pi 4 Model B Rev 1.2

Release:
Distributor ID: Raspbian
Description: Raspbian GNU/Linux 10 (buster)
Release: 10
Codename: buster

Firmware:
Sep 2 2020 21:13:13
Copyright (c) 2012 Broadcom
version 4439d2aaa6c376a2d1ef4402f142e1cf4de37c43 (clean) (release) (start)

Error (Full Traceback):

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1952, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_restx/api.py", line 652, in error_router
    return original_handler(e)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1821, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/_compat.py", line 39, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_login/utils.py", line 272, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_input.py", line 406, in page_input
    devices_1wire_w1thermsensor=devices_1wire_w1thermsensor)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/templating.py", line 140, in render_template
    ctx.app,
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/templating.py", line 120, in _render
    rv = template.render(context)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/jinja2/environment.py", line 1090, in render
    self.environment.handle_exception()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/jinja2/environment.py", line 832, in handle_exception
    reraise(*rewrite_traceback_stack(source=source))
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/jinja2/_compat.py", line 28, in reraise
    raise value.with_traceback(tb)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/pages/input.html", line 3, in top-level template code
    {% set help_page = ["https://kizniche.github.io/Mycodo/Inputs/", dict_translation['input']['title']] %}
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/layout.html", line 384, in top-level template code
    {%- block body %}{% endblock -%}
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/pages/input.html", line 77, in block "body"
    {% include 'pages/data_options/math.html' %}
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/pages/data_options/math.html", line 161, in top-level template code
    {{form_base.csrf_token}}
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/jinja2/environment.py", line 471, in getattr
    return getattr(obj, attribute)
jinja2.exceptions.UndefinedError: 'form_base' is undefined

[mycodoupgrade.log](https://github.com/kizniche/Mycodo/files/7093609/mycodoupgrade.log)

kizniche, 2021-09-01:
Thanks for the bug report. This should not be affecting the operation, but only the frontend to change settings. I can release a bugfix in a few hours. In the meantime, if you need to change settings on your Input page, I suggest you restore a backup.

emuehlstein, 2021-09-02:
Thanks Kyle!  I upgraded to master (610f40d) and tested this morning.  The inputs page loaded successfully and I was able to delete the old math inputs.

closed 2021-09-04 by kizniche