kizniche/Mycodo #1086: Camera DB model should store filename of its most recent still/timelapse images

kimbernator, 2021-09-02:
**Is your feature request related to a problem? Please describe.**
When a camera has taken a large number of timelapse images, querying for the most recent one, whether it's in the camera administration page or on a dashbaord, can take an extremely long time. This appears to be because it's running the following:

```PY
 latest_time_lapse_img_full_path = max(glob.iglob(
                '{path}/Timelapse-{cam_id}-*.jpg'.format(
                    path=tl_path,
                    cam_id=each_camera.id)), key=os.path.getmtime)
```

This is pretty rough since it's checking the mtime on each and every file individually, moreso when the timelapse image path is on something like an SMB share which is quite slow compared to local storage. With thousands or more images this can be a pretty frustrating thing to wait for since a timelapse dashboard widget or the camera administration page both require this function to return before they will even load. Often the whole system will lock up since the function is stuck running in the background.

**Describe the solution you'd like**
I think there are a few options that could feasibly improve this.

Add a field to the Camera model in the DB which is the filename (with or without the path) of the most recent timelapse. Update it whenever a new one is taken. When querying for a camera's most recent timelapse, all you'd need to do it get that field and maybe verify that the file exists.

**Describe alternatives you've considered**

1. Simply sort the filenames in the given path and take the last one. Since filenames are identical aside from their numbers, an alphanumeric sort will give you the last one. Only downside is if someone runs over 100,000 images and the filenames stop conforming to the 5 digit numeric minimum (e.g. img_99999.jpg would come before img_100000.jpg)

2. Somehow make the call to that method asynchronous so it doesn't bring the whole program to a halt. Probably not worth the work.

**Additional context**

A really simple benchmark I did on a dir which is within a mounted cifs/smb share. There are ~8000 images in the directory:

```PY
>>> def run_iglob():
...   start_time = time.time()
...   print(f"starting at {start_time}")
...   files = max(glob.iglob("/state/mycodo/timelapse_images/*.jpg"), key=os.path.getmtime)
...   end_time = time.time()
...   print(f"ended at {end_time}   (total: {end_time - start_time})")
...
>>> run_iglob()
starting at 1630622099.835114
ended at 1630622231.550066   (total: 131.7149519920349)
```

I'm willing to make this enhancement myself if I get some time in the next couple days, I just want to get feedback and make sure the current design isn't secretly integral to the system.

kizniche, 2021-09-02:
The first issue I see with the alternative approach is because filenames include the name of the camera controller, if the user changes the name, the most recently-acquired image is not necessarily the one that will appear if you merely sort by filename.

kizniche, 2021-09-02:
Considering the filename can change, there's only two solutions I can see to this:

1. Change the filenaming behavior, which since current users will have timelapse images already present, does not guarantee the solution will work for all without deleting or moving their current files, which I do not want to do.
2. Save the filename of the last timelapse image to the database, then have all functions that want the latest timelapse just reference this DB entry.

closed 2021-09-24 by kizniche