kizniche/Mycodo #1089: LCD-Control with a trigger

grux77, 2021-09-04:
RPi4
Linux Mycodo 5.10.17-v7l+
Mycodo Version: 8.12.6
Python Version: 3.7.3 (default, Jan 22 2021, 20:04:44) [GCC 8.3.0]

Hallo Kyle!
I would like to save electricity. I want to turn the backlighting or flashing of my display on and off with a trigger.
If I add Action "LCD:Backlight:On" or other LCD-actions in Setup/Function/Trigger(xxx) I only see devices that are in the old  
Setup/LCD area but not in the new Setup/Function/LCD. You can only control the display with "Controller:Activate/Deactivate". But it always stays light.

And I have a suggestion for clarity:
I would suggest moving the LCD-Setup to the "Setup/Output" area.. As a Mycodo beginner you would rather search there than  in the "Setup/Functions". I find that a display is more of an output than a function. Just like a camera is an input. Is just an idea of ​​mine.

kizniche, 2021-09-05:
Good suggestion. I just committed a [feature addition](https://github.com/kizniche/Mycodo/blob/4e873c9da0e69e5739e4455be47a1dce9a0f62b2/CHANGELOG.md#8127-unreleased) to add the ability to have Actions affect Functions, and the first to be added are the Backlight On/Off Actions. If you are able to [upgrade to master](https://github.com/kizniche/Mycodo/wiki/Updating-to-Github-Master-Branch), please test and let me know if this works.

kizniche, 2021-09-06:
Just committed some fixes and it should now work

grux77, 2021-09-07:
Yes, it works. Tested with Display:Generic LCD 16x2 (I2C) and Trigger:Edge. 
So I can turn the backlight on and off with two buttons. I think the trigger now reacts faster than before to the press of a button.
Thanks very much!

closed 2022-03-28 by kizniche