kizniche/Mycodo #1091: Dashboard not displaying after 8.12.6 upgrade

jacopienaar, 2021-09-17:
### Describe the problem/bug

After upgrading to the latest version v8.12.6 the Dashboard is no longer dispalying correctly - just empty boxes.  Issue on Chrome (desktop) and Safari (phone). 

I've tried adding and editing widgets to no avail. Seems to be an issue with pulling the data?

### Versions:

 - Mycodo Version: 8.12.6
 - Raspberry Pi Version: 3B+
 - Raspbian OS Version: Buster
 
### Screenshots

[image]

Thanks for the awesome work that you are doing with Mycodo. I've been using it for the past 5 years for growing mushrooms. I see you recently started a new project - really cool!

kizniche, 2021-09-17:
The only thing that could cause something that looks like this is a Javascript issue. Please copy any errors from the inspection console of your browser.

jacopienaar, 2021-09-17:
[image]

kizniche, 2021-09-17:
None of that is abnormal or could cause the issue. Unless you can replicate the issue, there's not much I can do. You have the ability to create dashboards if you want to experiment.

jacopienaar, 2021-09-17:
Not sure how to replicate the issue. I upgraded from 8.10.1 where the dashboards worked fine. I've created multiple dashboards but no joy, the problem persists.
I'll get stuck into the details a bit later today to see if I can get to the bottom of it.

kizniche, 2021-09-17:
You showed a complex dashboard with several different widgets. You're saying the same ting happens with a new dashboard with only one widget? And you tested this with every widget?

jacopienaar, 2021-09-17:
Yes, with every dashboard *new ones/existing ones) and with all widgets where I have data to feed into it - the issue persists.

kizniche, 2021-09-17:
You can post the log excerpts you are indicating have errors, this may aid in diagnosing the issue.

jacopienaar, 2021-09-18:
Eh, so I managed to stuff this up even further. I opted to roll back to 8.10.1 and then to run the upgrade again. After rolling back, Mycodo now says there is no internet connection which there is - I quickly googled something on chrome. See below screen shot. At least the widgets are displaying again...
I have a static IP assigned (192.168.1.70) - could this be something to do with Mycodo trying to connect to 192.168.1.65 ?

**Web Error log:**
2021/06/22 18:43:12 [crit] 534#534: *1 connect() to unix:/var/run/mycodoflask.sock failed (2: No such file or directory) while connecting to upstream, client: 192.168.1.65, server: , request: "GET /dashboard/7424f3fa-dd71-40cd-9b1d-f8af9afbaf96 HTTP/1.1", upstream: "http://unix:/var/run/mycodoflask.sock:/dashboard/7424f3fa-dd71-40cd-9b1d-f8af9afbaf96", host: "192.168.1.70", referrer: "https://192.168.1.70/settings/general"
2021/09/18 08:44:45 [alert] 530#530: *430 open socket #17 left in connection 4
2021/09/18 08:44:45 [alert] 530#530: *426 open socket #14 left in connection 6
2021/09/18 08:44:45 [alert] 530#530: *431 open socket #15 left in connection 9
2021/09/18 08:44:45 [alert] 530#530: aborting
2021/09/18 11:56:11 [notice] 9098#9098: using inherited sockets from "6;7;"

[image]

jacopienaar, 2021-09-18:
Ok I feel like a bit of a tool. I managed to update it via terminal and updated without issues - dashboard displaying correctly and internet working as normal.

I set and forget my control system and then months down the line I've forgotten everything and have to brush up on knowledge again. Not ideal. Will get into the weeds again over the next few weeks/months as there are quite a few things I want to get sorted in terms of set up and possibly new features with regard to mushroom growing so you'll see my name pop up regurarly.

Thanks again for your prompt response and support and for creating this amazing little piece of software!

closed 2021-09-18 by jacopienaar