kizniche/Mycodo #1092: Internal server error after upgrading from version 8.11.0 to 8.12.6

sjoerdschouten, 2021-09-22:
### Describe the problem/bug
Mycodo produces an internal server error on version `8.12.6` (after succesfully upgrading from `8.11.0`) with the following log:

```
Fout 500 - Interne Server Fout

Something bad happened but it's probably not your fault. Letting the developers know about these issues is crucial to supporting Mycodo. Please submit a new issue on GitHub with the following diagnostic information and error traceback (copy the entire traceback):

Version: 8.12.6
Database: 6e394f2e8fec
Model: Raspberry Pi 4 Model B Rev 1.4

Release:
Distributor ID: Raspbian
Description: Raspbian GNU/Linux 10 (buster)
Release: 10
Codename: buster

Firmware:
Aug 3 2021 18:14:56
Copyright (c) 2012 Broadcom
version 40787ee5905644f639a2a0f6e00ae12e517a2211 (clean) (release) (start)

Error (Full Traceback):

Traceback (most recent call last):
  File "/home/sjoerd/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/sjoerd/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1952, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/sjoerd/Mycodo/env/lib/python3.7/site-packages/flask_restx/api.py", line 652, in error_router
    return original_handler(e)
  File "/home/sjoerd/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1821, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/sjoerd/Mycodo/env/lib/python3.7/site-packages/flask/_compat.py", line 39, in reraise
    raise value
  File "/home/sjoerd/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/sjoerd/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/sjoerd/Mycodo/mycodo/mycodo_flask/routes_general.py", line 86, in index_page
    if not flask_login.current_user.index_page:
  File "/home/sjoerd/Mycodo/env/lib/python3.7/site-packages/werkzeug/local.py", line 347, in __getattr__
    return getattr(self._get_current_object(), name)
AttributeError: 'AnonymousUserMixin' object has no attribute 'index_page'

```
### Versions:

- Database: 6e394f2e8fec
- Model: Raspberry Pi 4 Model B Rev 1.4

Release:
- Distributor ID: Raspbian
- Description: Raspbian GNU/Linux 10 (buster)
- Release: 10
- Codename: buster

Firmware:
- Aug 3 2021 18:14:56
- Copyright (c) 2012 Broadcom
- version 40787ee5905644f639a2a0f6e00ae12e517a2211 (clean) (release) (start)

### Reproducibility

This error occurred after upgrading Mycodo from version `8.11.0` to version ` 8.12.6`. The updating process itself went fine and did not produce any errors.

### Expected behavior

I expect to use the latest version without errors after upgrading.

### Additional context

I was able to get Mycodo working again after restoring. I restored immediatly since my Shiitake are fruiting at the moment so I was not able to grab additional logs. I can reproduce the issue if the logs are desired. (Or perhaps they are still stored somewhere?)

sjoerdschouten, 2021-09-25:
Any idea?

kizniche, 2021-09-28:
Did you try loading any page other than /index_page?

sjoerdschouten, 2021-10-05:
No, I immediately restored the backup.

kizniche, 2021-10-05:
The .upgrade file is only used to indicate that the upgrade log should be displayed to the user on the upgrade page. It has no effect whatsoever on the upgrade itself or the detection of an internet connection prior to an upgrade.