kizniche/Mycodo #1093: Install error  invalid user influxdb.influxdb

Chomski99, 2021-09-23:
STOP right now, and please first look to see if the issue you're about to submit is already an open or recently closed issue at https://github.com/kizniche/Mycodo/issues

Please DO NOT OPEN AN ISSUE:

 - If your Mycodo version is not the latest release version, please update your device before submitting your issue (unless your issue is related to not being able to upgrade). Your problem might already be solved.

 - If your issue has been addressed before. If you have any new information that may aid in solving the issue, post it in the issue that already exists.

If you are going to post a new issue, next read How to Write a Good Bug Report at https://forum.kylegabriel.com/t/how-to-write-a-good-bug-report/71

Please complete as many of the sections below, if applicable, to provide the most information that may help with investigating your issue. Replace the text in brackets with your own text describing the issue. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help those providing assistance by reducing the time required to help you.

### Describe the problem/bug

*[A clear and concise description of what you were trying to do and what the problem/bug is - replace this with your own text]*

### Versions:

 - Mycodo Version: *[e.g. 8.0.3]*
 - Raspberry Pi Version: *[e.g. 3B+]*
 - Raspbian OS Version: *[e.g. Buster Lite]*

### Reproducibility

Please list specific setup details that are involved and the steps to reproduce the behavior:

1. *[Go to...]*
2. *[Click on...]*
3. *[Scroll down to...]*
4. *[See error...]*
5. *[replace this list with your own text]*

### Expected behavior

*[A clear and concise description of what you expected to happen - replace this with your own text]*

### Screenshots

*[If applicable, add screenshots to help explain your problem - replace this with your own text]*

### Additional context

*[Anything that should be added to make it easier to address this issue - replace this with your own text]*

Chomski99, 2021-09-23:
I have a clean raspberry pi os installed on virtualbox. I'm trying to install Mycodo which fails with an invalid user influxdb.influxdb 
[setup.log](https://github.com/kizniche/Mycodo/files/7217037/setup.log).

So far all I have tried it to reboot and and rerun the setup script again. I used the latest version of PI os and Mycodo. Can anyone shed some light on the failure please.

kizniche, 2021-09-23:
Notice all the 404 errors installing packages. That's your issue.

Chomski99, 2021-09-23:
Thanks for coming back to me. After logging the issue I upgraded the PI os in a new virtual machine and found that helped but I still got the error once I got in to Mycodo. I wan't sure if this was because it wasn't running on a real raspberry PI. Any thoughts?

closed 2021-09-25 by Chomski99