kizniche/Mycodo #1095: Install Dependencies for I/O Expander: MCP23017 (16 Channels): On/Off

grux77, 2021-09-25:
RPi4 
Linux Mycodo 5.10.60-v7l+ 
Mycodo Version: 8.12.6
Python Version: 3.7.3 (default, Jan 22 2021, 20:04:44) [GCC 8.3.0]

Failure:
The I/O Expander: MCP23017 cannot be added.

Reproduction:
Try adding the  I/O Expander: MCP23017 in Setup/Output. There is a hint that dependencies have to be installed:
_"Dependencies not installed: adafruit_mcp4728"_. After the installation you can still not add the MCP23017 although the log does not show any errors.

My solution was:
First install the dependencies for  Digital-to-Analog Converters: MCP4728. Then the Expander MCP23017 can also be added.

kizniche, 2021-09-28:
Please paste the dependency log excerpt when you were prompted to install the dependency and you pressed the Install button.

grux77, 2021-09-29:
[2021-09-21 12:28:57] Dependency installation beginning. Installing: adafruit_mcp4728

[2021-09-21 12:28:57] 
[2021-09-21 12:28:57] #### Installing/updating adafruit-circuitpython-mcp230xx==2.4.6 (pip-pypi)
[2021-09-21 12:29:01] Collecting adafruit-circuitpython-mcp230xx==2.4.6
[2021-09-21 12:29:01]   Downloading adafruit-circuitpython-mcp230xx-2.4.6.tar.gz (32 kB)
[2021-09-21 12:29:06] Requirement already satisfied: Adafruit-Blinka in ./env/lib/python3.7/site-packages (from adafruit-circuitpython-mcp230xx==2.4.6) (6.2.2)
[2021-09-21 12:29:06] Requirement already satisfied: adafruit-circuitpython-busdevice in ./env/lib/python3.7/site-packages (from adafruit-circuitpython-mcp230xx==2.4.6) (5.0.5)
[2021-09-21 12:29:06] Requirement already satisfied: Adafruit-PlatformDetect>=3.1.0 in ./env/lib/python3.7/site-packages (from Adafruit-Blinka->adafruit-circuitpython-mcp230xx==2.4.6) (3.1.1)
[2021-09-21 12:29:06] Requirement already satisfied: pyftdi>=0.40.0 in ./env/lib/python3.7/site-packages (from Adafruit-Blinka->adafruit-circuitpython-mcp230xx==2.4.6) (0.52.9)
[2021-09-21 12:29:06] Requirement already satisfied: Adafruit-PureIO>=1.1.7 in ./env/lib/python3.7/site-packages (from Adafruit-Blinka->adafruit-circuitpython-mcp230xx==2.4.6) (1.1.8)
[2021-09-21 12:29:06] Requirement already satisfied: sysv-ipc>=1.1.0 in ./env/lib/python3.7/site-packages (from Adafruit-Blinka->adafruit-circuitpython-mcp230xx==2.4.6) (1.1.0)
[2021-09-21 12:29:06] Requirement already satisfied: rpi-ws281x>=4.0.0 in ./env/lib/python3.7/site-packages (from Adafruit-Blinka->adafruit-circuitpython-mcp230xx==2.4.6) (4.2.5)
[2021-09-21 12:29:06] Requirement already satisfied: RPi.GPIO in ./env/lib/python3.7/site-packages (from Adafruit-Blinka->adafruit-circuitpython-mcp230xx==2.4.6) (0.7.0)
[2021-09-21 12:29:06] Requirement already satisfied: pyserial>=3.0 in ./env/lib/python3.7/site-packages (from pyftdi>=0.40.0->Adafruit-Blinka->adafruit-circuitpython-mcp230xx==2.4.6) (3.5)
[2021-09-21 12:29:06] Requirement already satisfied: pyusb>=1.0.0 in ./env/lib/python3.7/site-packages (from pyftdi>=0.40.0->Adafruit-Blinka->adafruit-circuitpython-mcp230xx==2.4.6) (1.1.1)
[2021-09-21 12:29:06] Building wheels for collected packages: adafruit-circuitpython-mcp230xx
[2021-09-21 12:29:06]   Building wheel for adafruit-circuitpython-mcp230xx (setup.py): started
[2021-09-21 12:29:08]   Building wheel for adafruit-circuitpython-mcp230xx (setup.py): finished with status 'done'
[2021-09-21 12:29:08]   Created wheel for adafruit-circuitpython-mcp230xx: filename=adafruit_circuitpython_mcp230xx-2.4.6-py3-none-any.whl size=11377 sha256=a708054dab31d2b2cecabb2ce9671daf17842e478329a1e9a69b98ead7ab95f8
[2021-09-21 12:29:08]   Stored in directory: /root/.cache/pip/wheels/57/d9/ec/746cdba3aba1b49b215f9f92cc9e967aaf4a7a88c7fbfe795a
[2021-09-21 12:29:08] Successfully built adafruit-circuitpython-mcp230xx
[2021-09-21 12:29:09] Installing collected packages: adafruit-circuitpython-mcp230xx
[2021-09-21 12:29:10] Successfully installed adafruit-circuitpython-mcp230xx-2.4.6

[2021-09-21 12:29:10] End install of adafruit_mcp4728

[2021-09-21 12:29:10] 
[2021-09-21 12:29:10] #### Setting permissions

[2021-09-21 12:29:15] #### Dependencies installed. Restarting frontend and backend...

[2021-09-21 12:29:20] #### Dependency install complete.

[2021-09-21 17:57:15] Dependency installation beginning. Installing: adafruit_mcp4728

[2021-09-21 17:57:15] 
[2021-09-21 17:57:15] #### Installing/updating adafruit-circuitpython-mcp230xx==2.4.6 (pip-pypi)
[2021-09-21 17:57:16] Requirement already satisfied: adafruit-circuitpython-mcp230xx==2.4.6 in ./env/lib/python3.7/site-packages (2.4.6)
[2021-09-21 17:57:17] Requirement already satisfied: adafruit-circuitpython-busdevice in ./env/lib/python3.7/site-packages (from adafruit-circuitpython-mcp230xx==2.4.6) (5.0.5)
[2021-09-21 17:57:17] Requirement already satisfied: Adafruit-Blinka in ./env/lib/python3.7/site-packages (from adafruit-circuitpython-mcp230xx==2.4.6) (6.2.2)
[2021-09-21 17:57:17] Requirement already satisfied: pyftdi>=0.40.0 in ./env/lib/python3.7/site-packages (from Adafruit-Blinka->adafruit-circuitpython-mcp230xx==2.4.6) (0.52.9)
[2021-09-21 17:57:17] Requirement already satisfied: RPi.GPIO in ./env/lib/python3.7/site-packages (from Adafruit-Blinka->adafruit-circuitpython-mcp230xx==2.4.6) (0.7.0)
[2021-09-21 17:57:17] Requirement already satisfied: Adafruit-PureIO>=1.1.7 in ./env/lib/python3.7/site-packages (from Adafruit-Blinka->adafruit-circuitpython-mcp230xx==2.4.6) (1.1.8)
[2021-09-21 17:57:17] Requirement already satisfied: sysv-ipc>=1.1.0 in ./env/lib/python3.7/site-packages (from Adafruit-Blinka->adafruit-circuitpython-mcp230xx==2.4.6) (1.1.0)
[2021-09-21 17:57:17] Requirement already satisfied: rpi-ws281x>=4.0.0 in ./env/lib/python3.7/site-packages (from Adafruit-Blinka->adafruit-circuitpython-mcp230xx==2.4.6) (4.2.5)
[2021-09-21 17:57:17] Requirement already satisfied: Adafruit-PlatformDetect>=3.1.0 in ./env/lib/python3.7/site-packages (from Adafruit-Blinka->adafruit-circuitpython-mcp230xx==2.4.6) (3.1.1)
[2021-09-21 17:57:17] Requirement already satisfied: pyserial>=3.0 in ./env/lib/python3.7/site-packages (from pyftdi>=0.40.0->Adafruit-Blinka->adafruit-circuitpython-mcp230xx==2.4.6) (3.5)
[2021-09-21 17:57:17] Requirement already satisfied: pyusb>=1.0.0 in ./env/lib/python3.7/site-packages (from pyftdi>=0.40.0->Adafruit-Blinka->adafruit-circuitpython-mcp230xx==2.4.6) (1.1.1)

[2021-09-21 17:57:19] End install of adafruit_mcp4728

[2021-09-21 17:57:19] 
[2021-09-21 17:57:19] #### Setting permissions

[2021-09-21 17:57:25] #### Dependencies installed. Restarting frontend and backend...

[2021-09-21 17:57:29] #### Dependency install complete.

grux77, 2021-09-29:
and that led to success:

[2021-09-26 00:11:35] Dependency installation beginning. Installing: adafruit_mcp4728

[2021-09-26 00:11:35] 
[2021-09-26 00:11:35] #### Installing/updating adafruit-circuitpython-mcp4728==1.1.5 (pip-pypi)
[2021-09-26 00:11:37] Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
[2021-09-26 00:11:37] Collecting adafruit-circuitpython-mcp4728==1.1.5
[2021-09-26 00:11:38]   Downloading https://www.piwheels.org/simple/adafruit-circuitpython-mcp4728/adafruit_circuitpython_mcp4728-1.1.5-py3-none-any.whl (7.7 kB)
[2021-09-26 00:11:38] Requirement already satisfied: adafruit-circuitpython-busdevice in ./env/lib/python3.7/site-packages (from adafruit-circuitpython-mcp4728==1.1.5) (5.0.5)
[2021-09-26 00:11:38] Requirement already satisfied: Adafruit-Blinka in ./env/lib/python3.7/site-packages (from adafruit-circuitpython-mcp4728==1.1.5) (6.2.2)and that led to success
[2021-09-26 00:11:38] Requirement already satisfied: pyftdi>=0.40.0 in ./env/lib/python3.7/site-packages (from Adafruit-Blinka->adafruit-circuitpython-mcp4728==1.1.5) (0.52.9)
[2021-09-26 00:11:38] Requirement already satisfied: RPi.GPIO in ./env/lib/python3.7/site-packages (from Adafruit-Blinka->adafruit-circuitpython-mcp4728==1.1.5) (0.7.0)
[2021-09-26 00:11:38] Requirement already satisfied: Adafruit-PlatformDetect>=3.1.0 in ./env/lib/python3.7/site-packages (from Adafruit-Blinka->adafruit-circuitpython-mcp4728==1.1.5) (3.1.1)
[2021-09-26 00:11:38] Requirement already satisfied: Adafruit-PureIO>=1.1.7 in ./env/lib/python3.7/site-packages (from Adafruit-Blinka->adafruit-circuitpython-mcp4728==1.1.5) (1.1.8)
[2021-09-26 00:11:38] Requirement already satisfied: sysv-ipc>=1.1.0 in ./env/lib/python3.7/site-packages (from Adafruit-Blinka->adafruit-circuitpython-mcp4728==1.1.5) (1.1.0)
[2021-09-26 00:11:38] Requirement already satisfied: rpi-ws281x>=4.0.0 in ./env/lib/python3.7/site-packages (from Adafruit-Blinka->adafruit-circuitpython-mcp4728==1.1.5) (4.2.5)
[2021-09-26 00:11:38] Requirement already satisfied: pyserial>=3.0 in ./env/lib/python3.7/site-packages (from pyftdi>=0.40.0->Adafruit-Blinka->adafruit-circuitpython-mcp4728==1.1.5) (3.5)
[2021-09-26 00:11:38] Requirement already satisfied: pyusb>=1.0.0 in ./env/lib/python3.7/site-packages (from pyftdi>=0.40.0->Adafruit-Blinka->adafruit-circuitpython-mcp4728==1.1.5) (1.1.1)
[2021-09-26 00:11:39] Installing collected packages: adafruit-circuitpython-mcp4728
[2021-09-26 00:11:40] Successfully installed adafruit-circuitpython-mcp4728-1.1.5

[2021-09-26 00:11:40] End install of adafruit_mcp4728

[2021-09-26 00:11:40] 
[2021-09-26 00:11:40] #### Setting permissions

[2021-09-26 00:11:41] #### Dependencies installed. Restarting frontend and backend...

[2021-09-26 00:11:44] #### Dependency install complete.

kizniche, 2021-09-29:
I'm not understanding the issue. You're showing the log of the dependency successfully installing, which is the expected behavior.

kizniche, 2021-09-29:
Oh, I see the issue. The be MCP23017 output has the MCP4728 listed as the dependency, which is the wrong dependency. I'll fix that for the next release. Thanks for the bug report.

grux77, 2021-09-30:
That's it

closed 2021-12-03 by kizniche