kizniche/Mycodo #1096: No internet connection after restoring previous version of Mycodo

sjoerdschouten, 2021-09-26:
After restoring (from version 8.12.6) a version 8.11.0 (see #1092) the `upgrade` page shows the message below and prevents an update of Mycodo.

```
No internet connection detected. To upgrade Mycodo automatically, you will need an internet connection. Refresh the page when one is connected. 
```
The raspberry is connected to the internet and I have verified this with `ping`.

### Versions:
- Mycodo Version: 8.11.0
- Model: Raspberry Pi 4 Model B Rev 1.4
- Release:
- Distributor ID: Raspbian
- Description: Raspbian GNU/Linux 10 (buster)
- Release: 10
- Codename: buster

Firmware:
- Aug 3 2021 18:14:56
- Copyright (c) 2012 Broadcom
- version 40787ee5905644f639a2a0f6e00ae12e517a2211 (clean) (release) (start)

### Reproducibility

Please list specific setup details that are involved and the steps to reproduce the behavior:

1. Restore a backup of a previous version (in my case from `8.12.6` to  8.11`)
2. Upgrade button becomes red (to indicate there is an update)
3. See update page presenting the error
4. 
### Expected behavior
Not receiving a message about not having an internet connection (even though the Pi is connect to the internet) and being able to upgrade.

### Screenshots
https://imgur.com/FBVwLSk

kizniche, 2021-09-28:
See the upgrade option in the configuration

sjoerdschouten, 2021-10-05:
From the mycodo configuration --> upgrade interface

``` 
Upgrade Options
Internet Test IP Address
1.1.1.1
Internet Test Port
53
Internet Test Timeout
3
```

directly on the device:
``` 
sjoerd@mycodo:~ $ ping 1.1.1.1
PING 1.1.1.1 (1.1.1.1) 56(84) bytes of data.
64 bytes from 1.1.1.1: icmp_seq=1 ttl=57 time=23.5 ms
64 bytes from 1.1.1.1: icmp_seq=2 ttl=57 time=20.4 ms
64 bytes from 1.1.1.1: icmp_seq=3 ttl=57 time=22.8 ms
64 bytes from 1.1.1.1: icmp_seq=4 ttl=57 time=19.8 ms
64 bytes from 1.1.1.1: icmp_seq=5 ttl=57 time=19.8 ms
^C
--- 1.1.1.1 ping statistics ---
5 packets transmitted, 5 received, 0% packet loss, time 9ms
rtt min/avg/max/mdev = 19.799/21.274/23.532/1.582 ms

```

Any idea?

kizniche, 2021-10-05:
It's not pinging, it's using socket. https://github.com/kizniche/Mycodo/blob/9a5677cdc11b737274604427e489095d4850c2b8/mycodo/utils/system_pi.py#L564-L579

sjoerdschouten, 2021-10-05:
I managed to get rid of the error by performing the following work-around.

1. Go to `Configuration`
2. Go to the Diagnostics tab
3. Press `Delete File .upgrade`

Might be worthwhile keeping this issue around in case other people run into this problem.

kizniche, 2021-10-05:
>Delete File .upgrade

The .upgrade file does not affect internet detection.

kizniche, 2021-10-29:
I found the cause of this issue and pushed a fix, as well as another issue that contributed to this issue which copied the .upgrade file during a backup.

closed 2021-12-03 by kizniche