kizniche/Mycodo #1098: Install Issue - Error 502: Bad Gateway Error

SixtyTwoThirty, 2021-09-29:
After running through the Mycodo install process, I have the following connection problem:

**Error 502: Bad Gateway Error**

The last 50 lines from the Mycodo frontend web log are displayed below:

_journalctl -u mycodoflask | tail -n 50_

-- Logs begin at Wed 2021-09-29 18:03:06 AEST, end at Wed 2021-09-29 18:35:22 AEST. --
Sep 29 18:24:55 raspberrypi systemd[1]: Started gunicorn daemon for Mycodo flask web user interface.
Sep 29 18:24:55 raspberrypi systemd[5199]: mycodoflask.service: Failed to execute command: No such file or directory
Sep 29 18:24:55 raspberrypi systemd[5199]: mycodoflask.service: Failed at step EXEC spawning /var/mycodo-root/env/bin/gunicorn: No such file or directory
Sep 29 18:24:55 raspberrypi systemd[1]: mycodoflask.service: Main process exited, code=exited, status=203/EXEC
Sep 29 18:24:55 raspberrypi systemd[1]: mycodoflask.service: Failed with result 'exit-code'.

Can anyone provide me a solution?

kizniche, 2021-09-29:
You need to inspect the entire install log to determine the issue that occurred during your install, not just the last 50 lines.

SixtyTwoThirty, 2021-09-30:
The reason I looked at the last 50  lines as that was the instructions on the attempt to display the Mycodo Web Interface after the   installation:

> **Error 502: Bad Gateway Error**
> 
> Something bad happened but it's probably not your fault. Try refreshing the page to see if it will load. If that doesn't work, try rebooting your system, then try refreshing it again. If that still doesn't work, submit a new issue on GitHub with any usefull information that can help determine the cause of this issue.
> 
> The last 50 lines from the Mycodo frontend web log can be viewed with the command:
> 
> journalctl -u mycodoflask | tail -n 50
> If you create an issue on github, provide this excerpt of the log for diagnostic purposes.

I've had a look at the setup logs. Nothing stands out. What exactly should I be searching for in the set up logs?

I've also looked at [Mycodo Issues (Bug Reports/Feature Requests)](https://github.com/kizniche/Mycodo/issues) but I can't find any helpful solutions.

SixtyTwoThirty, 2021-10-01:
The setup logs were no help - no exceptions were raised during the install.

To resolve this issue, I deleted the Mycodo directory at /home/pi.

I then reinstalled the Mycodo at the terminal using the following commands:

```
sudo apt update
sudo apt upgrade
curl -L https://kizniche.github.io/Mycodo/install | bash
```

I hope this helps someone facing the same issue.

closed 2021-10-01 by SixtyTwoThirty

kizniche, 2021-10-02:
>The reason I looked at the last 50 lines as that was the instructions on the attempt to display the Mycodo Web Interface after the installation:

That command shows the last 50 lines of the frontend log, which is different from the setup log. The frontend can fail at any time. However, if you just installed and it's not working, it's likely an issue with the install, which requires inspecting the setup log. At the end of the install, the following is shown: "If you experience issues, review the full install log is located at... setup.log". If there was no error at all during the install, then it should work. Rerunning the install likely installed something that didn't install due to a network error during the first install attempt. This should have been indicated in the initial install log.