kizniche/Mycodo #1109: Error 502: Bad Gateway Error after update request

Rockytfox, 2021-10-29:
I was attempting to update Mycodo from 8.12.6 to 8.12.8 today.  After selecting the update, the raspberry pi became unresponsive and when trying to access the login page, I'm greeted with the Error 502:Bad Gateway Error message.  Attaching the output from "journalctl -u mycodoflask | tail -n 50" as stated on the Error 502 page.

pi@raspberrypi:~ $ journalctl -u mycodoflask | tail -n 50
-- Logs begin at Fri 2021-10-29 14:17:01 CDT, end at Fri 2021-10-29 15:08:58 CDT. --
Oct 29 14:17:06 raspberrypi systemd[1]: Started gunicorn daemon for Mycodo flask web user interface.
Oct 29 14:17:06 raspberrypi systemd[508]: mycodoflask.service: Failed to execute command: No such file or directory
Oct 29 14:17:06 raspberrypi systemd[508]: mycodoflask.service: Failed at step EXEC spawning /var/mycodo-root/env/bin/gunicorn: No such file or directory
Oct 29 14:17:06 raspberrypi systemd[1]: mycodoflask.service: Main process exited, code=exited, status=203/EXEC
Oct 29 14:17:06 raspberrypi systemd[1]: mycodoflask.service: Failed with result 'exit-code'.
pi@raspberrypi:~ $

kizniche, 2021-10-29:
Please attach the excerpt from the upgrade log for the entirety of the most current upgrade at /var/log/mycodo/mycodoupgrade.log

Rockytfox, 2021-10-29:
I'm sorry Kyle...can you remind me how to access that file?  I'm logged into the pi via terminal but can't remember the command

kizniche, 2021-10-29:
To obtain just the relevant excerpt, you should download the file with an FTP client (Filezilla is a good cross-platform client).

susingz, 2021-10-30:
I have attached the mycodoupgrade.log file
I have the same issue. 
[mycodoupgrade.log](https://github.com/kizniche/Mycodo/files/7445981/mycodoupgrade.log)

kizniche, 2021-10-30:
502 is a generic error that the frontend is not running. I.e. you have no if your issues are related.

kizniche, 2021-10-30:
@susingz no idea what's wrong with your system.

Rockytfox, 2021-10-30:
[mycodoupgrade.log](https://github.com/kizniche/Mycodo/files/7446849/mycodoupgrade.log)

Kyle, here is the upgrade log you requested.

Rockytfox, 2021-10-30:
Update.  I am back up and running.

closed 2021-10-30 by Rockytfox