kizniche/Mycodo #1112: Security Vulnerability

haby0, 2021-11-16:
The program has a log injection vulnerability.

kizniche, 2021-11-21:
Thank you. I received your email with the details. I've been busy lately and haven't had a chance to address it.

kizniche, 2021-12-12:
Is there a way to set this so it gets inherited by all loggers that use logging.getLogger()?

I see 100+ results searching for "logger = logging", and the proposed change is only one of them.

haby0, 2021-12-13:
There should be a Logger tool class.

haby0, 2021-12-13:
mycodo\utils\logging_utils.py
```python
import logging
import anticrlf

def set_logger(logger_):
    handler = logging.StreamHandler()
    handler.setFormatter(anticrlf.LogFormatter('%(levelname)s:%(name)s:%(message)s'))
    logger_.addHandler(handler)
    return logger_

```

mycodo\mycodo_client.py
```python
from mycodo.utils.logging_utils import set_logger

logging.basicConfig(
    stream=sys.stdout,
    level=logging.INFO,
    format='%(asctime)s %(message)s'
)
logger = set_logger(logging.getLogger(__name__))
```

@kizniche Is this way possible?

kizniche, 2021-12-13:
I will test. Thanks for the example.

haby0, 2021-12-21:
I will close this issus first.

closed 2021-12-21 by haby0