kizniche/Mycodo #1114: 502 Gateway error

Toastybunz1, 2021-11-24:
### Describe the problem/bug
My problem is that mycodo will not install properly. I get a 502 gateway error. here's the journal.
-- Journal begins at Sat 2021-10-30 12:29:42 BST, ends at Wed 2021-11-24 02:51:33 GMT. --
Nov 24 01:04:07 raspberrypi systemd[1]: Started gunicorn daemon for Mycodo flask web user interface.
Nov 24 01:04:07 raspberrypi systemd[4398]: mycodoflask.service: Failed to locate executable /var/mycodo-root/env/bin/gunicorn: No such file or directory
Nov 24 01:04:07 raspberrypi systemd[4398]: mycodoflask.service: Failed at step EXEC spawning /var/mycodo-root/env/bin/gunicorn: No such file or directory
Nov 24 01:04:07 raspberrypi systemd[1]: mycodoflask.service: Main process exited, code=exited, status=203/EXEC
Nov 24 01:04:07 raspberrypi systemd[1]: mycodoflask.service: Failed with result 
'exit-code'.

### Versions:

 - Mycodo Version: 8.12.8
 - Raspberry Pi Version: Zero
 - Raspbian OS Version: Bullseye Lite
### Reproducibility

Please list specific setup details that are involved and the steps to reproduce the behavior:

Installing, deleting, reinstalling. Tried this 5 times today.

Also I seen previous issues on the forum like this and Ive tried all those options too. Any command with "pip" in it wont work. either dose not do anything or its not a directory.

kizniche, 2021-11-24:
You will need to inspect the setup.log that's generated in the install/ directory for the exact issue during install.

Toastybunz1, 2021-11-24:
Says -bash: /home/pi/Mycodo/install/setup.log: Permission denied. I also tried the root way and same outcome

Toastybunz1, 2021-11-24:
Just tried a fresh install aswell

kizniche, 2021-11-25:
Use an FTP client to connect and download setup.log and attach it to a comment here via a web browser.

kizniche, 2021-11-25:
Or, if you have a monitor and mouse connected to your Pi, just make a comment from the browser of the Pi.

Toastybunz1, 2021-11-25:
Solved the issue by reinstalling the full OS vs the new lite

closed 2021-12-06 by kizniche