kizniche/Mycodo #1117: Raspberry Pi OS - Bullseye libcamera changes

cstallings, 2021-11-28:
This is perhaps less a bug and more a functionality issue due to major camera changes in the latest version of Raspberry Pi OS (Bullseye). As you probably know, the newest version of RPi OS has completely changed the camera libraries to libcamera (https://libcamera.org/entries/2020-05-05.html) . I'm currently working with a Pi Camera Rev 1.3 on both a RPi4b and a Zero2w. I'm able to get the camera to work with the RPi4b and the Zero 2w however I'm unsure if its compatible with Mycodo currently? 
1. Does Mycodo work with libcamera?
2. If not, are there plans to support it?
3. 
Thanks!

**Versions:**

 - Mycodo Version: 8.12.8 
 - Raspberry Pi Version: 4b 8gb  and Zero2w
 - Raspbian OS Version: Bullseye

kizniche, 2021-12-03:
This issue has been mentioned on **DIY Automation Forum**. There might be relevant details there:

https://forum.kylegabriel.com/t/can-i-use-libcamera-library-on-bullseye-os/510/3

kizniche, 2021-12-19:
This issue has been resolved with commit https://github.com/kizniche/Mycodo/commit/fd7d07b47d881a3ccc42368f9282ef571c92c83a

closed 2022-03-28 by kizniche