kizniche/Mycodo #1120: ModuleNotFoundError: No module named '_sqlite3'

anyunfeifei, 2021-12-07:
I have finished install the Mycodo,when I run it on 127.0.0.1,it runs out Error 502: Bad Gateway Error,in /home/pi/Mycodo/mycodo,I  runned sudo /var/mycodo-root/env/bin/gunicorn --workers 1 --worker-class gthread --pid /var/lock/mycodoflask.pid --bind unix:/var/run/mycodoflask.sock start_flask_ui:app,it comes out from _sqlite3 import *,When  journalctl -u mycodoflask | tail -n 50 ,it comes out the same error "_sqlite3 import *"
,

anyunfeifei, 2021-12-07:
When I import sqlite3,there is no error occur。And I have install it by use "sudo apt-get install libsqlite3-dev"

kizniche, 2021-12-08:
What operating system are you installing on?

kizniche, 2021-12-08:
>I have install it by use "sudo apt-get install libsqlite3-dev"

Mycodo runs in a virtualenv so it's best you don't attempt to install packages unless you are very familiar with pip and virtualenvironments.

anyunfeifei, 2021-12-09:
> What operating system are you installing on?

I installed it on Raspberry

anyunfeifei, 2021-12-09:
I used "curl -L https://kizniche.github.io/Mycodo/install | bash" to install Mycodo,then it comes out the error

kizniche, 2022-03-28:
v8.13.0 was just released. If you have the same issue with the latest version, feel free to reopen.

closed 2022-03-28 by kizniche