kizniche/Mycodo #1121: Cant access to Dashboard

ser6iogl92, 2021-12-08:
Hello everyone, today i decide to install mycodo in my new raspberry 4b, all the installation looks success.
1-i decide to access via 127.0.0.1 and shows 127.0.0.1/create_admin.
2- I put all the user data(really simple characters)
3-i press enter or press ''create login'' button 
4-shows again the same page with black fields. I tried some times in diferents Web browsers and still the same. All the services is running good, dont know where the problem came.
Any suggestion? 
Note: whe i access localhost 127.0.0.1 shows message the connection is not secure but i read other posts thats normal then i didnt put so much interest in that.

Versions:

 - Mycodo Version: 8.12.9
 - Raspberry Pi Version: 4B
 - Raspbian OS Version: Debian bullseye

thanks in advance

kizniche, 2021-12-08:
What command(s) did you use to install?

ser6iogl92, 2021-12-08:
i follow instructions in your web page , the commands:
curl -L https://kizniche.github.io/Mycodo/install | bash

kizniche, 2021-12-08:
What is the output from the command ```ls -l ~/Mycodo/databases``` ?

ser6iogl92, 2021-12-08:
total 552
drwxrwxr-x 3 mycodo mycodo   4096 dic  3 04:09 alembic
-rw-rw-r-- 1 mycodo mycodo   1446 dic  3 04:09 alembic.ini
-rw-rw-r-- 1 mycodo mycodo  71966 dic  3 04:09 alembic_post.py
-rw-rw-r-- 1 mycodo mycodo    555 dic  3 04:09 alembic_post_utils.py
-rw-rw-r-- 1 mycodo mycodo     64 dic  8 20:09 flask_secret_key
-rw-rw-r-- 1 mycodo mycodo     15 dic  3 04:09 __init__.py
-rw-rw-r-- 1 mycodo mycodo 458752 dic  8 20:53 mycodo.db
-rw-rw-r-- 1 mycodo mycodo    520 dic  8 20:55 statistics.csv
-rw-rw-r-- 1 mycodo mycodo     10 dic  8 20:10 statistics.id

kizniche, 2021-12-08:
nothing looks out of the ordinary. I've never seen this issue before. I recommend reinstalling the OS and Mycodo.

kizniche, 2021-12-08:
Did you install the Lite or Desktop version of the OS?

kizniche, 2021-12-08:
I always use the Lite version and have never had an issue. Others that have used the desktop version have reported numerous issues. You can always install the GUI after install of the Lite version.

ser6iogl92, 2021-12-08:
okay i will try that. I think maybe is delivered from something about SO. I tried to install updates and says the clock is not sychronized - cannot install updates. try again in a few minutes. Thats normal? the clock is in the good date

ser6iogl92, 2021-12-09:
i installed raspbian lite and run all the process again and i still have the same issue. I put the name, email, password push enter and blank field again. I will try to do the same out of this localNet.

ser6iogl92, 2021-12-09:
i just was thinking maybe dont have permissions to write data to create a new user in mycodo. then cant go trough the program and stuck on the same action. i really dont have knowledge about that, just what my imagination can get.

ser6iogl92, 2021-12-09:
Is strange but i will try resolve. Maybe sd card is not good. I just bought raspberry pi 4B this week and connect in a monitor screen and put this commands. that the reason i can not think in what i did bad because i didnt manipulate  it before.

ser6iogl92, 2021-12-09:
i changed keyboard to spanish and select the brand of the keyboard to have acces to this character  "|" for enter the install commands. Thats the only thing i did different from the Mycodo installing instructions. I changed SD Card and I test in another Raspberry pi 3, Still the same problem. Other option can be if i have a routing problem. the only thing i know about my router is the technician add a static ip. I think it can not make interference in the access of the database.
I will apreciate a comment where can tell me a idea about what can fix that.
Thanks so much

kizniche, 2021-12-09:
Try entering English characters.

ser6iogl92, 2021-12-09:
Happy i say i can create user now!  the problem was i did so short password... omg so simple 
thank for all your help i am exciting to do the proyect :D

ser6iogl92, 2021-12-09:
the thing is i used to find the message that is short password other times i had this issue. I think will be good the program shows that message when is not satisfying the rules.

kizniche, 2021-12-09:
Glad to hear it's working. It should provide the error message if you enter erroneous data in the form fields. I'll check on that.

craSH, 2022-01-24:
Hi - just wanted to say that I had the exact same issue here, and it also surprisingly worked once I used a 100 character random password, vs. a ~16 character generated one.
Raspberry Pi OS 11 / Full on a Pi4 8GB. Installed via the same curl/bash method above. Nothing seen in log files, including when restarting Mycodo in debug mode.

closed 2022-03-18 by kizniche