kizniche/Mycodo #1122: part of mcp324x using python2 syntax?

nccappello, 2021-12-09:
### Describe the problem/bug

mcp342x (analog input) input does not populate values

mycodo.log contains the following:
ERROR - mycodo.inputs.mcp342x_ab51a19d - InputModule raised an exception when taking a reading: 'dict' object has no attribute 'iteritems'

### Versions:

 - Mycodo Version: 8.12.9
 - Raspberry Pi Version: 4 Model B
 - Raspbian OS Version: Raspbian GNU/Linux 10 (buster)

### Reproducibility

Please list specific setup details that are involved and the steps to reproduce the behavior:

1. add input
2. activate input

### Expected behavior

values should populate (ADS1115_CP is similar in that it's a 4ch ADC which works, SHT4X populates values without issue as well)

### Screenshots

N/A

### Additional context

new to this software, may be missing something. Can't determine where 'iteritems' is called.. searched through scripts within the home/pi/Mycodo folder with no hits

kizniche, 2021-12-19:
I can't find iteritems() anywhere in the library that's used ([python-MCP342x](https://github.com/stevemarple/python-MCP342x)).

nccappello, 2022-01-06:
I added a traceback to the mycodo/inputs/base_input.py file where the error message originated and it lead me to /var/mycodo-root/env/lib/python3.7/site-packages/MCP342x/__init__.py - I changed the syntax in this file and the sensor works fine. I will push the changes in the next few days after I wrap up a few other things.

kizniche, 2022-01-28:
What do you mean you'll push the changes? What you changed is a file of the MCP342x library, not Mycodo.

closed 2022-03-28 by kizniche