kizniche/Mycodo #1123: Conditional Controller function throws an error Conditional ID not found

vipul-khatana, 2021-12-10:
STOP right now, and please first look to see if the issue you're about to submit is already an open or recently closed issue at https://github.com/kizniche/Mycodo/issues

Please DO NOT OPEN AN ISSUE:

 - If your Mycodo version is not the latest release version, please update your device before submitting your issue (unless your issue is related to not being able to upgrade). Your problem might already be solved.

 - If your issue has been addressed before. If you have any new information that may aid in solving the issue, post it in the issue that already exists.

If you are going to post a new issue, next read How to Write a Good Bug Report at https://forum.kylegabriel.com/t/how-to-write-a-good-bug-report/71

Please complete as many of the sections below, if applicable, to provide the most information that may help with investigating your issue. Replace the text in brackets with your own text describing the issue. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help those providing assistance by reducing the time required to help you.

### Describe the problem/bug

*I am using a raspberry pi along with Atlas pH sensor and Atlas pump and LCD display to automate the monitoring process of just pH of the water. All the devices are recognized when I use `sudo i2cdetect -y 1` I have added pH sensor in the input and pump in the outputs. As indicated in your project, I have created a controller function to implement the algorithm. 

However, when I activate the function, first of all I get a warning `Warning: pylint returned with code 127`. Then when I check the Daemon logs it shows : `2021-12-10 10:32:00,777 - DEBUG - mycodo.controllers.controller_conditional_0109b937 - Conditional check. pH: None
2021-12-10 10:33:00,628 - ERROR - mycodo.function_actions - Condition ID not found`*

### Versions:

 - Mycodo Version: *8.12.9*
 - Raspberry Pi Version: *4*
 - Raspbian OS Version: *Bullseye*

### Reproducibility

Please list specific setup details that are involved and the steps to reproduce the behavior:

1. *Go to Setup*
2. *Click on Functions*
3. *Chose Conditional Controller. Remove the lines where EC measurement is required, as we do not use an EC sensor. Replace the ids of the components with the ID from Setup -> Input and click on  "Atlas pH" option to copy the UUID*
4. *After compiling the function, go to Daemon Logs to see the error which is also pasted below*
5. *
[image]
*

### Expected behavior

*Condtional ID being recognized and emails being sent for dangerously high pH values*

### Screenshots

*`2021-12-10 10:28:50,533 - DEBUG - mycodo.controllers.controller_conditional_0109b937 - Conditional Statement (pre-replacement):
### Edit below to set desired ranges for pH and electrical conductivity ###

# Desired range for pH
range_ph_high = 6.2
range_ph_low = 5.5
# pH range that will immediately cause a pH correction
range_ph_high_danger = 7.0
range_ph_low_danger = 5.0

print(range_ph_high)
### Edit below to set the IDs for Conditions and Actions ###
condition_id_measurement_ph_id = "8398f815-8356-4a4a-b475-b44d7a5c67aa"  # Condition: measurement, last, pH Input

action_id_pump_1_acid = "bfb22612-153b-4603-b907-112630869018"  # Action: Pump 1 (Acid)

action_id_email_notification = "{EMAIL_ID}"  # Action: Email Notification
### DO NOT EDIT BELOW THIS LINE UNLESS YOU KNOW WHAT YOU ARE DOING ###
import time
if 'notify_ph' not in self.variables:  # Initiate pH notification timer
    self.variables['notify_ph'] = 0
if 'notify_none' not in self.variables:  # Initiate None measurement notification timer
    self.variables['notify_none'] = 0

measure_ph = self.condition(condition_id_measurement_ph_id)
self.logger.debug("Conditional check. pH: {}".format(measure_ph))
if None in [measure_ph]:
[... 102 lines cut ...]
            self.logger.debug("pH is {}. Should be > {}. Dispensing 1 ml base".format(measure_ph, range_ph_low))

        elif measure_ph > range_ph_high:  # pH too high, add acid (pH down)
            self.logger.debug("pH is {}. Should be < {}. Dispensing 1 ml acid".format(measure_ph, range_ph_high))
            self.run_action(action_id_pump_1_acid)  # Dispense 1 ml acid (pH down)

    def function_status(self):
        # Example code to provide a return status for other controllers and widgets.
        status_dict = {
            'string_status': "This is the demo status of the conditional controller. "
                             "The controller has looped {} times".format(self.loop_count),
            'loop_count': self.loop_count,
            'error': []
        }
        return status_dict

2021-12-10 10:29:00,709 - ERROR - mycodo.function_actions - Condition ID not found
2021-12-10 10:29:00,710 - DEBUG - mycodo.controllers.controller_conditional_0109b937 - Conditional check. pH: None
2021-12-10 10:30:00,576 - ERROR - mycodo.function_actions - Condition ID not found`*

### Additional context

*Am I using the wrong `PH_MEASURE_ID`? Is the line wrong `ondition_id_measurement_ph_id = "8398f815-8356-4a4a-b475-b44d7a5c67aa"`? 

I am pasting the UUID I get when I click at Setup->Input and then Atlas pH option which automatically copies the UUID to the clipboard *

kizniche, 2021-12-10:
>Replace the ids of the components with the ID from Setup -> Input

You are using Input IDs instead of the Condition IDs. Conditions are created within the Conditional Controller, therefore you need to get the IDs from the Conditions listed in the Conditional Controller you are editing the Conditional Statement of.

vipul-khatana, 2021-12-10:
[image]

Thanks for the reply. Is this highlighted term the one that I need to copy?

kizniche, 2021-12-10:
Correct

closed 2021-12-10 by vipul-khatana