kizniche/Mycodo #1124: RockPI / ubuntu installation issues: netcat, wget, daemon

amirdt22, 2021-12-18:
### Describe the problem/bug

There are a few issues I encountered during the install on a RockPI flashed with Ubuntu 20.04

1. curl install script assumes wget is installed (minor)
2. setup never finished, hung on restarting the daemon (logs below)
3. nc (netcat) was not installed and I got errors when creating an input (could not connect to :9090)

Once I installed netcat and rebooted I was able to create an input without error.

### Versions:

 - Mycodo Version: *8.12.9*
 - Raspberry Pi Version: *RockPi 4B+ v1.72*
 - Raspbian OS Version: *Ubuntu 20.04* (I now realize, maybe this only works with raspbian?)

### Expected behavior

Perhaps I shouldn't expect it to work on ubuntu, in that case would debian be a better choice?  I see there is a docker option as well as some discussion here about other OSes: https://github.com/kizniche/Mycodo/issues/897

I don't think I have the option to install "raspbian"
https://wiki.radxa.com/Rockpi4/downloads

kizniche, 2021-12-18:
It should work with most debian-based operating systems, which includes Ubuntu.

netcat was included in the install recently in https://github.com/kizniche/Mycodo/commit/91a871fea225cb0c7c25ea50170e2fa07636fea4 and has yet to be released.

I just updated the install script to install wget. Thanks.

>setup never finished, hung on restarting the daemon (logs below)

I don't see any logs attached.

kizniche, 2021-12-18:
>I see there is a docker option

This is still pretty experimental. I have slowly been improving the docker setup, though you are likely to find issues. If you have experience with docker and want to give it a shot and potentially help fix some bugs, try it out. It's definitely nicer to work with in a lot of ways since the build environment is stable.

amirdt22, 2021-12-18:
Thanks, logs included this time.

I do have and love docker, I'll give it a go.

Btw, really appreciate your work here and excited to get started!

<details>
  <summary>setup.log</summary>
### Mycodo installation initiated Sat Dec 18 17:36:46 UTC 2021
#### Mycodo installation began Sat Dec 18 17:37:00 UTC 2021

#### Checking if swap size is 100 MB and needs to be changed to 512 MB
#### Swap not currently set to 100 MB. Not changing.

#### Updating apt repositories
Hit:1 http://apt.radxa.com/focal-stable focal InRelease
Hit:2 http://ports.ubuntu.com/ubuntu-ports focal InRelease
Hit:3 http://ports.ubuntu.com/ubuntu-ports focal-security InRelease
Hit:4 http://ports.ubuntu.com/ubuntu-ports focal-updates InRelease
Hit:5 http://ports.ubuntu.com/ubuntu-ports focal-backports InRelease
Reading package lists...

#### Uninstalling apt version of pip (if installed)
Reading package lists...
Building dependency tree...
Reading state information...
Package 'python-pip' is not installed, so not removed
0 upgraded, 0 newly installed, 0 to remove and 217 not upgraded.

#### Installing prerequisite apt packages and update pip
Reading package lists...
Building dependency tree...
Reading state information...
Package 'python3-cffi-backend' is not installed, so not removed
Package 'python-cffi-backend' is not installed, so not removed
Package 'apache2' is not installed, so not removed
0 upgraded, 0 newly installed, 0 to remove and 217 not upgraded.
Reading package lists...
Building dependency tree...
Reading state information...
python3 is already the newest version (3.8.2-0ubuntu2).
wget is already the newest version (1.20.3-1ubuntu2).
The following additional packages will be installed:
  blt build-essential bzip2 cpp cpp-9 dpkg-dev fakeroot g++-9 gcc-10-base
  gcc-9 gcc-9-base git-man less libalgorithm-diff-perl
  libalgorithm-diff-xs-perl libalgorithm-merge-perl libasan5 libatomic1
  libc-dev-bin libc6 libc6-dev libcc1-0 libcrypt-dev liberror-perl
  libexpat1-dev libfakeroot libgcc-9-dev libgcc-s1 libgcc1 libgd3 libgomp1
  libio-pty-perl libipc-run-perl libisl22 libitm1 liblsan0 libmpc3 libmpfr6
  libnginx-mod-http-image-filter libnginx-mod-http-xslt-filter
  libnginx-mod-mail libnginx-mod-stream libpython3-dev libpython3.8
  libpython3.8-dev libpython3.8-minimal libpython3.8-stdlib libsqlite3-0
  libstdc++-9-dev libstdc++6 libtcl8.6 libtime-duration-perl libtimedate-perl
  libtk8.6 libtsan0 libubsan1 libxpm4 libxslt1.1 linux-libc-dev make manpages
  manpages-dev nginx-common nginx-core patch pylint python-pip-whl
  python3-astroid python3-distutils python3-isort python3-lazy-object-proxy
  python3-lib2to3 python3-logilab-common python3-mccabe
  python3-pylint-plugin-utils python3-tk python3-wheel python3-wrapt python3.8
  python3.8-dev python3.8-minimal tk8.6-blt2.5 zlib1g zlib1g-dev
Suggested packages:
  blt-demo bzip2-doc cpp-doc gcc-9-locales debian-keyring gcc-9-doc gawk-doc
  gcc-multilib automake libtool flex bison gdb gcc-doc gettext-base
  git-daemon-run | git-daemon-sysvinit git-doc git-el git-email git-gui gitk
  gitweb git-cvs git-mediawiki git-svn glibc-doc locales libgd-tools
  libstdc++-9-doc tcl8.6 tk8.6 bsd-mailx | mailx make-doc man-browser fcgiwrap
  nginx-doc ssl-cert ed diffutils-doc pylint-doc python-setuptools-doc tix
  python3-tk-dbg python3.8-venv python3.8-doc binfmt-support sqlite3-doc
The following NEW packages will be installed:
  blt build-essential bzip2 cpp cpp-9 dpkg-dev fakeroot g++ g++-9 gawk gcc
  gcc-9 gcc-9-base git git-man less libalgorithm-diff-perl
[... 1042 lines cut ...]

#### Connecting to http://localhost (creates Mycodo database if it doesn't exist)
#### Could not connect to http://localhost. Waiting 60 seconds then trying again (up to 10 times)...
#### Trying again...
#### Successfully connected to http://localhost

#### Setting permissions

#### Restarting the Mycodo daemon

**********************************
** ERROR During Mycodo Install! **
**********************************

An error occurred that may have prevented Mycodo from
being installed properly!

Open to the end of the setup log to view the full error:
/home/amir/Mycodo/install/setup.log

Please contact the developer by submitting a bug report
at https://github.com/kizniche/Mycodo/issues with the
pertinent excerpts from the setup log located at:
/home/amir/Mycodo/install/setup.log
</details>

amirdt22, 2021-12-18:
And for context here:

```
Restarting the Mycodo daemon
** ERROR During Mycodo Install! **

An error occurred that may have prevented Mycodo from
being installed properly!
```

This was hung for 5-10 minutes.  I hit CTRL+C and that's the "error".  I can try a fresh install with netcat included and perhaps that will take care of it?

kizniche, 2021-12-18:
I just checked and the daemon indeed could not start without netcat, which was the impetus for https://github.com/kizniche/Mycodo/commit/91a871fea225cb0c7c25ea50170e2fa07636fea4.

closed 2021-12-18 by amirdt22