kizniche/Mycodo #1125: i2C Multiplexer TCA9545a, SHT31 address constraints

jacopienaar, 2021-12-21:
I have a a sht31 sensor and a generic 16x2 display connected to an TCA9545a multiplexer and can see in system information that the 2 devices are picked up on bus 2, but am unable to select the below addresses when adding a sensor.

When adding the display I cannot get anyhting to display on it using the  below 0x50 and 0x3a adresses. I'm not sure what I am doing wrong.

I2C Bus 1: i2cdetect -y 1
       0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         08 09 0a 0b 0c 0d 0e 0f 
10: 10 11 12 13 14 15 16 17 18 19 1a 1b 1c 1d 1e 1f 
20: 20 21 22 23 24 25 26 27 28 29 2a 2b 2c 2d 2e 2f 
30: 30 31 32 33 34 35 36 37 38 39 3a 3b 3c 3d 3e 3f 
40: 40 41 42 43 44 45 46 47 48 49 4a 4b 4c 4d 4e 4f 
50: 50 51 52 53 54 55 56 57 58 59 5a 5b 5c 5d 5e 5f 
60: 60 61 62 63 64 65 66 67 68 69 6a 6b 6c 6d 6e 6f 
70: 70 71 72 73 74 75 76 77                         

I2C Bus 2: i2cdetect -y 2
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- 3a -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: 50 -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         

When trying to add the SHT31x, I am limited to choosing between 0x44 and 0x45 adresses and cannot add anything else. How can I add this sensor?

kizniche, 2021-12-21:
>When trying to add the SHT31x, I am limited to choosing between 0x44 and 0x45 adresses

Those are the only addresses that sensor has. It's not capable of having alternate addresses.

jacopienaar, 2021-12-21:
Thanks Kyle. So I cnnot use it with a multiplexer then? I also noticed when Unplugging the inputs that the i2cdetect remains unchanged - think something is wrong with it.

kizniche, 2021-12-21:
You can use it with a multiplexer, you just cant change the I2C address to anything except 0x44 or 0x45.

kizniche, 2021-12-21:
It doesn't look like your multiplexer is correctly set up and working. For instance, you likely do not have 127 devices connected to bus 1.

jacopienaar, 2022-02-06:
I've come to the conclusion the the multiplexer(s) are faulty. Cannot get sensor addresses to feed through the multiplexer

closed 2022-02-06 by jacopienaar