kizniche/Mycodo #1127: I reinstall the mycodo,there is a new Issues run out.

anyunfeifei, 2021-12-21:
pi@raspberrypi:~ $ journalctl -u mycodoflask | tail -n 50
-- Logs begin at Thu 2019-02-14 18:11:59 CST, end at Tue 2021-12-21 22:26:23 CST. --
12月 21 22:22:32 raspberrypi systemd[1]: Started gunicorn daemon for Mycodo flask web user interface.
12月 21 22:22:36 raspberrypi gunicorn[507]: Traceback (most recent call last):
12月 21 22:22:36 raspberrypi gunicorn[507]:   File "/var/mycodo-root/env/bin/gunicorn", line 5, in <module>
12月 21 22:22:36 raspberrypi gunicorn[507]:     from gunicorn.app.wsgiapp import run
12月 21 22:22:36 raspberrypi gunicorn[507]: ModuleNotFoundError: No module named 'gunicorn'
12月 21 22:22:36 raspberrypi systemd[1]: mycodoflask.service: Main process exited, code=exited, status=1/FAILURE
12月 21 22:22:36 raspberrypi systemd[1]: mycodoflask.service: Failed with result 'exit-code'.

kizniche, 2021-12-22:
>No module named 'gunicorn'

There's your problem

anyunfeifei, 2021-12-23:
I‘ve installed the gunicorn.’

kizniche, 2022-01-28:
gunicorn must be installed in the virtualenvironment that Mycodo runs in. If it's not installed during setup, you will need to review the setup.log to determine what issue is preventing it form being installed. More often than not it's an network issue related to your internet connection.

closed 2022-01-28 by kizniche