kizniche/Mycodo #1132: SCD41

jacopienaar, 2022-01-06:
### Describe the problem/bug

Installed SCD4x dependency and connected Seeed SCD41 Sensor (https://wiki.seeedstudio.com/Grove-CO2_%26_Temperature_%26_Humidity_Sensor-SCD41/). Sensor is detected but no reading is obtained. 

### Versions:

 - Mycodo Version: 8.12.9
 - Raspberry Pi Version:  3B+
 - Raspbian OS Version: Latest

### Reproducibility
 This is done on a fresh install with dtparam=i2c_vc=on added in boot config.txt. 
I have three of these sensors and have tried them all - same result.

### Expected behavior

To obtain sensor readings

### Screenshots

[image]
[image]

### Additional context
The daemon log show the following error message:

`2022-01-07 09:24:47,450 - ERROR - mycodo.inputs.scd4x_circuitpython_24681be8 - Cannot set a value of None. Must be a float or string representing a float. Check the sensor and Input module is working correctly.
2022-01-07 09:24:47,453 - ERROR - mycodo.inputs.scd4x_circuitpython_24681be8 - Cannot set a value of None. Must be a float or string representing a float. Check the sensor and Input module is working correctly.
2022-01-07 09:24:47,456 - ERROR - mycodo.inputs.scd4x_circuitpython_24681be8 - Cannot set a value of None. Must be a float or string representing a float. Check the sensor and Input module is working correctly.
2022-01-07 09:24:47,457 - ERROR - mycodo.inputs.scd4x_circuitpython_24681be8 - Cannot set a value of None. Must be a float or string representing a float. Check the sensor and Input module is working correctly.
2022-01-07 09:24:47,457 - ERROR - mycodo.inputs.scd4x_circuitpython_24681be8 - Cannot set a value of None. Must be a float or string representing a float. Check the sensor and Input module is working correctly.
2022-01-07 09:24:47,668 - ERROR - mycodo.inputs.sht31d_circuitpython_e269eaca - Input not set up
2022-01-07 09:24:47,759 - ERROR - mycodo.inputs.scd30_b462b47a - InputModule.get_measurement() method raised IOError: [Errno 121] Remote I/O error`

I think it might be something to do with the dependencies/library of the SCD41. For ease of reference, the [adruino i2c scd4x library can be found here]( https://github.com/Sensirion/arduino-i2c-scd4x) and the [adruino core dependency can be found here.](https://github.com/Sensirion/arduino-core) Both these links are given on the sensor product page (link posted above).

jacopienaar, 2022-01-08:
Another issue I noticed about the SCD41 is that it doesn't show up when connected over a cable with 2-3m length. I have a the SCD30 connected over the same cable and works just fine. I encountered the same with the SHT31 which make me think it is a sensor limitation. I've searched far and wide for info on cable length limitation and cannot find anyhting and was wondering by the off chance if you might be able to shed some light on this?

aaronarnold2, 2022-01-25:
Hi. I had/ have the same issue as described above (with same error logs). I verified that my SCD41 sensor was working properly on another RPi. I "fixed" the issue by copying and pasting the input from the git repository and uploading it as a custom input (renamed it to "test_scd41). The only change I made was to add "measurements_select" to the options enabled list. It now works, but I don't know why.

jacopienaar, 2022-01-25:
This is a huge relief that you got it working - I bought a whole bunch of these sensors that's just sitting around atm!

Do you mind sharing the test_scd41 file? Not sure what you mean with 'input' and where to copy it from, else i'd try and do it myself.

kizniche, 2022-01-25:
>The only change I made was to add "measurements_select" to the options enabled list. It now works, but I don't know why.

Doing this doesn't have any affect on how measurements are acquired. You're likely just associating something you changed at the same time it started working.

kizniche, 2022-01-25:
>IOError: [Errno 121] Remote I/O error

This indicates there's an issue communicating with the sensor.

jacopienaar, 2022-01-25:
That's what I thought but it is showing up in i2cdetect and like aaronarnold2 said, I also confirmed the sensor is working on another pi (not running Mycodo). I'm a bit stumped tbh...

kizniche, 2022-01-25:
I just pushed a change that updates adafruit-circuitpython-scd4x from 1.2.1 to 1.2.2, which may fix issues with the library that communicates with the sensor. You will need to [upgrade to master](https://github.com/kizniche/Mycodo/wiki/Updating-to-Github-Master-Branch) to test if the new code works.

jacopienaar, 2022-01-26:
Thanks Kyle - I'll gave it a try but no joy. Daemon log as follows:

`During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/base_controller.py", line 74, in run
    self.initialize_variables()
  File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 317, in initialize_variables
    self.measure_input = input_loaded.InputModule(self.input_dev)
  File "/home/pi/Mycodo/mycodo/inputs/scd4x_circuitpython.py", line 69, in __init__
    self.initialize_input()
  File "/home/pi/Mycodo/mycodo/inputs/scd4x_circuitpython.py", line 75, in initialize_input
    self.sensor = adafruit_scd4x.SCD4X(
  File "/var/mycodo-root/env/lib/python3.9/site-packages/adafruit_scd4x.py", line 97, in __init__
    self.i2c_device = i2c_device.I2CDevice(i2c_bus, address)
  File "/var/mycodo-root/env/lib/python3.9/site-packages/adafruit_bus_device/i2c_device.py", line 50, in __init__
    self.__probe_for_device()
  File "/var/mycodo-root/env/lib/python3.9/site-packages/adafruit_bus_device/i2c_device.py", line 163, in __probe_for_device
    raise ValueError("No I2C device at address: 0x%x" % self.device_address)
ValueError: No I2C device at address: 0x62
2022-01-26 00:14:42,196 - INFO - mycodo.controllers.controller_input_5cada7bc - Activated in 681.2 ms
2022-01-26 00:14:42,197 - ERROR - mycodo.controllers.controller_input_5cada7bc - Mycodo is attempting to acquire measurement(s) from an Input that has already critically errored. Review the log lines following Input Activation to investigate why this happened.
2022-01-26 00:14:42,197 - INFO - mycodo.daemon - All activated Input controllers started
2022-01-26 00:14:42,198 - INFO - mycodo.daemon - All activated Math controllers started
2022-01-26 00:14:42,198 - INFO - mycodo.daemon - All activated PID controllers started
2022-01-26 00:14:42,199 - INFO - mycodo.daemon - All activated LCD controllers started
2022-01-26 00:14:42,199 - INFO - mycodo.daemon - All activated Function controllers started
2022-01-26 00:14:42,370 - INFO - mycodo.controllers.controller_widget - Activated in 170.4 ms
2022-01-26 00:14:43,373 - INFO - mycodo.daemon - Mycodo daemon started in 5.050 seconds
2022-01-26 00:14:43,395 - INFO - mycodo.daemon - 45.77 MB RAM in use
2022-01-26 00:14:56,927 - ERROR - mycodo.controllers.controller_input_5cada7bc - Mycodo is attempting to acquire measurement(s) from an Input that has already critically errored. Review the log lines following Input Activation to investigate why this happened.`

kizniche, 2022-01-26:
>ValueError: No I2C device at address: 0x62

See above. Additionally, I added measurements_select (as it should be included to allow disabling measurements), but like I said, has no affect on actually acquiring measurements.

Your issue likely has nothing to do with Mycodo and is a result of your setup of the sensor, wiring, power supply, or operating system.

kizniche, 2022-01-26:
>`During handling of the above exception, another exception occurred:

You're also not proving the entire log.

jacopienaar, 2022-01-26:
That's because it just repeats:

OSError: [Errno 121] Remote I/O error

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/var/mycodo-root/env/lib/python3.9/site-packages/adafruit_bus_device/i2c_device.py", line 160, in __probe_for_device
    self.i2c.readfrom_into(self.device_address, result)
  File "/var/mycodo-root/env/lib/python3.9/site-packages/busio.py", line 156, in readfrom_into
    return self._i2c.readfrom_into(address, buffer, stop=stop)
  File "/var/mycodo-root/env/lib/python3.9/site-packages/adafruit_blinka/microcontroller/generic_linux/i2c.py", line 56, in readfrom_into
    readin = self._i2c_bus.read_bytes(address, end - start)
  File "/var/mycodo-root/env/lib/python3.9/site-packages/Adafruit_PureIO/smbus.py", line 181, in read_bytes
    return self._device.read(number)
OSError: [Errno 121] Remote I/O error

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/base_controller.py", line 74, in run
    self.initialize_variables()
  File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 317, in initialize_variables

kizniche, 2022-01-26:
"OSError: [Errno 121] Remote I/O error" is a different error than "ValueError: No I2C device at address: 0x62".

Have you begin testing parts of your system for faults? wiring, power supply, operating system?

kizniche, 2022-01-26:
Additionally, what does "Raspbian OS Version: Latest" mean? Can you provide better detail?

kizniche, 2022-01-26:
Also, why did you put "dtparam=i2c_vc=on" in config.txt and not use raspi-config? That is not the line raspi-config adds to enable I2C.

jacopienaar, 2022-01-26:
I currently have the sensor connected to a vanilla install of the latest raspbian (Raspbian Linux version 11 'bullseye) and mycodo  (8.12.9). The SHT31D and SCD30 sensors both work when connected but not the SCD41. On this basis I assume that the the issue either sits with sensor connections or sensor libraries. I've checked and confirmed that it is correct and even used the breakout pins on the side of the sensor board to see if it might be the grove connection, but also no luck.

> Also, why did you put "dtparam=i2c_vc=on" in config.txt and not use raspi-config? That is not the line raspi-config adds to enable I2C.

I did this to open the other i2c bus 0 (breakout pins 27 & 28) to be able to connect two i2c devices since the i2c multiplexer I bought is not working (or Im doing something wrong). I have successfully acquired measurements from both bus 0 & 1 with two sensors connected at the same time. I've also had a generic  12x8 LCD connected to bus 0 which worked fine for a while but would then stop working after a while. 

I can disable "dtparam=i2c_vc=on" and see if that works, however, the sensor is currently connected to i2c bus 1 and not 0?

kizniche, 2022-01-26:
I would try to simplify your setup in order to get just this sensor working. This means disconnecting all other devices except for the single sensor. Reducing wire lengths as much as you can. Ensuring the power supply is sufficient to power the Pi and sensor. Using default settings such as enabling I2C from raspi-config. Once the sensor is working you can start making changes, one or few at a time, until you're back at your current configuration and you determine at what step along the way causes the issue with your sensor.

jacopienaar, 2022-01-26:
I have two test setups running with the sensor connected on i2c bus 1 and detected by Mycodo (8.12.9) on 0x62. Both are vanilla installs, i.e new micro SD card, Raspbian Linux V.11 followed by sudo apt get upgrade and update commands.  Both i2c and ssh is enabled. Then Mycodo is installed followed by the SCD41x dependency installation. The sensor is connected, detected but no reading obtained. I've repeated this on a Pi Zero 2 W and 3B+ models with the same error messages and outcomes. With both I am using the official raspberry pi power supply. 
Subsequent edits that you may see above ("dtparam=i2c_vc=on" or other dependency installs) were made after the initial installs and sensor test outlined here.

With both setups I've use 20cm (7.87") jumper wires and which I have replaced with brand new ones a few times to make sure these are not the issue. I've tried connecting through the grove port and through the breakout on the side of the sensor - same results - sensor is detected but no reading obtained.
Other sensors (SCD30 & SHT31D) when connected to the same pins the SCD41 were connected to does work (detected and obtain reading).

All the above is leading me to believe that it is software related - but this is an uneducated guess really, I'm taking shots in the dark here - any other suggestions / potential root causes?

kizniche, 2022-01-26:
Can you get measurements to be acquired with the [adafruit-circuitpython-scd4x](https://github.com/adafruit/Adafruit_CircuitPython_SCD4X) library outside Mycodo?

jacopienaar, 2022-01-26:
I did manage to get it working outside of Mycodo in December, but cannot recall what I did/which tutorial I followed. Will try and recreate it over to weekend and confirm with all sensors (previously only tested 1). I only have basic working knowledge of these things and have to re-learn every time I pick it up again and with my two brain cells fighting over third place doesn't help matters either...

jacopienaar, 2022-01-29:
So I confirmed that the sensor is working outside of Mycodo

[image]

kizniche, 2022-01-30:
I just noticed your log's error lines (https://github.com/kizniche/Mycodo/issues/1132#issuecomment-1021731133) indicate you did not upgrade to master as I instructed (https://github.com/kizniche/Mycodo/issues/1132#issuecomment-1021721204). You must upgrade to master to get the latest code. If not, you will assuredly keep getting the same errors. Follow my instruction in comment

jacopienaar, 2022-01-30:
Idiot over here didn't read the last line where I had to mannually go hit the ugrade button. I just updated config.py and restarted mycodoflask. So completed all of it correctly and ran into some errors and restarted and it worked!!!

[image]

jacopienaar, 2022-01-30:
Hey Kyle - thank you so much for you patience and help - it is much appreciated!

jacopienaar, 2022-01-30:
So I updated the Period and after activating the input again it couldn't reactivate and gave the following error message "Error: Altitude(m) is require to be set". A restart didn't fix it. Any idea why it is askinf for alititude?

kizniche, 2022-01-30:
>Any idea why it is askinf for alititude?

Altitude must be set to a value in order to activate the Input. Did you change or delete the value?

jacopienaar, 2022-01-30:
No - I deactivated it so I could change the period and tried reactivating it but was met with that message. I didn't see a setting for altitude in the sensor configuration

kizniche, 2022-01-30:
[image]

slashie, 2022-01-30:
I had exactly the same problem
`2022-01-30 08:07:03,341 - ERROR - mycodo.inputs.scd4x_circuitpython_98f8eb1f - Cannot set a value of None. Must be a float or string representing a float. Check the sensor and Input module is working correctly.`

Dropped this into shell
`sudo mycodo-commands upgrade-master`

Then rebooted.  

Things to note: I did have a SCD4x input before updating,  but no alt or co2 calibration options available in UI.  Install script doesn't pull the very latest sensor module. I suspect next release will have it baked in.

It worked on reboot, with alt of 0, deactivated, and would not reactivate without putting alt in meters.
Entered the required field of altitude and all is well.

Using both Adafruit SCD40 and SCD41 modules

[image]

jacopienaar, 2022-01-30:
> [image]

Well wadda you know!  Sorry about that, I was halfway out the house yesterday with a screaming todler in tow - the brain circuitry melts and the ability to multitask goes out the windows. All sorted.

Thanks for all the help!

closed 2022-01-30 by jacopienaar