kizniche/Mycodo #1134: set baudrate of i2c bus

lawn-gnome, 2022-01-07:
**Is your feature request related to a problem? Please describe.**
YES, I need to retain the custom baudrate on the i2c bus, that is being set by my line in the config.txt file:
"dtparm=i2c_baudrate=10000"  <<<This allows my css811 adafruit sensor to work with i2cdetect. HOWEVER as soon as the daemon starts the bus speed settings changes (I think), and the sensor is no longer present on the bus. This causes "remote i/o" errors, and device not present errors.

**Describe the solution you'd like**
I deally this might be an option for i2c settigns in the control panel. 

**Describe alternatives you've considered**
For the time being,  can You point me to the place where I can change this in the backend?

**Additional context**
Not gonna lie this software is a bit complicated for Me, But eventually I will get it. It is pretty bad ass!

lawn-gnome, 2022-01-07:
Sooo.........

I found out that for some reason my pi is intializing the primary i2c bus twice. Once at my 10k setting during boot, and then again once mycodo starts up at 100k. the 100k setting is not compatible with my adafruit sensor. I looked into activating the secondary i2c bus, but I have not figured out how to set the baudrate on the secondary bus.

kizniche, 2022-01-07:
Mycodo itself does not do anything with I2C. Perhaps you're referring to an Input that you added that uses a 3rd party library.

kizniche, 2022-01-28:
I found that ExtendedI2C() that is used with circuitpython in Mycodo Inputs sets a default I2C frequency of 400kHz. This was discovered when updating the SCD-30 Input, when I was adding the ability to set the frequency (see https://github.com/kizniche/Mycodo/commit/79220879a48ecf44e6b9599612e56e236bce682e#diff-b84791a758463732231da4d8d1164fb60d326f3015da42d6f95d320d6c078a8cR175). This is something you may want to look into that could be causing your issue.

kizniche, 2022-02-13:
Here is the Input with the changes I applied to the SCD-30. You will need to rename the excension from txt to py, then inport on the Config -> Custom Inputs page, then add it on the Inputs page. It has in its name "FREQUENCY TEST 01".

[ccs811_CP_test1.txt](https://github.com/kizniche/Mycodo/files/8055975/ccs811_CP_test1.txt)

closed 2022-03-28 by kizniche