kizniche/Mycodo #1135: AnyLeaf pH Probe Won't Calibrate

bviolante, 2022-01-07:
Hey Kyle 

Before i explain the issue i'm having i would like to say thank you for creating such an awesome open source piece of software. Very easy to use and so useful. My Issue, I'm trying to calibrate my AnyLeaf pH probe but i keep getting this error after i click the calibrate, slot 2 button. I enter my first buffer of 4.0, hit the calibrate, slot button 1 and then notice my reading changes to around 4pH. I enter my second buffer solution of 7.0, hit the calibrate, slot button 2 and then i have no data on the live data viewing page for that input. The following logs: 

2022-01-07 18:20:25,467 - ERROR - mycodo.inputs.anyleaf_ph_9ad5adea - InputModule raised an exception when taking a reading: float division by zero
2022-01-07 18:20:29,797 - ERROR - mycodo.inputs.anyleaf_ph_9ad5adea - InputModule raised an exception when taking a reading: float division by zero
2022-01-07 18:20:44,855 - ERROR - mycodo.inputs.anyleaf_ph_9ad5adea - InputModule raised an exception when taking a reading: float division by zero
2022-01-07 18:20:44,856 - ERROR - mycodo.controllers.controller_input_9ad5adea - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.

Now, if i clear the calibration slots, deactivate my input, reactive my input. I get a reading and i have communication again thhe the probe. So far I've tried

Calibrating 1 slot at a time and deactivating then activating the input, and it still fails
Tried Calibrating from a different slot first, and it still fails. 
Tried Calibrating all three slots, and this still fails 

Any help would be appreciated 

Thanks Mate!

kizniche, 2022-01-07:
Try enabling Log Level: Debug and you should get a traceback of the error that will be more informative.

bviolante, 2022-01-07:
Thanks for the Tip. I can see an OSError: [Errno 121] Remote I/O error so may be an issue with my cabling :(

kizniche, 2022-01-07:
The traceback indicates there's something about your Cal1 and/or Cal2 values that are causing a division by zero in that equation. Perhaps you should change the default second calibration point value prior to going through the calibration to prevent the issue.

bviolante, 2022-01-10:
Unfortunately still no luck. Do you think if I had accidentally been sent an orp probe instead of a pH probe this could possibly cause that error?

bviolante, 2022-01-10:
Ok. I think I've been doing that wrong. How do you change the default second calibration value prior to calibration?

bviolante, 2022-01-11:
Looks like there is an issue with the probe after speaking to the Manufacture. They are sending out a new one so hopefully that resolves this issue.

closed 2022-03-28 by kizniche

bviolante, 2022-05-20:
FYI I ordered a new any leaf PH module. This resolved the issue as the one I had was faulty.