kizniche/Mycodo #1138: Cannot update/delete a camera

mattimatti, 2022-01-08:
### Describe the problem/bug

A camera created cannot be deleted  nor its parameters updated

### Versions:

 - Mycodo Version: 8.12.9
 - Raspberry Pi Version: 3B+
 - Raspbian OS Version: bullseye

### Reproducibility

Please list specific setup details that are involved and the steps to reproduce the behavior:

1. Go to the camera page "/camera"
2. Click on an existing camera plus button too expand the panel
3. click the "Delete" button at first row of the panel or modify any field and clik Save

### Expected behavior

I expect the camera to be deleted or updated 

### Screenshots

### Additional context

Is it possible that the form action does not match the route?

Form action declared 

https://github.com/kizniche/Mycodo/blob/09351dcbc11aba1c84be7765a8050792a3df4222/mycodo/mycodo_flask/templates/pages/camera.html#L113

Route hanlinh delete/update

https://github.com/kizniche/Mycodo/blob/671e67b47509bd3c2f90f4edb093396b39900a79/mycodo/mycodo_flask/routes_page.py#L135

kizniche, 2022-01-08:
I cannot reproduce the issue. The camera page buttons work fine for me. Perhaps you're using an out of date browser.

mattimatti, 2022-01-08:
i'm using safari  Version 13.1.3 (15609.4.1) on macos catalina 10.15.7. 
yes indeed it was safari somehow preventing the call.  iIswitched to chrome after allowing the certificate. (tricky on mac)
and all works. So far so good!

Btw thank you for this big thing you carry on. Wil try to help out.

closed 2022-01-08 by mattimatti