kizniche/Mycodo #1139: Mycodo installation hangs. Possibly database related.

CedricHanebaum, 2022-01-16:
### Description
I was trying to install Mycodo, using the command from the readme, but the install hangs after a while. I've waited for more than an hour and nothing more has happened. The last messages being printed are
`
Could not connect to http://localhost. Waiting 60 seconds then trying again (up to 10 times)...
Trying again...
`

I've looked into the install script and tried to run
`wget --no-check-certificate -p http://localhost/`
manually. This is the output:
`
--2022-01-16 17:56:33--  http://localhost/
Resolving localhost (localhost)... ::1, 127.0.0.1
Connecting to localhost (localhost)|::1|:80... failed: Connection refused.
Connecting to localhost (localhost)|127.0.0.1|:80... connected.
HTTP request sent, awaiting response... 
`
After that it just hangs. I assume for the same reason as the install script.

If I try to open Mycodo in my browser from my PC I get a 504 Gateway time-out. nginx seems to be working.

I have taken a look into /var/log/mycodo/mycodo.log and it is full of errors. Most of it being repetitions of this one:
`
2022-01-16 17:21:31,010 - ERROR - mycodo.database - The Mycodo database is locked. Trying to access again in 1 second...
2022-01-16 17:21:32,209 - ERROR - mycodo.databases.utils - Error raised in session_scope.  Session will be rolled back: 
`
I'll attach the full log below.

I have tried reinstalling Mycodo by moving the original Mycodo directory and executing the command from the readme again. The results were the same.

### Versions:

 - Mycodo Version: 8.12.9 (Or so I assume. The version the install script in the readme installs)
 - Raspberry Pi Version: Pi Zero W
 - Raspbian OS Version: Raspbian 11 Bullseye

### Reproducibility
1. Execute the command `curl -L https://kizniche.github.io/Mycodo/install | bash`
2. Wait

I've tried it twice and at least on my system I can reproduce this. I assume that's not true for other systems though, or else there would probably be a bug report for this already.

### Expected behavior
The installation finishes and Mycodo can be accessed through the browser.

### Additional context
The Raspbian install is relatively fresh. I have however done two things with the Pi so far.
1. Experiment with and test the sensors I want to use on my project. This required installing some python libraries
2. Create a Mycodo user

Not sure If that could have any influence on the installation process though.

### Log file
[mycodo.log](https://github.com/kizniche/Mycodo/files/7877614/mycodo.log)
I can supply further logs if necessary.

kizniche, 2022-01-28:
The daemon log is not much use to diagnose a setup issue. Can you please attach the setup.log found in the install directory?

dylandn, 2022-02-09:
Cedric, are you on the 64bit Raspbian?

kizniche, 2022-03-18:
Possible duplicate of #1149

closed 2022-03-28 by kizniche

CedricHanebaum, 2022-03-28:
Sorry for not keeping up with this. Just to give this some closure from my end, I wiped my SD card and reinstalled Raspbian. On the fresh install, Mycodo installed without problems.

kizniche, 2022-03-28:
Good to hear. I just released v8.13.0 and was cleaning up old issues.