kizniche/Mycodo #1140: Cannot remove LCD hardware without crashing Mycodo

littlepackage, 2022-01-19:
### Describe the problem/bug

I am not needing the LCD input so I thought I'd deactivate the LCD function and unplug the screen. No such luck. See below -

### Versions:

 - Mycodo Version: 8.12.9
 - Raspberry Pi Version: 11
 - Raspbian OS Version: Bullseye

### Reproducibility

Please list specific setup details that are involved and the steps to reproduce the behavior:

1. Configure a Pi with Mycodo as per your instructions at https://kylegabriel.com/projects/2021/09/mushroom-cultivation-automation.html#Install_Raspberry_Pi_OS. Nothing special
2. Add a function to add a 20x4 LCD screen
3. Run it a week or two. The LCD and Mycodo work fine, outputs as expected, no surprises.
4. Uh oh: Try to delete the LCD from the installation because you no longer want/need LCD output. Unplug it.
5. Entire system breaks when next function runs:

`
2022-01-18 17:46:18,433 - DEBUG - mycodo.devices.atlas_scientific_i2c_111_1 - AtlasScientificI2C raised an exception when taking a reading: [Errno 121] Remote I/O error
2022-01-18 17:46:18,434 - DEBUG - mycodo.devices.atlas_scientific_i2c_111_1 - AtlasScientificI2C raised an exception when taking a reading: [Errno 121] Remote I/O error
2022-01-18 17:46:19,436 - DEBUG - mycodo.devices.atlas_scientific_i2c_111_1 - AtlasScientificI2C raised an exception when taking a reading: [Errno 121] Remote I/O error
2022-01-18 17:46:19,436 - ERROR - mycodo.inputs.atlas_humidity_8322b362 - Sensor read unsuccessful (after 2 attempts): [Errno 121] Remote I/O error
2022-01-18 17:46:19,437 - DEBUG - mycodo.devices.atlas_scientific_i2c_111_1 - AtlasScientificI2C raised an exception when taking a reading: [Errno 121] Remote I/O error
2022-01-18 17:46:19,437 - ERROR - mycodo.inputs.atlas_humidity_8322b362 - InputModule raised an exception when taking a reading: argument of type 'OSError' is not iterable
2022-01-18 17:46:19,438 - ERROR - mycodo.controllers.controller_input_8322b362 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2022-01-18 17:47:59,048 - WARNING - mycodo.controllers.controller_pid_2fac27b9 - No data returned from influxdb
2022-01-18 17:47:59,048 - DEBUG - mycodo.controllers.controller_pid_2fac27b9 - Last measurement unsuccessful. Turning outputs off.
2022-01-18 17:47:59,055 - DEBUG - mycodo.controllers.controller_output - output_on_off(c912abf0-d051-4e68-b846-c568898f9e2e, off, 2, None, 0.0, 0.0, True)
2022-01-18 17:48:18,489 - DEBUG - mycodo.devices.atlas_scientific_i2c_111_1 - AtlasScientificI2C raised an exception when taking a reading: [Errno 121] Remote I/O error
2022-01-18 17:48:18,490 - DEBUG - mycodo.devices.atlas_scientific_i2c_111_1 - AtlasScientificI2C raised an exception when taking a reading: [Errno 121] Remote I/O error
2022-01-18 17:48:19,491 - DEBUG - mycodo.devices.atlas_scientific_i2c_111_1 - AtlasScientificI2C raised an exception when taking a reading: [Errno 121] Remote I/O error
2022-01-18 17:48:19,492 - ERROR - mycodo.inputs.atlas_humidity_8322b362 - Sensor read unsuccessful (after 2 attempts): [Errno 121] Remote I/O error
2022-01-18 17:48:19,493 - DEBUG - mycodo.devices.atlas_scientific_i2c_111_1 - AtlasScientificI2C raised an exception when taking a reading: [Errno 121] Remote I/O error
2022-01-18 17:48:19,493 - ERROR - mycodo.inputs.atlas_humidity_8322b362 - InputModule raised an exception when taking a reading: argument of type 'OSError' is not iterable
2022-01-18 17:50:18,434 - DEBUG - mycodo.devices.atlas_scientific_i2c_111_1 - AtlasScientificI2C raised an exception when taking a reading: [Errno 121] Remote I/O error
2022-01-18 17:50:18,435 - DEBUG - mycodo.devices.atlas_scientific_i2c_111_1 - AtlasScientificI2C raised an exception when taking a reading: [Errno 121] Remote I/O error
2022-01-18 17:50:19,437 - DEBUG - mycodo.devices.atlas_scientific_i2c_111_1 - AtlasScientificI2C raised an exception when taking a reading: [Errno 121] Remote I/O error
2022-01-18 17:50:19,438 - ERROR - mycodo.inputs.atlas_humidity_8322b362 - Sensor read unsuccessful (after 2 attempts): [Errno 121] Remote I/O error
2022-01-18 17:50:19,439 - DEBUG - mycodo.devices.atlas_scientific_i2c_111_1 - AtlasScientificI2C raised an exception when taking a reading: [Errno 121] Remote I/O error
2022-01-18 17:50:19,439 - ERROR - mycodo.inputs.atlas_humidity_8322b362 - InputModule raised an exception when taking a reading: argument of type 'OSError' is not iterable
2022-01-18 17:50:58,990 - WARNING - mycodo.controllers.controller_pid_2fac27b9 - No data returned from influxdb
2022-01-18 17:50:58,991 - DEBUG - mycodo.controllers.controller_pid_2fac27b9 - Last measurement unsuccessful. Turning outputs off.
2022-01-18 17:50:58,998 - DEBUG - mycodo.controllers.controller_output - output_on_off(c912abf0-d051-4e68-b846-c568898f9e2e, off, 2, None, 0.0, 0.0, True)
2022-01-18 17:51:04,210 - WARNING - mycodo.controllers.controller_pid_9b6b04b6 - No data returned from influxdb
2022-01-18 17:51:04,211 - DEBUG - mycodo.controllers.controller_pid_9b6b04b6 - Last measurement unsuccessful. Turning outputs off.`

6. Delete LCD function and restart system frontend/backend. No joy. 
7. Plug the LCD back in and restart the system and things will go back to functioning.

### Expected behavior

Is there a graceful way to delete/deactivate components such as this LCD? Other components, such as the CO2 pump, did not cause similar issues.

### Additional context

THANK YOU. I'm really enjoying this software and appreciate your work.

kizniche, 2022-01-28:
>unplug the screen

Unplugging an I2C device while the system is energized is not recommended, as you may cause issues with other sensors. There's no reason why adding or removing an LCD Function would disrupt I2C devices. I'm not sure what the issue is, but it's likely the result of wiring, power supply, sensor choice, or other configuration.

littlepackage, 2022-01-30:
The system is not energized while doing this. It's shut down and unplugged, after deactivating the LCD function. Say I wanted to remove an LCD from the system to replace with a different one. Any recommendations - because I'm stumped - or stupid.

kizniche, 2022-01-31:
I have no idea what could be causing your issue, but it's likely not Mycodo. It's likely your circuit. Have you disconnected everything and connected one sensor at a time?

closed 2022-03-28 by kizniche