kizniche/Mycodo #1142: Adding CH4 UART (similar to K30 CO2) for soil GHG sensor

alonrab, 2022-01-21:
**Got new research to test soil greenhouse gas, that needs CH4 reading

**Describe the solution you'd like**

I would like to add the  Cubic CU-SJH-5 sensor provided by GASLAB, https://www.co2meter.com/collections/ch4-sensors/products/infrared-methane-ch4-gas-sensor?variant=31489082163318
It is the same company that makes the K30 CO2 sensor. I have purchased one CH4 CU-SJH-5 with a USB UART connection and looking into placing it in the field this spring

**Describe alternatives you've considered**
There are a lot of similar infrared sensors, however, I think since this one uses similar code to the K30 sensor it made sense.
**Additional context** 
Here is the datasheet
[Manual-Industrial-Grade-NDIR-Gas-Sensors-Cubic-GasLab.pdf](https://github.com/kizniche/Mycodo/files/7915174/Manual-Industrial-Grade-NDIR-Gas-Sensors-Cubic-GasLab.pdf)

Many thanks!

kizniche, 2022-01-21:
Do you have any Python code that works with communicating with the sensor? If so, creating a new Input would be relatively simple.

alonrab, 2022-01-21:
I see, I'm trying to get one from the manufacturer and searching for one online,