kizniche/Mycodo #1144: PID Controller: band is stopping calculation in one direction mode

chiefbrain, 2022-01-23:
The PID calculation stops if the following conditions are true:
- Direction: Raise or Lower
- Band: not 0
- Measurement > Setpoint + Band (Raise) or Measurement < Setpoint - Band (Lower)

Thus, the integral value never falls.

[image]

Example for temperature and Raise: If the temperature exceeds the green line (setpoint + band), the calculation of the error is not done, so the integral is not updated. Since only positive errors are added, the integral only raises.
If the band is 0, the PID calculates the expected error and the expected integral.
The integral is approximated by straight lines in the graph but should show the idea.

This is caused by check_hysteresis returning no setpoint in e.g. https://github.com/kizniche/Mycodo/blob/master/mycodo/utils/pid_controller_default.py#L131, which stops the calculation in https://github.com/kizniche/Mycodo/blob/master/mycodo/utils/pid_controller_default.py#L60.

#1008 seems related. 

Mycodo Version: 8.12.9

I'm looking for a fix and publish a merge request once done.

kizniche, 2022-01-23:
Interesting. I haven't thoroughly explored use cases, but it seems what you describe is undesirable behavior. I would need to re-familiarize myself with my PID code to figure out a solution. If you have a fix in mind, please share and I'll review it.

chiefbrain, 2022-01-28:
On direction raise, this is the current behaviour (`resultingError = self.setpoint - measure`):

```
#         input           #        output           #
# measure  # allowRaising # setpoint # allowRaising #  resultingError  #
########################################################################
#  < min   #    both      #   max    #     true     #     > 2*band     #
#    min   #    both      #   none   #   noChange   #       none       #
# between  #    false     #   none   #     false    #       none       #
# between  #    true      #   max    #     true     #  0 < e < 2*band  #
#    max   #    both      #   none   #   noChange   #       none       #
#  > max   #    both      #   none   #     false    #       none       #
```

To prevent an always rising integral, the following change is needed:
```
#         input           #        output           #
# measure  # allowRaising # setpoint # allowRaising #  resultingError  #
########################################################################
#  < min   #    both      #   max    #     true     #     > 2*band     #
#    min   #    both      #   none   #   noChange   #       none       #
# between  #    false     #   none   #     false    #       none       #
# between  #    true      #   max    #     true     #  0 < e < 2*band  #
#    max   #    both      #   none   #   noChange   #       none       #
#  > max   #    both      #   min    #     false    #     < 2*band     # <= changed
```

To keep the output steady, the pid should not abort but force the error to zero, thus the integral does not change:
```
#         input           #        output           #
# measure  # allowRaising # setpoint # allowRaising #  resultingError  #
########################################################################
#  < min   #    both      #   max    #     true     #     > 2*band     #
#    min   #    both      #   max    #     true     #      2*band      # <= changed
# between  #    false     # measure  #     false    #         0        # <= changed
# between  #    true      #   max    #     true     #  0 < e < 2*band  #
#    max   #    both      #   max    #     true    #         0        # <= changed
#  > max   #    both      #   min    #     false    #     < -2*band    #
```
This is the same as
```
#         input           #        output           #
# measure  # allowRaising # setpoint # allowRaising #  resultingError  #
########################################################################
#  <= min  #    both      #   max    #     true     #     > 2*band     #
# between  #    false     # measure  #   noChange   #         0        # <= changed
# between  #    true      #   max    #     true     #  0 < e < 2*band  #
#  > max   #    both      #   min    #     false    #     < -2*band    #
```
My solution:
```
        # direction raise/lower
        # measure  # setpoint # resultingError #
        ########################################
        #  < min   # max      #   > 2*band     #
        # between  # measure  #      0         #
        #  > max   # min      #   < -2*band    #

chiefbrain, 2022-01-28:
On direction raise, this is the current behaviour (`resultingError = self.setpoint - measure`):

```
#         input           #        output           #
# measure  # allowRaising # setpoint # allowRaising #  resultingError  #
########################################################################
#  < min   #    both      #   max    #     true     #     > 2*band     #
#    min   #    both      #   none   #   noChange   #       none       #
# between  #    false     #   none   #     false    #       none       #
# between  #    true      #   max    #     true     #  0 < e < 2*band  #
#    max   #    both      #   none   #   noChange   #       none       #
#  > max   #    both      #   none   #     false    #       none       #

Note: resultingError > 0 if setpoint < measure < max
```

To prevent an always rising integral, the following change is needed:
```
#         input           #        output           #
# measure  # allowRaising # setpoint # allowRaising #  resultingError  #
########################################################################
#  < min   #    both      #   max    #     true     #     > 2*band     #
#    min   #    both      #   none   #   noChange   #       none       #
# between  #    false     #   none   #     false    #       none       #
# between  #    true      #   max    #     true     #  0 < e < 2*band  #
#    max   #    both      #   none   #   noChange   #       none       #
#  > max   #    both      #   min    #     false    #     < 2*band     # <= changed
```

To keep the output steady, the pid should not abort but force the error to zero, thus the integral does not change:
```
#         input           #        output           #
# measure  # allowRaising # setpoint # allowRaising #  resultingError  #
########################################################################
#  < min   #    both      #   max    #     true     #     > 2*band     #
#    min   #    both      #   max    #   noChange   #      2*band      # <= changed
# between  #    false     # measure  #     false    #         0        # <= changed
# between  #    true      #   max    #     true     #  0 < e < 2*band  #
#    max   #    both      #   max    #  noChange    #         0        # <= changed
#  > max   #    both      #   min    #     false    #     < -2*band    #
```

This is the same as:
```
#         input           #        output           #
# measure  # allowRaising # setpoint # allowRaising #  resultingError  #
########################################################################
#  < min   #    both      #   max    #     true     #     > 2*band     #
# between  #    false     # measure  #     false    #         0        #
# between  #    true      #   max    #     true     # 0 <= e <= 2*band #
#  > max   #    both      #   min    #     false    #     < -2*band    #

Note: resultingError still > 0 if setpoint < measure < max
```

I do not understand why the setpoint is set to the upper bound on raising while the measurement is within the band.
If I understand correctly, the error should be 0 within the band to keep the integrator constant, like in:
```
# measure  # setpoint # resultingError #
########################################
#  < min   # max      #   > 2*band     #
# between  # measure  #      0         #
#  > max   # min      #   < -2*band    #
```

kizniche, 2022-11-06:
Closing because a fix has been committed.

closed 2022-11-06 by kizniche