kizniche/Mycodo #1145: Server 500's when loading "Function" page after upgrading directly from 8.8.8 to 8.12.9

AKAMEDIASYSTEM, 2022-01-23:
Hi! Thanks again for making Mycodo, it rocks.

After upgrading from 8.8.8 to the latest (ie, skipping several releases in between), clicking the "Function" link on the "Setup" drop-down results in an Error 500 from the server (full server output appended below)

### Versions:

 - Mycodo Version: *[e.g. 8.0.3]*
 - Raspberry Pi Version: *[e.g. 3B+]*
 - Raspbian OS Version: *[e.g. Buster Lite]*

### Reproducibility

Please list specific setup details that are involved and the steps to reproduce the behavior:

1) Have a functioning setup using Mycodo 8.8.8
2) Turn everything off for several months because you needed the space for another project
3) Turn the system on again, immediately upgrading from 8.8.8 to 8.12.9
4) Try to access the "Functions" page to alter the timing of a function (that is still successfully running after the upgrade - just wanted to change its interval)
5) Get 500 page form server

### Expected behavior

Able to load Function page and alter settings

### Additional context

Here is the content of the 500 message:

```
Version: 8.12.9
Database: 0187ea22dc4b
Model: Raspberry Pi 3 Model B Rev 1.2

Release:
Distributor ID: Raspbian
Description: Raspbian GNU/Linux 10 (buster)
Release: 10
Codename: buster

Firmware:
Sep 2 2020 21:16:35
Copyright (c) 2012 Broadcom
version 4439d2aaa6c376a2d1ef4402f142e1cf4de37c43 (clean) (release) (start_x)

Error (Full Traceback):

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 2073, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1518, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_restx/api.py", line 672, in error_router
    return original_handler(e)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1516, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1502, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**req.view_args)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_login/utils.py", line 272, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_function.py", line 609, in page_function
    user=user)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/templating.py", line 150, in render_template
    ctx.app,
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/templating.py", line 128, in _render
    rv = template.render(context)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/jinja2/environment.py", line 1291, in render
    self.environment.handle_exception()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/jinja2/environment.py", line 925, in handle_exception
    raise rewrite_traceback_stack(source=source)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/pages/function.html", line 3, in top-level template code
    {% set help_page = ["https://kizniche.github.io/Mycodo/Functions/", dict_translation['function']['title']] %}
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/layout.html", line 384, in top-level template code
    {%- block body %}{% endblock -%}
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/pages/function.html", line 121, in block 'body'
    {% include 'pages/function_options/function_entry.html' %}
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/pages/function_options/function_entry.html", line 40, in top-level template code
    {% include 'pages/function_options/function_options.html' %}
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/pages/function_options/function_options.html", line 58, in top-level template code
    {% include 'pages/function_options/actions.html' %}
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/pages/function_options/actions.html", line 15, in top-level template code
    {% include 'pages/function_options/action_options/'+each_action.action_type+'.html' %}
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/templating.py", line 59, in get_source
    return self._get_source_fast(environment, template)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/templating.py", line 95, in _get_source_fast
    raise TemplateNotFound(template)
jinja2.exceptions.TemplateNotFound: pages/function_options/action_options/infrared_send.html
```

kizniche, 2022-01-23:
The IR action doesn't exist in the latest version. This was not accounted for in upgrades. To fix this, you will need to downgrade, delete the action, then upgrade again.

kizniche, 2022-01-23:
Alternatively, I can make a fix to prevent the error, so the page will load, and you can delete the offending Action. This could take a while to implement and you would have to upgrade to the master branch to use the code.

closed 2022-01-23 by kizniche

AKAMEDIASYSTEM, 2022-01-23:
As always, thanks so much for fast and helpful answers!

Totally makes sense re: irsend going away, that's what I suspected based on the error output but I wondered if I had missed a mitigating action that would've occurred if I went through all the upgrades one by one.

I'll try to downgrade and then re-upgrade (I assume I just restore my last backup?), no need to do custom work on my behalf.

Thanks again,

AKA

kizniche, 2022-01-23:
I must have missed that when considering upgrades. The best solution I can think of at the moment is to have better error-handling, so missing templates don't prevent the page from rendering. Then at least users can delete the offending action with the missing template.

AKAMEDIASYSTEM, 2022-01-23:
That sounds like a good improvement for future releases, no rush on my part though, I am happy to try the downgrade/upgrade (although now I'm getting a worrying error `sqlite3.OperationalError: no such column: users.code` after trying to restore from my last 8.8.8 backup, but I will look around for clues etc before I bug you again)

AKAMEDIASYSTEM, 2022-01-23:
I was impatient! sqlite error resolved, back up and running on 8.8.8 now...

kizniche, 2022-01-23:
Just pushed a fix, at least for the Function Action templates. I need to extend it to all templates that are included on pages.