kizniche/Mycodo #1147: Install Dependencies for I/O Expander: MCP23017 (16 Channels): On/Off #1095

ePortageLLC, 2022-01-28:
I have run into the same is as described and resolved in #1095.  
Checking my version: 
No upgrade is available. You are running the latest release, version 8.12.9

I attempted loading MCP4728 as a dependency with no luck. 

Question: is the resolution to this issue in version 8.12.9?

Thank you  - 
Peter

kizniche, 2022-01-28:
Yes, Mycodo v8.12.9 fixed this issue. What do you mean you have the same issue? Can you describe the issue?

ePortageLLC, 2022-01-28:
Thank you for getting back (so fast)

I’ve added the output:  Output Configuration: I/O Expander: MCP23017 (16 Channels): On/Off.  

It required a library or dependency install which I approved.  At the time I did not look closely at what was being performed and I’ve not been able to see the request again.  I assume the output particulars have been satisfied.

I now see the output available in the list of outputs. 

However, When the 16 outputs are listed each channel is Unconfigured and can not be changed to On or Off.

This is the error reported when attempting to turn on or off:
Can not manipulate Output - Output: not set up.  (see attached screen shot)

I then tried loading MCP4728, as described in the prior issue report, with no success.

I was not able to see this error in the Daemon log.

Thoughts?

[image]

kizniche, 2022-01-28:
You can retrieve the upgrade log on the Log Page. Also, review the dameon log on the same page for exeact erorr that produces that popup error.

ePortageLLC, 2022-01-28:
The upgrade log -last few entries reads: (no errors)

[2021-12-10 05:37:24] Stage 3 execution time summary:
[2021-12-10 05:37:24] initialize:                   5 s
[2021-12-10 05:37:24] update-swap-size:             0 s
[2021-12-10 05:37:24] setup-virtualenv:             0 s
[2021-12-10 05:37:24] update-apt:                   7 s
[2021-12-10 05:37:24] update-packages:              21 s
[2021-12-10 05:37:24] web-server-update:            4 s
[2021-12-10 05:37:24] update-logrotate:             0 s
[2021-12-10 05:37:24] update-pip3:                  7 s
[2021-12-10 05:37:24] update-pip3-packages:         53 s
[2021-12-10 05:37:24] update-dependencies:          27 s
[2021-12-10 05:37:24] update-influxdb:              0 s
[2021-12-10 05:37:24] update-alembic:               2 s
[2021-12-10 05:37:24] update-alembic-post:          1 s
[2021-12-10 05:37:24] update-mycodo-startup-script: 1 s
[2021-12-10 05:37:24] compile-translations:         4 s
[2021-12-10 05:37:24] generate-widget-html:         2 s
[2021-12-10 05:37:24] update-cron:                  1 s
[2021-12-10 05:37:24] update-permissions:           1 s
[2021-12-10 05:37:24] restart-daemon:               3 s
[2021-12-10 05:37:24] web-server_reload:            5 s
[2021-12-10 05:37:24] web-server-connect:           5 s
[2021-12-10 05:37:24] 
[2021-12-10 05:37:24] #### Completed Upgrade Stage 3 of 3 in 149 seconds ####
[2021-12-10 05:37:24] 
[2021-12-10 05:37:24] Upgrade completed. Review the log to ensure no critical errors were encountered

Below is what I found in the Daemon log - I searched for error: mycodo.outputs.mcp23017_123760b4 

2022-01-28 17:18:59,494 - INFO - mycodo.controllers.controller_conditional_20d214af - This INFO log entry will appear in the Daemon Log
2022-01-28 17:18:59,494 - ERROR - mycodo.controllers.controller_conditional_20d214af - This ERROR log entry will appear in the Daemon Log
2022-01-28 17:18:59,558 - INFO - mycodo.controllers.controller_conditional_20d214af - Check this measurement in the Daemon Log. The value is 0.0
**2022-01-28 17:19:01,754 - ERROR - mycodo.outputs.mcp23017_123760b4 - Cannot manipulate Output 123760b4-359b-4896-be4e-73228bbf9856: Output not set up.**
2022-01-28 17:19:04,675 - INFO - mycodo.controllers.controller_conditional_20d214af - This INFO log entry will appear in the Daemon Log
2022-01-28 17:19:04,675 - ERROR - mycodo.controllers.controller_conditional_20d214af - This ERROR log entry will appear in the Daemon Log

Here's the error in the log:
2022-01-28 17:18:59,494 - INFO - mycodo.controllers.controller_conditional_20d214af - This INFO log entry will appear in the Daemon Log
2022-01-28 17:18:59,494 - ERROR - mycodo.controllers.controller_conditional_20d214af - This ERROR log entry will appear in the Daemon Log
2022-01-28 17:18:59,558 - INFO - mycodo.controllers.controller_conditional_20d214af - Check this measurement in the Daemon Log. The value is 0.0
2022-01-28 17:19:01,754 - ERROR - mycodo.outputs.mcp23017_123760b4 - Cannot manipulate Output 123760b4-359b-4896-be4e-73228bbf9856: Output not set up.
2022-01-28 17:19:04,675 - INFO - mycodo.controllers.controller_conditional_20d214af - This INFO log entry will appear in the Daemon Log
2022-01-28 17:19:04,675 - ERROR - mycodo.controllers.controller_conditional_20d214af - This ERROR log entry will appear in the Daemon Log

I'm not exactly sure what I'm looking for so I appreciate you walking me through this.  I suspect you need more information.
Regards

kizniche, 2022-01-28:
Sorry, I meant dependency log for the dependency install.

kizniche, 2022-01-28:
You also aren't showing where the output issue occurred. Like I mentioed previously, you will need to find when the daemon first started, ebcause this is where the output controller starts. If you can't just restart the daemon and it will be at the base fo th elog again.

ePortageLLC, 2022-01-29:
Output log below - I will dig into the daemon today.

Last 100 lines of /var/log/mycodo/mycododependency.log:
[2021-11-17 13:27:27] Setting up libopencv-photo4.5:armhf (4.5.1+dfsg-5) ...
[2021-11-17 13:27:27] Setting up libhdf5-hl-100:armhf (1.10.6+repack-4+deb11u1) ...
[2021-11-17 13:27:27] Setting up libspatialite7:armhf (5.0.1-2) ...
[2021-11-17 13:27:27] Setting up libopencv-features2d4.5:armhf (4.5.1+dfsg-5) ...
[2021-11-17 13:27:27] Setting up libnetcdf18:armhf (1:4.7.4-1) ...
[2021-11-17 13:27:27] Setting up libopencv-calib3d4.5:armhf (4.5.1+dfsg-5) ...
[2021-11-17 13:27:27] Setting up libopencv-shape4.5:armhf (4.5.1+dfsg-5) ...
[2021-11-17 13:27:27] Setting up libopencv-video4.5:armhf (4.5.1+dfsg-5) ...
[2021-11-17 13:27:27] Setting up libopencv-stitching4.5:armhf (4.5.1+dfsg-5) ...
[2021-11-17 13:27:27] Setting up libopencv-objdetect4.5:armhf (4.5.1+dfsg-5) ...
[2021-11-17 13:27:27] Setting up odbcinst (2.3.6-0.1+b1) ...
[2021-11-17 13:27:27] Setting up odbcinst1debian2:armhf (2.3.6-0.1+b1) ...
[2021-11-17 13:27:27] Setting up libgdal28 (3.2.2+dfsg-2) ...
[2021-11-17 13:27:27] Setting up libopencv-imgcodecs4.5:armhf (4.5.1+dfsg-5) ...
[2021-11-17 13:27:27] Setting up libopencv-highgui4.5:armhf (4.5.1+dfsg-5) ...
[2021-11-17 13:27:28] Setting up libopencv-videoio4.5:armhf (4.5.1+dfsg-5) ...
[2021-11-17 13:27:28] Setting up libopencv-contrib4.5:armhf (4.5.1+dfsg-5) ...
[2021-11-17 13:27:28] Setting up python3-opencv (4.5.1+dfsg-5) ...
[2021-11-17 13:27:28] Processing triggers for libc-bin (2.31-13+rpt2+rpi1) ...
[2021-11-17 13:27:28] Processing triggers for man-db (2.9.4-2) ...

[2021-11-17 13:27:30] End install of python3-opencv

[2021-11-17 13:27:30] 
[2021-11-17 13:27:30] #### Setting permissions

[2021-11-17 13:27:32] #### Dependencies installed. Restarting frontend and backend...

[2021-11-17 13:27:36] #### Dependency install complete.

[2022-01-28 11:02:44] Dependency installation beginning. Installing: adafruit_mcp230xx

[2022-01-28 11:02:44] 
[2022-01-28 11:02:44] #### Installing/updating adafruit-circuitpython-mcp230xx==2.4.6 (pip-pypi)
[2022-01-28 11:02:46] Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
[2022-01-28 11:02:48] Collecting adafruit-circuitpython-mcp230xx==2.4.6
[2022-01-28 11:02:48]   Downloading https://www.piwheels.org/simple/adafruit-circuitpython-mcp230xx/adafruit_circuitpython_mcp230xx-2.4.6-py3-none-any.whl (12 kB)
[2022-01-28 11:02:48] Requirement already satisfied: adafruit-circuitpython-busdevice in ./env/lib/python3.9/site-packages (from adafruit-circuitpython-mcp230xx==2.4.6) (5.1.0)
[2022-01-28 11:02:48] Requirement already satisfied: Adafruit-Blinka in ./env/lib/python3.9/site-packages (from adafruit-circuitpython-mcp230xx==2.4.6) (6.15.0)
[2022-01-28 11:02:48] Requirement already satisfied: Adafruit-PureIO>=1.1.7 in ./env/lib/python3.9/site-packages (from Adafruit-Blinka->adafruit-circuitpython-mcp230xx==2.4.6) (1.1.8)
[2022-01-28 11:02:48] Requirement already satisfied: rpi-ws281x>=4.0.0 in ./env/lib/python3.9/site-packages (from Adafruit-Blinka->adafruit-circuitpython-mcp230xx==2.4.6) (4.3.1)
[2022-01-28 11:02:48] Requirement already satisfied: sysv-ipc>=1.1.0 in ./env/lib/python3.9/site-packages (from Adafruit-Blinka->adafruit-circuitpython-mcp230xx==2.4.6) (1.1.0)
[2022-01-28 11:02:48] Requirement already satisfied: Adafruit-PlatformDetect>=3.13.0 in ./env/lib/python3.9/site-packages (from Adafruit-Blinka->adafruit-circuitpython-mcp230xx==2.4.6) (3.17.2)
[2022-01-28 11:02:48] Requirement already satisfied: pyftdi>=0.40.0 in ./env/lib/python3.9/site-packages (from Adafruit-Blinka->adafruit-circuitpython-mcp230xx==2.4.6) (0.53.3)
[2022-01-28 11:02:48] Requirement already satisfied: RPi.GPIO in /usr/lib/python3/dist-packages (from Adafruit-Blinka->adafruit-circuitpython-mcp230xx==2.4.6) (0.7.0)
[2022-01-28 11:02:48] Requirement already satisfied: pyusb!=1.2.0,>=1.0.0 in ./env/lib/python3.9/site-packages (from pyftdi>=0.40.0->Adafruit-Blinka->adafruit-circuitpython-mcp230xx==2.4.6) (1.1.1)
[2022-01-28 11:02:48] Requirement already satisfied: pyserial>=3.0 in ./env/lib/python3.9/site-packages (from pyftdi>=0.40.0->Adafruit-Blinka->adafruit-circuitpython-mcp230xx==2.4.6) (3.5)
[2022-01-28 11:02:50] Installing collected packages: adafruit-circuitpython-mcp230xx
[2022-01-28 11:02:50] Successfully installed adafruit-circuitpython-mcp230xx-2.4.6

[2022-01-28 11:02:52] End install of adafruit_mcp230xx

[2022-01-28 11:02:52] 
[2022-01-28 11:02:52] #### Setting permissions

[2022-01-28 11:02:53] Dependency installation beginning. Installing: 

[2022-01-28 11:02:54] 
[2022-01-28 11:02:54] #### Setting permissions

[2022-01-28 11:02:56] #### Dependencies installed. Restarting frontend and backend...
[2022-01-28 11:02:56] #### Dependencies installed. Restarting frontend and backend...

[2022-01-28 11:03:01] #### Dependency install complete.

[2022-01-28 11:03:01] #### Dependency install complete.

[... 3 lines cut ...]
[2022-01-28 15:06:30] #### Installing/updating adafruit-circuitpython-mcp4728==1.2.0 (pip-pypi)
[2022-01-28 15:06:32] Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
[2022-01-28 15:06:33] Collecting adafruit-circuitpython-mcp4728==1.2.0
[2022-01-28 15:06:33]   Downloading https://www.piwheels.org/simple/adafruit-circuitpython-mcp4728/adafruit_circuitpython_mcp4728-1.2.0-py3-none-any.whl (8.3 kB)
[... cut ...]
[2022-01-28 15:06:33] Requirement already satisfied: pyserial>=3.0 in ./env/lib/python3.9/site-packages (from pyftdi>=0.40.0->Adafruit-Blinka->adafruit-circuitpython-mcp4728==1.2.0) (3.5)
[2022-01-28 15:06:35] Installing collected packages: adafruit-circuitpython-mcp4728
[2022-01-28 15:06:35] Successfully installed adafruit-circuitpython-mcp4728-1.2.0

[2022-01-28 15:06:35] End install of adafruit_mcp4728

[2022-01-28 15:06:36] 
[2022-01-28 15:06:36] #### Setting permissions

[2022-01-28 15:06:37] #### Dependencies installed. Restarting frontend and backend...

[2022-01-28 15:06:40] #### Dependency install complete.

kizniche, 2022-01-30:
You're still not showing the relevant daemon log lines. You need to find when it starts, as I've mentioned, the output controller starts when the daemon starts. If you cannot provide the log lines when the daemon starts, your issue cannot possibly be diagnosed.

ePortageLLC, 2022-01-30:
This required some remedial work on my part to familiarize myself with the daemon.  The output below was generated after a executing "restart system" on the configure page.  I hope this helps.  Thank You.

**2022-01-30 08:01:35,754 - INFO - mycodo.daemon - Mycodo daemon v8.12.9 starting**
2022-01-30 08:01:36,497 - INFO - mycodo.outputs.on_off_gpio_a1448432 - Output setup on pin 12 and turned OFF (ON=HIGH)
2022-01-30 08:01:36,497 - INFO - mycodo.outputs.on_off_gpio_a1448432 - Initialized in 154.8 ms
2022-01-30 08:01:36,622 - INFO - mycodo.outputs.on_off_gpio_155d0166 - Output setup on pin 20 and turned OFF (ON=HIGH)
2022-01-30 08:01:36,622 - INFO - mycodo.outputs.on_off_gpio_155d0166 - Initialized in 123.7 ms
2022-01-30 08:01:36,747 - INFO - mycodo.outputs.on_off_gpio_50b8bee1 - Output setup on pin 19 and turned OFF (ON=HIGH)
2022-01-30 08:01:36,748 - INFO - mycodo.outputs.on_off_gpio_50b8bee1 - Initialized in 124.5 ms
2022-01-30 08:01:36,879 - INFO - mycodo.outputs.on_off_gpio_8199c1b7 - Output setup on pin 13 and turned OFF (ON=HIGH)
2022-01-30 08:01:36,879 - INFO - mycodo.outputs.on_off_gpio_8199c1b7 - Initialized in 130.3 ms
2022-01-30 08:01:37,006 - INFO - mycodo.outputs.on_off_gpio_06a3caf8 - Output setup on pin 21 and turned OFF (ON=HIGH)
2022-01-30 08:01:37,006 - INFO - mycodo.outputs.on_off_gpio_06a3caf8 - Initialized in 125.8 ms
2022-01-30 08:01:37,143 - INFO - mycodo.outputs.on_off_gpio_be6a3990 - Output setup on pin 22 and turned OFF (ON=HIGH)
2022-01-30 08:01:37,144 - INFO - mycodo.outputs.on_off_gpio_be6a3990 - Initialized in 136.1 ms
2022-01-30 08:01:37,263 - INFO - mycodo.outputs.on_off_gpio_936a6d97 - Output setup on pin 24 and turned ON (ON=HIGH)
2022-01-30 08:01:37,264 - INFO - mycodo.outputs.on_off_gpio_936a6d97 - Initialized in 118.8 ms
2022-01-30 08:01:37,428 - INFO - mycodo.outputs.on_off_python_a47c7e94 - ID: a47c7e94-a9b6-49da-8cbf-20e071cd1fc3: OFF
2022-01-30 08:01:37,428 - INFO - mycodo.outputs.on_off_python_a47c7e94 - Initialized in 163.1 ms
2022-01-30 08:01:37,551 - INFO - mycodo.outputs.on_off_gpio_5ef1f849 - Output setup on pin 14 and turned OFF (ON=HIGH)
2022-01-30 08:01:37,551 - INFO - mycodo.outputs.on_off_gpio_5ef1f849 - Initialized in 122.0 ms
2022-01-30 08:01:37,748 - ERROR - mycodo.outputs.mcp23017_123760b4 - Could not set up output
Traceback (most recent call last):
  File "/var/mycodo-root/env/lib/python3.9/site-packages/adafruit_bus_device/i2c_device.py", line 154, in __probe_for_device
    self.i2c.writeto(self.device_address, b"")
  File "/var/mycodo-root/env/lib/python3.9/site-packages/busio.py", line 166, in writeto
    return self._i2c.writeto(address, buffer, stop=stop)
  File "/var/mycodo-root/env/lib/python3.9/site-packages/adafruit_blinka/microcontroller/generic_linux/i2c.py", line 49, in writeto
    self._i2c_bus.write_bytes(address, buffer[start:end])
  File "/var/mycodo-root/env/lib/python3.9/site-packages/Adafruit_PureIO/smbus.py", line 308, in write_bytes
    self._device.write(buf)
OSError: [Errno 121] Remote I/O error

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/var/mycodo-root/env/lib/python3.9/site-packages/adafruit_bus_device/i2c_device.py", line 160, in __probe_for_device
    self.i2c.readfrom_into(self.device_address, result)
  File "/var/mycodo-root/env/lib/python3.9/site-packages/busio.py", line 156, in readfrom_into
    return self._i2c.readfrom_into(address, buffer, stop=stop)
  File "/var/mycodo-root/env/lib/python3.9/site-packages/adafruit_blinka/microcontroller/generic_linux/i2c.py", line 56, in readfrom_into
    readin = self._i2c_bus.read_bytes(address, end - start)
  File "/var/mycodo-root/env/lib/python3.9/site-packages/Adafruit_PureIO/smbus.py", line 179, in read_bytes
    return self._device.read(number)
OSError: [Errno 121] Remote I/O error

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/Mycodo/mycodo/outputs/mcp23017.py", line 146, in setup_output
    self.sensor = MCP23017(
  File "/var/mycodo-root/env/lib/python3.9/site-packages/adafruit_mcp230xx/mcp23017.py", line 49, in __init__
    super().__init__(i2c, address)
  File "/var/mycodo-root/env/lib/python3.9/site-packages/adafruit_mcp230xx/mcp230xx.py", line 31, in __init__
    self._device = i2c_device.I2CDevice(i2c, address)
  File "/var/mycodo-root/env/lib/python3.9/site-packages/adafruit_bus_device/i2c_device.py", line 50, in __init__
    self.__probe_for_device()
  File "/var/mycodo-root/env/lib/python3.9/site-packages/adafruit_bus_device/i2c_device.py", line 163, in __probe_for_device
    raise ValueError("No I2C device at address: 0x%x" % self.device_address)
ValueError: No I2C device at address: 0x20
2022-01-30 08:01:37,756 - INFO - mycodo.outputs.mcp23017_123760b4 - Initialized in 203.1 ms
2022-01-30 08:01:37,756 - INFO - mycodo.controllers.controller_output - Activated in 1696.0 ms
2022-01-30 08:01:38,420 - INFO - mycodo.controllers.controller_conditional_228c7cd5 - Activated in 44.7 ms
2022-01-30 08:01:38,572 - INFO - mycodo.controllers.controller_conditional_9b65f5ce - Activated in 45.8 ms
2022-01-30 08:01:38,747 - INFO - mycodo.controllers.controller_conditional_37e4f43a - Activated in 40.2 ms
2022-01-30 08:01:38,898 - INFO - mycodo.controllers.controller_conditional_20d214af - Activated in 38.9 ms
2022-01-30 08:01:38,898 - INFO - mycodo.daemon - All activated Conditional controllers started
[... cut ...]
2022-01-30 08:05:20,404 - DEBUG - mycodo.controllers.controller_input_45b57277 - Adding measurements to InfluxDB with ID 45b57277-2fa4-47af-b1b2-94ac6607bba0: {0: {'measurement': None, 'unit': 'F', 'value': 72.14, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 95.9, 'timestamp_utc': None}, 2: {'measurement': None, 'unit': 'F', 'value': 70.90396, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 110.37537124612825, 'timestamp_utc': None}}

kizniche, 2022-01-30:
The error indicates the device isn't detected. Does it appear in the deivce tree when you execute ```sudo i2cdetect -y 0``? This is also printed on the System Information page. Have you ever got this device to work outside Mycodo?

ePortageLLC, 2022-01-31:
You were correct.  The eval board I was using evidently gave up the ghost.  Two have been ordered from DigiKey and should arrive later this week.  I'll provide an update after one has been installed.  Thank You again!

ePortageLLC, 2022-02-04:
That was it - We're good to go here.  Thanks Again!

closed 2022-03-28 by kizniche